adafruit/adafruit-beaglebone-io-python #192: Typo in pin ID

fivdi, 2017-11-15:
The following [line of code from commin.c](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/1.0.7/source/common.c#L304) defines ehrpwm0 channel 1 as being available on P1_13 on a PocketBeagle.
```c
  { "ehrpwm0", 1, 1, 1, "ehrpwm.0:1", "EHRPWM0B", "48300000", "48300200", "P1_13"},
```
However, P1_13 is reserved for USB1-ID. It looks like there is a typo here and P1_13 should be P1_33.
```c
  { "ehrpwm0", 1, 1, 1, "ehrpwm.0:1", "EHRPWM0B", "48300000", "48300200", "P1_33"},
```

pdp7, 2017-11-15:
@fivdi thanks for opening this issue.

@RobertCNelson could you take a look?

RobertCNelson, 2017-11-15:
Confirmed p1_13 is the usb pin, looks like a day finger write by me.. traveling so I can only asked the change.

Regards

pdp7, 2017-11-17:
@fivdi  @RobertCNelson Thanks, I've pushed b328cddd13504c137265a417dce28353b07119eb to fix the typo

closed 2017-11-17 by pdp7