adafruit/adafruit-beaglebone-io-python #193: Not able to control some of the GPIOs on PocketBeagle using Adafruit library

eshtaartha, 2017-11-16:
I am trying to control some of the GPIO pins of PocketBeagle using a simple test case:

```
import Adafruit_BBIO.GPIO as GPIO

GPIO.setup("P1_8", GPIO.OUT)

GPIO.output("P1_8", GPIO.HIGH)

GPIO.cleanup()
```

The GPIOs that do not work with the above test case are:
P1.3, P1.6, P1.8, P1.10, P1.12, P1.26, P1.28, P1.30, P1.32, P1.36
P2.1, P2.5, P2.7, P2.9, P2.11, P2.25, P2.27, P2.29, P2.31.

I installed the recommended Debian Image and Adafruit_BBIO library using the steps given in 
[https://github.com/adafruit/adafruit-beaglebone-io-python](url)

I have not installed Linux Kernel 4.14.x though. Any ideas?

My kernel version and python version are as follows:

[image]

fivdi, 2017-11-16:
This is probably happening because the pins are not in the correct mode. Below is a picture of the PocketBeagle expansion headers. In most cases the dashed lines in the picture show the default mode for a pin. For example, for pins P1_2 and P1_4 there are dashed lines around the cells 87 and 89 so these pins are in gpio mode. For pins P1_6, P1_8, P1_10, P1_12 the dashed lines are around the SPI0 pins so these pins are in spi mode. There are exceptions to this "dashed lines" rule. The picture indicates that the default mode for some of the pins is pru mode but this isn't the case as the default mode for the corresponding pins is actually gpio.

An additional point to be aware of is that the image indicates that GPIO109 is available on P1_3. This is not the case as P1_3 is reserved for usb1_vbus_out and can't be configured as a GPIO.

To get things to work configure the mode for the required pins to gpio before the python program is run. For example, to set the mode for P1_8 to gpio run the following command

```
config-pin P1_8 gpio
```

[image]

fivdi, 2017-11-16:
See also https://github.com/adafruit/adafruit-beaglebone-io-python/issues/158

eshtaartha, 2017-11-16:
@fivdi It works! Thank you for the detailed explanation.

closed 2017-11-16 by eshtaartha