adafruit/adafruit-beaglebone-io-python #194: Encoder position cannot be set

dplanella, 2017-11-16:
I'm using a rotary encoder based on the KY-040 module, with the `Adafruit_BBIO.Encoder.RotaryEncoder` class. I'm running the latest **IoT Debian image** on a **Beaglebone Black**:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.2 (stretch)
Release:	9.2
Codename:	stretch

$ uname -r
4.4.91-ti-r133

$ python --version
Python 2.7.13

$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[a5d5f437baeaa7f8e14d5ae037268abf7a67e88e]
eeprom:[A335BNLT00C03614BBBK0298]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.4.91-ti-r133]
nodejs:[v6.12.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171107.0-0rcnee1~stretch+20171107]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
```

To enable the QEP module on the hardware, I added these lines to `/boot/uEnv.txt` as instructed in the `Encoder` module source code and rebooted:

```
cmdline=coherent_pool=1M quiet cape_universal=enable
cape_enable=bone_capemgr.enable_partno=cape-universala
```

I then configured the `equep2` pins on the command line. I don't think this step is necessary, though, as the Encoder module seems to set them up as `qep` in code already.

```
$ config-pin P8.11 qep 
$ config-pin P8.12 qep 
```

And with these inputs connected to the encoder, I ran this simple code via an SSH connection: 

```
import time
import Adafruit_BBIO.Encoder as Encoder

encoder = Encoder.RotaryEncoder(Encoder.RotaryEncoder.EQEP2)

encoder.setAbsolute()
encoder.zero()

while True:
    print("Encoder position: " + encoder.getPosition())
    time.sleep(1)
```

- **Expected**: the position is 0 upon every start of the app
- **Actual**: the position is not 0 and is remembered upon app restarts

The `zero()` method calls the `setPosition()` method to set the position to 0. The issue here is that currently the `setPosition()` method does nothing else than printing debug output. The new position is not set. I'm assuming that it's because the python function to write sysfs files has not yet been implemented?

Just for comparison, I tested the original PyBBIO code, which is the same except for the uncommented part that does set the position, and has a Python API for writing to sysfs. Unfortunately, that caused a segmentation fault.  

Thanks!

pdp7, 2017-11-17:
> The zero() method calls the setPosition() method to set the position to 0. The issue here is that currently the setPosition() method does nothing else than printing debug output. The new position is [...]

Yeah, this is something I still need to implement.  Thanks for bringing it up.

dplanella, 2017-11-17:
@pdp7 I'm not sure if you were thinking of an implementation in C or Python, but there's a Python module to read/write sysfs attributes at https://github.com/ponycloud/python-sysfs that could be imported into the project, at least as an incremental enhancement if later replaced by a C implementation.

It is MIT-licensed, so I assume it should be compatible with the code in this repo.

pdp7, 2017-11-18:
@dplanella thanks for pointing this out

dplanella, 2017-11-24:
WIP, but working branch at https://github.com/dplanella/adafruit-beaglebone-io-python/blob/encoder-sysfs/Adafruit_BBIO/Encoder.py.

Will wait for PR #201 to be reviewed and approved before submitting, as it depends on it.

pdp7, 2017-11-28:
@dplanella can you create a PR from your working branch?

pdp7, 2017-11-30:
Thanks, I've merged #205

pdp7, 2017-11-30:
@dplanella can this issue be closed now?

dplanella, 2017-11-30:
@pdp7 if the project's policy is to close issues once the fixes have been committed ahead of a release (i.e. "fix committed" vs. "fix released"), then +1. I'd also say issue #122 could be closed along the way.

closed 2017-12-01 by pdp7