adafruit/adafruit-beaglebone-io-python #204: Python - Beagle Bone - Timer

miliy, 2017-11-28:
Hi everyone, 
I want to configure a timer on green beagle bone.
I only found library to configure PWM, how can I configure a timer on beagle bone using python please?
Thank you for you help, 
Regards

pdp7, 2017-11-28:
@miliy hi, support for the AM3358 timer peripherals is not currently implemented in Adafruit_BBIO.

It might possible to add support, but I'd like to know more about your use case.  Could you describe how you intend to use timers in your project?

miliy, 2017-11-28:
Hi, 
Thank you for your quick answer.
I want to measure the frequency of a square signal. (signal command- LED blinking at F=3Hz) Initially I wanted to use ADC, but I am not sure it is best.
If I can do that using the ADC on the beagle bone, what is the best way to implement it please?
Thank you

pdp7, 2017-11-28:
@miliy I'm not sure of the best solution but I have a started a discussion on the [BeagleBoard.org community mailing list](https://beagleboard.org/discuss):

**["Measure frequency of a square wave input without PRU?"](https://groups.google.com/d/msg/beagleboard/c46KI__VONw/Hq7bOXi9AgAJ)**

If C code can be written to achieve this goal from the ARM processor, then it may be possible to include that functionality in Adafruit_BBIO.  I'm going to close this issue for now.  A new `enhancement` issue should be opened once there is a proof of concept in C.

NOTE: it appears that there is a solution using `libpruio` library but this is unrelated to the `Adafruit_BBIO` library:

["How to read PWM in BBB GPIO....?"](https://groups.google.com/d/msg/beagleboard/QawyZ23J6sM/FpaYePlAAwAJ)
> You don't need GPIOs. Instead analyse the pulse train by hardware. The eCAP modules in the PWMSS subsystems are designed for that purpose. The simplest (and most powerful) solution is to use libpruio. Check out the example pwm_cap.

pdp7, 2017-11-28:
@miliy oops, I misread what you wrote, I was thinking of 3kHz but you actually wrote 3Hz.  That would be a period of 333 milliseconds which is much easier to handle.  I believe both GPIO input or ADC would work.

Could you please post a schematic and/or a photo of how the components are wired up to the BeagleBone?

miliy, 2017-11-29:
[image]

Goog Morning, 
Please find attached, I hope it will help.
Regards

pdp7, 2017-11-30:
@miliy thanks, looking at the schematic, I am thinking it might be simpler to connect a GPIO input line to the BeagleBone in parallel to the LED.  It looks like this would be it would normally be pulled up to 3.3V.  The V_LED_OK will pull the GPIO input down when active.

I think you could then use the `GPIO.wait_for_edge()` to detect when the input line changes.

The ADC method that you have drawn in the schematic also would work I believe, but it is more complex than needed if it just a GND to 3.3V digital signal.

pdp7, 2018-01-26:
@miliy  please re-open if this is still an issue

closed 2018-01-26 by pdp7