adafruit/adafruit-beaglebone-io-python #206: GPIO setup to in problems

Grincas, 2017-11-30:
Hello i have a problem with GPIO.  I cant set up  on input mode.  Get this error:  ValueError: Set gpio mode failed, missing file or invalid permissions.

 `import Adafruit_BBIO.GPIO as GPIO
import sys
import time

def main(argv):

    Frequency = 10
    # CS
    GPIO.setup("P9_28", GPIO.OUT)
    GPIO.output("P9_28", GPIO.HIGH)
    # CLK
    GPIO.setup("P9_31", GPIO.OUT)
    # MOSI
    GPIO.setup("P9_29", GPIO.OUT)
    # MISO
    GPIO.setup("P8_14",GPIO.IN)

    count = 0
    while (count < 9):
        if GPIO.input("P9_27"):
            break
        GPIO.output("P9_28", GPIO.LOW)
        time.sleep(5)

        GPIO.output("P9_28", GPIO.HIGH)
        time.sleep(5)

        count = count + 1

    GPIO.output("P9_28", GPIO.HIGH)
    print "Good bye!"

if __name__ == "__main__":
    main(sys.argv)
`

sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[2ce750d881941c5189db9e189af90517e11c079f]
eeprom:[A335BNLTEIA04816BBBK376B]
dogtag:[BeagleBoard.org Debian Image 2017-08-31]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-rc2-00002-g7c9353]
kernel:[4.4.84-ti-r120]
nodejs:[v6.11.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-UART2-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20170728.0-0rcnee1~stretch+20170728]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]

pdp7, 2017-12-01:
Please try running your Python program as root using sudo.

Alternatively, you can run the [Debian 9.2 "Stretch" iot (2017-10-29)](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#microSD.2FStandalone:_.28stretch-iot.29_.28All_BeagleBone_Variants_.26_PocketBeagle.29) image and install [Linux kernel](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Kernel_Options) [4.14.x](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Mainline_.284.14.x_lts.29) to enable [non-root control of GPIO](https://github.com/rcn-ee/repos/blob/master/bb-customizations/suite/stretch/debian/80-gpio-noroot.rules) and [PWM](https://github.com/rcn-ee/repos/blob/master/bb-customizations/suite/stretch/debian/81-pwm-noroot.rules) [_(commit)_](https://github.com/adafruit/adafruit-beaglebone-io-python/commit/b65cbf8e41b444bad7c4ef6cfd4f88a30210fd78)

Grincas, 2017-12-01:
I have tried  to run as root get the same error , i look for a file in sys directory and i sow that pin have been changed to a input but get that error , so solution to downgrade from 4.4 to 4.1 ?

Grincas, 2017-12-01:
root@beaglebone:/# sudo python SPI_new.py
Traceback (most recent call last):
  File "SPI_new.py", line 18, in main
    GPIO.setup("P8_14",GPIO.IN)
ValueError: Set gpio mode failed, missing file or invalid permissions. 

Out put is the same

pdp7, 2017-12-01:
Please upgrade to the 4.14.x kernel, reboot and try running the Python program again:
```
sudo apt-get update
sudo apt-get upgrade
cd /opt/scripts/tools/update_kernel.sh
git pull
sudo ./update_kernel.sh --lts-4_14
```

If you get the error message again, then please try running strace to see what /sys files it is trying to open:
```
sudo strace -e open python ./SPI_new.py 2>&1 | grep /sys/
```

If strace is not installed, then run:
```
sudo apt-get install strace
```

pdp7, 2017-12-02:
@Grincas Also, I noticed in your example code at the top of this issue that `P9_27` is used as input but never setup:
```
    if GPIO.input("P9_27"):
```
There needs to be a call to `GPIO.setup()` first such as:
```
GPIO.setup("P9_27")
```

rd-tyler, 2018-01-23:
Hi @pdp7 ,

Not to interject but I too am experiencing some issues and they are likely related to this issue as they are reporting similar errors to @Grincas .

When trying to run the prebuilt Adafruit_CharLCD example "char_lcd_rgb_pwm.py" as shown here from the Cloud9 IDE:
https://github.com/adafruit/Adafruit_Python_CharLCD/blob/master/examples/char_lcd_rgb_pwm.py

I get the following:
```
debian@beaglebone:~/Adafruit_Python_CharLCD/examples$ sudo ./char_lcd_rgb_pwm.py
Traceback (most recent call last):
  File "./char_lcd_rgb_pwm.py", line 71, in <module>
    enable_pwm=True)
  File "build/bdist.linux-armv7l/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 367, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 152, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/GPIO.py", line 278, in setup
ValueError: Set gpio value failed, missing file or invalid permissions.
```

Same for if I attempt to execute using:
```
sudo python ./char_lcd_rgb_pwm.py
```

So, following your attempts at a repair I have attempted to update and upgrade everything, including the kernel as I'm currently using 4.4.91 as included in the latest image on the BBB main site.

When attempting to execute the update_kernel script I get the following:
```
debian@beaglebone:/opt/scripts/tools$ sudo ./update_kernel.sh --lts-4_14
info: checking archive
2018-01-23 23:47:42 URL:https://rcn-ee.com/repos/latest/stretch-armhf/LATEST-ti [136/136] -> "LATEST-ti" [1]
-----------------------------
Kernel Options:
ABI:1 LTS41 4.1.30-ti-r70
ABI:1 LTS44 4.4.110-ti-r143
ABI:1 LTS49 4.9.77-ti-r92
ABI:1 LTS414 4.14.14-ti-r27
-----------------------------
Kernel version options:
-----------------------------
LTS44: --lts-4_4
LTS49: --lts-4_9
LTS414: --lts-4_14
STABLE: --stable
TESTING: --testing
-----------------------------
info: you are running: [4.4.91-ti-r133], latest is: [4.14.14-ti-r27] updating...
Ign:1 http://deb.debian.org/debian stretch InRelease                          
Hit:2 http://repos.rcn-ee.com/debian stretch InRelease                        
Hit:3 http://deb.debian.org/debian stretch-updates InRelease
Hit:4 http://deb.debian.org/debian-security stretch/updates InRelease
Hit:5 http://deb.debian.org/debian stretch Release
Reading package lists... Done
Building dependency tree       
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Checking dpkg...
Checking apt-cache...
debug: installing: [linux-image-4.14.14-ti-r27]
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
```

Blindly following the direction to execute 
```
sudo dpkg --configure -a
```
just temporarily disconnects the beaglebone, resets it to the standard landing screen and doesn't actually resolve the issue as it persists and repeats the message. My initial attempts at finding a fix for that has yielded very little. Any insight you can offer would be greatly appreciated.

RobertCNelson, 2018-01-24:
@rd-tyler  please list what this shows:

```
apt list --upgradable
```

Regards,

rd-tyler, 2018-01-24:
Hi @RobertCNelson,

from home:
```
debian@beaglebone:~$ sudo apt list --upgradable
[sudo] password for debian:
Listing... Done
bb-node-red-installer/unknown 0.17.5-0rcnee5~stretch+20180120 armhf [upgradable from: 0.17.5-0rcnee0~stretch+20170816]
N: There is 1 additional version. Please use the '-a' switch to see it
```

RobertCNelson, 2018-01-24:
@rd-tyler that's strange:

try:

```
sudo apt remove bb-node-red-installer --purge
sudo apt update
sudo apt install bb-node-red-installer
```
Regards,

rd-tyler, 2018-01-24:
The plot thickens:

```
debian@beaglebone:~$ sudo apt remove bb-node-red-installer --purge
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
```

StokesMIDE, 2018-01-25:
I was having GPIO trouble as originally described, and I updated the kernel to 4.14 as per the instructions here. Not only did it fail to fix the problem, it made the UART overlays unusable; the system won't boot if they're loaded via /boot/uEnv.txt, and attempting to load them manually produces a similar seg fault. I've updated/upgraded, tried rebuilding the overlays, but it hasn't helped.

I know this is getting beyond the scope of the original issue, but it was this issue that led me here, and I'm utterly stuck.

(BTW, I also had the thing with upgrading bb-node-red-installer, but removing and reinstalling it worked for me.)

pdp7, 2018-01-26:
@StokesMIDE **Please create a new issue so we can troubleshoot your problem.**

In that issue, please post complete the Python source code that you are trying to run.  Use the "Insert code" button so that the whitespace is preserved.

Please also paste the output of these commands in that issue:
```
sudo /opt/scripts/tools/version.sh
```
```
cat /boot/uEnv.txt
```

If you are not already, I would recommend trying this image:
**[Debian 9.3 "Stretch" iot (2018-01-21)](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Stretch_Snapshot_iot)**

pdp7, 2018-01-26:
@rd-tyler  **Please create a new issue so we can troubleshoot your problem with `Adafruit_Python_CharLCD`**.

Please paste the output of these commands in that issue:
```
sudo /opt/scripts/tools/version.sh
```
```
cat /boot/uEnv.txt
```

If you are not already, I would recommend trying this image:
**[Debian 9.3 "Stretch" iot (2018-01-21)](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Stretch_Snapshot_iot)**

pdp7, 2018-01-26:
@Grincas **Are you still experiencing this issue?**

I noticed in your example code at the top of this issue that `P9_27` is used as input but never setup:
```
    if GPIO.input("P9_27"):
```
There needs to be a call to `GPIO.setup()` first such as:
```
GPIO.setup("P9_27")
```

If you get the error message again, then please try running strace to see what /sys files it is trying to open:
```
sudo strace -e open python ./SPI_new.py 2>&1 | grep /sys/
```

If strace is not installed, then run:
```
sudo apt-get install strace
```
Also, please paste the output of this command:
```
cat /boot/uEnv.txt
```

Finally, if you are able to, then I would recommend trying this image:
**[Debian 9.3 "Stretch" iot (2018-01-21)](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Stretch_Snapshot_iot)**

pdp7, 2018-02-02:
@Grincas are you still having this problem?

pdp7, 2018-02-08:
@Grincas is it OK to close this issue?

closed 2018-02-13 by pdp7

gert-lauritsen, 2018-03-22:
Hi Im having the same problem
Short program
import Adafruit_BBIO.GPIO as GPIO
GPIO.setup("P9_23", GPIO.IN)

giving this
 sudo strace -e open python ./testGPIO.py 2>&1 | grep /sys/
open("/sys/class/gpio/gpio49/direction", O_WRONLY) = 4
open("/sys/devices/platform/ocp/ocp:P9_23_pinmux/state", O_WRONLY|O_CREAT|O_TRUN                         C, 0666) = -1 ENOENT (No such file or directory)

the system is:
Linux beaglebone 4.14.25-ti-r38 #1 SMP Sat Mar 10 07:43:50 UTC 2018 armv7l GNU/Linux

origtech, 2018-09-19:
Hi I'm having the same problem and have tried all the suggestions in this thread without any success.

Here is a sample of my test program:
#!/user/bin/python
import Adafruit_BBIO.GPIO as GPIO

GPIO.setup('P8_11', GPIO.IN)
if(GPIO.input('P8_11')):
    print 'high'
else:
    print 'low'

When I run the program I get this error message:
  File "ioTest.py", line 4, in <module>
    GPIO.setup('P8_11', GPIO.IN)
ValueError: Set gpio mode failed, missing file or invalid permissions.

sudo /opt/scripts/tools/version.sh produces the following:

git:/opt/scripts/:[e307a944e0be0610ff5296e0abe4ad31a6e70daa]
eeprom:[A335BNLT000C1826BBBG0624]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
kernel:[4.9.82-ti-r102]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180914.0-0rcnee0~stretch+20180914]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BB-BONE-4D7R-01, coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.344481] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.345071] gpio-of-helper ocp:gpio_helper: ready
[    1.345687] gpio-of-helper ocp:cape-universal: ready
END

cat /boot/uEnv.txt produces the following:
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.9.82-ti-r102
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
dtb_overlay=/lib/firmware/BBB-UMC-TESTER-00A0.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

To try and correct the problem I have run:
sudo apt-get update
sudo apt-get upgrade

But the problem remains. I am running Debian Version 9.3 IoT Stretch.
Any suggestions on what to try?

pdp7, 2018-09-20:
@origtech please open a new issue so it is easier to keep track of the problem you experienced

@RobertCNelson is 2018.01 too old of u-boot version for u-boot device tree overlays to work?