adafruit/adafruit-beaglebone-io-python #208: No module named 'sysfs' in newest version of Adafruit_BBIO

AKAMEDIASYSTEM, 2017-11-30:
Hello, I've been following the recent improvements to this library with interest and gratitude - thanks for all the work you've done on this!

I updated to the most recent version (via git clone) today, and now I receive this error when running a script that polls two encoders:
```
Traceback (most recent call last):
  File "bbb-enc.py", line 9, in <module>
    import Adafruit_BBIO.Encoder as Encoder
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_BBIO-1.0.8-py3.5-linux-armv7l.egg/Adafruit_BBIO/Encoder.py", line 7, in <module>
    import sysfs
ImportError: No module named 'sysfs'
```

...I see that sysfs.py exists at Adafruit_BBIO/sysfs.py but I'm not sure how to point this library at that; it's probably something simple, but I have looked around online for hints all morning and not found much.

Thanks in advance for any advice you might have!

AKA

dplanella, 2017-11-30:
Oh, wow, this hasn't even been released and folks are already testing it, excellent :)

Is this perhaps Python 3 related? I used Python 2.7 to test the module, as my app uses `Adafruit_GPIO`, which is not Python3-compatible.

If you hack the `Encoder.py` file and add this snippet before the `import sysfs` line, does it all work then?

```python
import os
import sys

file_dir = os.path.dirname(__file__)
sys.path.append(file_dir)
```

AKAMEDIASYSTEM, 2017-11-30:
Oh I thought I'd give it a try since there's such quick and helpful support (thanks again!)

I added the suggested lines (and had to change one instance of `iteritems` to `items` in Encoder.py) and now things seem to be working in python3.

Thanks!

AKA

AKAMEDIASYSTEM, 2017-11-30:
Ah, one wrinkle to this solution - I now need to be root to run the script, as when I run as user debian, I see:
```
PermissionError: [Errno 13] Permission denied: '/sys/devices/platform/ocp/48300000.epwmss/48300180.eqep/enabled'
```
...which seems logical, since sysfs is doing these changes now.

I don't mind running this all as root, though I know that's not the Best Practice. Thought I should add this to the issue, though.

Thanks again!

AKA

RobertCNelson, 2017-11-30:
@AKAMEDIASYSTEM that should have been taken care of with:

https://github.com/rcn-ee/repos/commit/e30fb68cd43fb40d49207f264e62ff6222868cc4

Double check that the directory has "eqep" permissions, and then that your user is part of the eqep group.

```
cat /etc/group | grep eqep
eqep:x:997:debian
```

https://github.com/rcn-ee/repos/commit/1bf3004d6b835670b9f64e810fa95858dfa91bc6

Regards,

dplanella, 2017-11-30:
@AKAMEDIASYSTEM thanks for the feedback. Two things:

- Would you mind pulling from my [`fix-sysfs-import`](https://github.com/dplanella/adafruit-beaglebone-io-python/blob/fix-sysfs-import/Adafruit_BBIO/Encoder.py) branch? It now includes the fixes to the two issues you mentioned. If that works, I'd send a PR against upstream.
- You don't need to be root to access the eqep modules, but you need to be part of the `eqep` group, as you've already found out :) Great feedback again, the documentation needs to be updated to mention that.

AKAMEDIASYSTEM, 2017-11-30:
Thanks! I wasn't part of `eqep` group, but now I think I am:
```
debian@beagleboat:~$ cat /etc/group | grep eqep
eqep:x:992:debian
```
...however, after pulling the `fis-sysfs-import` (and adding the four lines to Encoder.py per @dplanella advice in this thread) (and changing `iteritems` to `items` in Encoder.py), I still see the permissions error. Running as root results in successful execution, though. The error is 
```
PermissionError: [Errno 13] Permission denied: '/sys/devices/platform/ocp/48300000.epwmss/48300180.eqep/enabled'
```
Happy to try whatever you suggest - my apologies that I'm not able to contribute more, I'm still a bit of a Linux/embedded beginner...

Thanks again,

AKA

AKAMEDIASYSTEM, 2017-11-30:
...confirming for @dplanella that I did indeed run 
```
sudo apt update
sudo apt upgrade bb-cape-overlays bb-customizations
```
before trying the `fix-sysfs-import` branch...

dplanella, 2017-11-30:
@AKAMEDIASYSTEM thanks. You shouldn't need to change anything after pulling the `fix-sysfs-import` branch, that was the point of creating it.

The branch already contains these two fixes (e70a7fde00ae258f33508d390993f1c46b9c8611 and 37f7204f3fd9d429822b541a9f783eb0684b559e). Are you sure you pulled it correctly (and that you switched to it)?

AKAMEDIASYSTEM, 2017-11-30:
Heh, sorry, my ineptitude strikes again ;-)

I thought that navigating to https://github.com/dplanella/adafruit-beaglebone-io-python/tree/fix-sysfs-import and then using `git clone` on the HTTPS link there would get me the correct branch, but it does not. If I instead `wget` the ZIP file and install from there, you're right, no alterations are necessary.

Thanks!

AKA

dplanella, 2017-11-30:
No worries, you were nearly there. After cloning, you would have just needed to switch to the branch. But the zip file works well too, I'll submit a PR now. Thanks for confirming it works!

pdp7, 2017-12-01:
@dplanella @AKAMEDIASYSTEM  I've merged #212

pdp7, 2017-12-01:
FYI: I have created a new [Release 1.0.9](https://github.com/adafruit/adafruit-beaglebone-io-python/releases/tag/1.0.9), and it has been [published on PyPI](https://pypi.python.org/pypi/Adafruit_BBIO)

pdp7, 2018-01-26:
@AKAMEDIASYSTEM please re-open if this is still an issue

closed 2018-01-26 by pdp7