adafruit/adafruit-beaglebone-io-python #209: eQEP  and Encoder.py support for an index channel?

AKAMEDIASYSTEM, 2017-11-30:
Hello, this is almost certainly more of an enhancement than a bug - please close/ignore/deprecate as appropriate.

My application is using two hi-resolution encoders and is working well thanks to this library. We are finding that we could use the index channel, however, to correct for small drifts in the counter (most likely the result of mechanical mounting inaccuracies).

I have read a good deal on the eQEP modules and their capabilities, and I believe that the Beaglebone Black eQEP library can natively handle an index channel on each encoder. This would be preferable to my current solution, attaching each encoder's index channel to a separate GPIO and configuring an interrupt via `add_event_detect`. However, I can't readily find what pins each encoder's index channel shall be connected to, or how to tell the eQEP that an index channel is involved.

My solution still loses steps, unfortunately (I can tell this because the encoder count should never be odd when the index channel fires), so I'm hoping there's a way to allow the eQEP handle the index channel - could anyone point me to hints/techniques for enabling this functionality?

Thanks again for the awesome library!

AKA

dplanella, 2017-11-30:
Regarding the pins, while I'm sure there is a way to figure it out from the `cape-universal` overlay, I've found that the separate [eQEP overlay files here](https://github.com/Teknoman117/beaglebot/tree/master/encoders/dts) give you a good overview of all pins/IO associated to a given eQEP module. For instance `P8_39` is the index pin for eQEP2:

https://github.com/Teknoman117/beaglebot/blob/master/encoders/dts/bone_eqep2.dts#L25

Other than that, I have no experience in handling the index channel from code.

AKAMEDIASYSTEM, 2017-11-30:
Oh thank you! I had perused that repo but hadn't seen the index pin callouts. I'll mess around with those and see if the eQEP unit just naturally handles them (the python api in that same repo doesn't seem to take any extra setup steps)

dplanella, 2017-11-30:
@AKAMEDIASYSTEM you might also need to use the `config-pin` tool to set the index channel's pin to `qep` mode, e.g.:

```
config-pin P8.16 qep
```

pdp7, 2017-12-01:
FYI: I have created a new [Release 1.0.9](https://github.com/adafruit/adafruit-beaglebone-io-python/releases/tag/1.0.9), and it has been [published on PyPI](https://pypi.python.org/pypi/Adafruit_BBIO)

pdp7, 2018-01-26:
@AKAMEDIASYSTEM please re-open if this is still an issue

closed 2018-01-26 by pdp7

czyskows, 2018-02-03:
Hi. I'm having issues getting Encoder.py to work. I think it's because I can't seem to configure the pins:
root@bela:~/Robut# config-pin P8.33 qep
P8_33 pinmux file not found!
bash: /sys/devices/platform/ocp/ocp*P8_33_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P8_33_pinmux/state

Is there a known solution to this? I can't seem to find anything. 
Thanks for any help!

pdp7, 2018-02-03:
@czyskows  Please create a new issue so we can troubleshoot your problem

Please also paste the output of these commands in that issue:
```
sudo /opt/scripts/tools/version.sh
```
```
cat /boot/uEnv.txt
```

If you are not already, I would recommend trying this image:
**[Debian 9.3 "Stretch" iot (2018-01-21)](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Stretch_Snapshot_iot)**