adafruit/adafruit-beaglebone-io-python #211: frequency and period not initialized

AKAMEDIASYSTEM, 2017-11-30:
Hello, I'm trying to use the newly-supported `frequency` getters and setters, but running into some difficulty.

My script begins by initializing eQEP0 and eQEP1. Because I wasn't sure what the default encoder polling frequency was, I then go:
```
    print(qep0.frequency)
    qep0.frequency(1000000)
    qep1.frequency(1000000)
```
...this resulted in some python3 errors (I needed to wrap int() around some inputs in the `frequency` methods), and after clearing those up I encountered an error in which `self._eqep.node.period` is not initialized.
More specifically, I get the error
```
File "/usr/local/lib/python3.5/dist-packages/Adafruit_BBIO-1.0.8-py3.5-linux-armv7l.egg/Adafruit_BBIO/Encoder.py", line 279, in frequency
    self._eqep.channel, frequency, period))
```

...thanks in advance for any advice/help you can provide!

AKA

dplanella, 2017-11-30:
@AKAMEDIASYSTEM thanks for the feedback. A few things:

- I've fixed the int() cast issue and the frequency calculation on the `fix-encoder-frequency` branch you can test on PR #213
- There is a bug in your test snippet: `frequency` is a property, not a function. Thus to set the frequency to 1 MHz as you're intending to do, you'd do an assignment `qep0.frequency = 1000000`.
- If I'm not mistaken, this sample frequency is most relevant if you set your eQEP channel to *relative* with the `setRelative()` function. The default mode is *absolute*. Quoting from [this documentation](https://github.com/Teknoman117/beaglebot/tree/master/encoders) (note: the API only exposes `frequency` as the inverse of the `period` described below):

> __mode__ - the capture mode of the eQEP hardware. 0 - Absolute mode.  The position starts at zero and is incremented or decremented by the encoder's movement. 1 - Relative mode.  The position is reset when the unit timer overflows.  This allows for tracking of the encoders movement per unit time.  Useful for velocity regulation

> __period__ - the time in nanoseconds for the unit timer to overflow.  The internal clock to the eQEP hardware runs at 100 MHz.  Since the unit timer is a 32 bit timer, this puts the maximum value of this parameter at around 40 seconds.  Note the value for this would be 40,000,000,000 nanoseconds.

> __position__ - the position of the encoder.  In absolute mode, this attribute represents the current position of the encoder.  In relative mode, this attribute represents the position of the encoder at the last unit timer overflow.  This attribute is pollable. In either operating mode, the unit timer overflow event wakes up any process polling on this attribute.  This feature is most useful in relative mode.  It allows the process or thread to awake when there is measurement available in the last time period.

pdp7, 2017-12-01:
@AKAMEDIASYSTEM please try again. I've merged #213 from @dplanella

pdp7, 2017-12-01:
there is a new release 1.0.9 that includes #213 which is now on PyPI: https://pypi.python.org/pypi/Adafruit_BBIO

pdp7, 2018-01-26:
@AKAMEDIASYSTEM please re-open if this is still an issue

closed 2018-01-26 by pdp7