adafruit/adafruit-beaglebone-io-python #215: Out of date Adafruit_I2C.py

dplanella, 2017-12-01:
There is an `Adafruit_I2C.py` module at the root of the source tree, which seem to be several years old. A comment inside directs developers to keep it in sync with https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/tree/master/Adafruit_I2C (which is now https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/I2C.py), which has much more recent changes.

As far I can tell, `Adafruit_I2C.py` here is obsolete and could be removed. Alternatively, it would need to be brought up to date and possibly moved to the `Adafruit_BBIO` folder for consistency with the other pure Python modules.

pdp7, 2017-12-01:
I have pushed commit b1557e2 to deprecate Adafruit_I2C in favor of Adafruit_GPIO.I2C

closed 2017-12-01 by pdp7

silver2row, 2018-02-26:
Hello @dplanella and @pdp7,

Seth here. I noticed that your deviation away from Adafruit_I2C.py has broken compatibility w/ the Motor Bridge Cape library, i.e. MotorBridge.py from https://github.com/Seeed-Studio/MotorBridgeCapeforBBG_BBB. 

I tried switching Adafruit_I2C in the Seeed Studio, MotorBridge.py library to Adafruit_BBIO.I2C and all while keeping my software I typed up, w/ the help of Seeed Studio, intact like in previous revisions of the Adafruit_BBIO module. 

I like the Motor Bridge Cape. Send me in the direction(s) of where you think I should start on this project.

I sure could use some support on making patches or whatever.

Seth

P.S. See here: 
[image]

...That is the beginning of the software so far. I can change it to suit your needs but...