adafruit/adafruit-beaglebone-io-python #218: Publish API documentation

dplanella, 2017-12-01:
It struck me that there was no online API documentation, which I'm assuming is simply because no one has worked on it yet.

I've put together a [`readthedocs` branch](https://github.com/dplanella/adafruit-beaglebone-io-python/tree/readthedocs) with Python docstrings for the Encoder module (the only pure Python module in the repo), and I've manually added API documentation for the PWM module. I used sphinx to build the HTML docs locally, with which the changes are really minimal (just 3 new files under the `docs` directory).

I've then imported this branch to readthedocs for a proof of concept, and here it is:

http://adafruit-beaglebone-io-python.rtfd.io

Before I submit the branch, feedback and discussion would be welcome.

pdp7, 2017-12-02:
Thanks, great idea. I was able to build the html files with those commands.  Please open a pull reuqest.

dplanella, 2017-12-02:
@pdp7 thanks for the feedback. I've now submitted PR #219 to address this.

pdp7, 2017-12-03:
> I added the documentation generation config under docs, as it seems to be the adopted convention for Sphinx. There is a doc directory in the source tree that contains some markdown files with testing instructions. Right now the source tree layout is a bit confusing with these two, but I wouldn't want to mix all these files. Could the markdown files be moved to the test directory?

I've moved the 'doc' dir to 'tests/notes'.

pdp7, 2018-02-08:
@dplanella I tried the access http://adafruit-bbio.readthedocs.io/en/latest/ and see this message:
```
Permission Denied

You don't have the proper permissions to view this page. Please contact the owner of this project to request permission.
```
Do you know how I can make it publicly accessible?

dplanella, 2018-02-08:
@pdp7 I disabled that URL in purpose. I used it for the initial testing, and when my docs branch was merged I switched to http://adafruit-beaglebone-io-python.readthedocs.io/en/latest. It's a longer URL to type, but for some reason it also got better Google search ranking results when looking for e.g. "beaglebone python api documentation". I can remove the adafruit-bbio project altogether if that causes confusion.

You should have full admin permissions on http://adafruit-beaglebone-io-python.readthedocs.io. If that does not work for some reason, let me know and we can sort it out. Also, I believe that is the only project you should see when you log into readthedocs.io.

All that has made me notice that the **GitHub webhook on readthedocs seems to have stopped working after 1.0.9 for no particular reason**. That means:

- The 1.0.10 label does not automatically appear on the active upstream versions page
- Documentation builds of the active branches/tags do not happen automatically upon upstream commits. (Currently only the upstream master branch is enabled, as the '/latest' URL.) 

We might need to take this to the readthedocs folks, as as far as I can tell, there is no visibility on the admin interface on what the webhook is doing. One can simply delete, resync or create a webhook. I've tried resyncing, but that gives me a "Fail.  Check back in a bit!" :/

Active upstream versions:
https://readthedocs.org/dashboard/adafruit-beaglebone-io-python/versions/

GitHub webhook:
https://readthedocs.org/dashboard/adafruit-beaglebone-io-python/integrations/

That's a bit of a bummer, as one of the main goals was for the API documentation to be nearly maintenance free...

pdp7, 2018-02-08:
@dplanella thanks for the detailed explanation.

dplanella, 2018-02-08:
@pdp7 no worries. I filed an RT on readthedocs - https://github.com/rtfd/readthedocs.org/issues/3587

pdp7, 2018-07-20:
@tannewt @tdicola @ladyada 

Hi, it does not seem that I can access this URL:
https://github.com/adafruit/adafruit-beaglebone-io-python/settings/hooks

**Does this work for you?**

@dplanella had [asked me check that URL](https://github.com/rtfd/readthedocs.org/issues/3587#issuecomment-404221048)

dplanella, 2018-07-20:
@pdp7 great, thanks!

Depending on how busy these folks are, another alternative that we have is to do a commit on adafruit-beaglebone-io-python. That should trigger a documentation build now that the readthedocs bug is reportedly fixed. If it does, we're all good. If not, then either the rtd bug is not fixed or it was not the cause for the automatic builds not being triggered.

pdp7, 2018-07-24:
@dplanella here is the reply from ladyada:
https://github.com/rtfd/readthedocs.org/issues/3587#issuecomment-406837681
> here's the current hooks
https://user-images.githubusercontent.com/1214161/43041857-a75e0fac-8d3a-11e8-9f51-f7dd53d1da7c.png
> what do you want added?

dplanella, 2018-07-25:
@pdp7 sorry, I had seen the reply, but was not able to look at it in detail until today.

Let me have a play with my forked repo, where I do have permissions to see and add/remove webhooks and see if I can find out more.

tannewt, 2018-07-25:
I believe you need to copy the webhook url from readthedocs integrations page into github webhooks. https://docs.readthedocs.io/en/latest/webhooks.html

pdp7, 2018-07-25:
@tannewt Is that something that you or @ladyada can do?  I don't believe I have permissions to do so.  Thanks.
https://github.com/rtfd/readthedocs.org/issues/3587#issuecomment-406837681

tannewt, 2018-07-25:
I don't think I have RTD permissions to get the webhook url. Feel free to email it to me or grant me admin permission on RTD.

ladyada, 2018-07-25:
@pdp7 i just made you an admin of this repo, can you see if its possible for you to see the webhooks now?

dplanella, 2018-07-25:
@pdp7 I think to get the RTD URL to use as Payload on the GitHub settings, you'll need to first delete the existing "GitHub incoming webhook" on RTD.

Otherwise I can't see a way to find out that URL. I think it's because RTD automatically sets up the webhook on both sides when you first set up a project. If, alternatively, you set up a webhook manually, the webhook screen looks different and it then shows the URL. I've just checked that out with my test project at RTD and my fork at GitHub. I believe the fact that webhooks are shown differently on RTD is a bug.

In summary, I think that's what needs to happen:

1. On RTD: delete the "GitHub incoming webhook" at https://readthedocs.org/dashboard/adafruit-beaglebone-io-python/integrations
2. On RTD: click on the "Add integration" button on that URL. Then do it again (the UI is somewhat confusing there).
3. On RTD: on the same screen, copy the https://readthedocs.org/api/v2/webhook/adafruit-beaglebone-io/nnnnn URL
4. On GitHub: go to https://github.com/adafruit/adafruit-beaglebone-io-python/settings/hooks and click on "Add webhook"
5. On GitHub: paste the RTD URL you copied on 3. into the "Payload URL" text field. The other fields can be left with the defaults, as they seem to match what's required on https://docs.readthedocs.io/en/latest/webhooks.html
6. On GitHub: Click on "Add webhook" 

After that, we should be all set. The webhook won't be triggered (and thus docs rebuilt) until the next push event, though. So only then, after looking at the RTD webhook logs, we'll know if it's worked.

We both have access to the RTD project settions, but I don't have access to the GitHub repo settings (I don't need them either).

So if that's ok with you, I'd suggest it's you who follows those steps above. Simply for the sake of being one single person changing both sides at the same time. Alternatively, I'm more than happy to do the RTD changes myself.

pdp7, 2018-07-26:
@ladyada thanks!  yes, I can edit the Webhooks now.

ladyada, 2018-07-26:
gr8! im outtie

dplanella, 2018-08-01:
Thanks @ladyada. @pdp7 have you had the chance to restore the webhooks? Is there anything I can do on my side to help?