adafruit/adafruit-beaglebone-io-python #223: Use libiio to interface with ADC

pdp7, 2018-01-12:
I've been considering if Adafruit_BBIO should be using [`libiio`](https://wiki.analog.com/resources/tools-software/linux-software/libiio) to interact with the ADC, instead of accessing it via `/sys`.

@RobertCNelson commented that this could allow Adafruit_BBIO to access more ADC functioality than a single raw read:  [ADC User Guide: Continuous Mode](http://processors.wiki.ti.com/index.php/Linux_Core_ADC_Users_Guide#Continuous_Mode)

pdp7, 2018-01-12:
There is already [*libiio python binding*](https://github.com/analogdevicesinc/libiio/blob/master/bindings/python/iio.py) 

thanks to @RobertCNelson for the link!

pdp7, 2018-01-12:
This kernel patch added DMA support for the BeagleBone's ADC peripheral:
[**drivers: iio: ti_am335x_adc: add dma support**](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/iio/adc/ti_am335x_adc.c?h=v4.15-rc7&id=f438b9da75eb80eb6c4095a5b75324cc9a7f0570)
> This patch adds the required pieces to ti_am335x_adc driver for DMA support

MarkAYoder, 2020-07-03:
The link above needs updating:  [ADC User Guide: Continuous Mode](http://software-dl.ti.com/processor-sdk-linux/esd/docs/latest/linux/Foundational_Components/Kernel/Kernel_Drivers/ADC.html#Continuous%20Mode)

MarkAYoder, 2020-07-03:
@pdp7 @RobertCNelson I'm trying to understand this project.  The **Continuous Mode** example in the updated link helped a lot.  Do you see adding an interface to Adafruit_BBIO that allows Continuous Mode?  What would that interface look like?

How fast does the adc sample when running continuous?

I don't see how the **libiio python binding** fits in.

MarkAYoder, 2020-07-08:
@pdp7 I've hacked together a python example that uses continuous analog input. https://github.com/beagleboard/cloud9-examples/blob/v2020.08/BeagleBone/Black/analogInContinuous.py It uses gnuplot to plot the signal.  

How do you see BBIO to be changed to implement things such as enabling inputs, changing buffer size, etc.?

pdp7, 2020-07-09:
@MarkAYoder I'm thinking that https://github.com/analogdevicesinc/libiio/tree/master/bindings/python might be useful.  Such as https://github.com/analogdevicesinc/libiio/blob/master/bindings/python/examples/iio_readdev.py