adafruit/adafruit-beaglebone-io-python #224: Problems with specific GPIO pins

boineg, 2018-01-13:
I'm new to the Beaglebone Black platform and linux boards in general so forgive my noob mistakes and inability to debug properly. I whoever will help me will give me specific instructions as to what to do and what to specifically type in the terminal.

I'm running newly flashed Debian 9.1 2017-08-31 4GB SD LXQT and am having problems configuring specific pins, namely P8_7, P8_8, P9_27. I have tested the other pins and they seem to be working fine. 

I am using a very simple code.
`import Adafruit_BBIO.GPIO as GPIO
GPIO.setup("P8_7", GPIO.OUT)
GPIO.output("P8_7",GPIO.HIGH)
GPIO.cleanup()
`

and the terminal gives me this error

> Traceback (most recent call last):
  File "gpio.py", line 5, in <module>
    GPIO.setup("P8_7", GPIO.OUT)
ValueError: Set gpio value failed, missing file or invalid permissions.

I have also tried configuring them as inputs using this code.
`import Adafruit_BBIO.GPIO as GPIO
GPIO.setup("P8_8", GPIO.IN)
GPIO.add_event_detect("P8_8", GPIO.RISING)
GPIO.cleanup()
`

and I get this error

> Traceback (most recent call last):
  File "gpio.py", line 3, in <module>
    GPIO.add_event_detect("P8_8", GPIO.RISING)
RuntimeError: Failed to add edge detection

pdp7, 2018-01-20:
Hi @boineg, sorry for the delay. Have you tried running Python with sudo?

I've been traveling but will be home on Sunday and will try to reproduce this issue.

pdp7, 2018-01-20:
It would be helpful if you could run strace so that I can better understand what is happening.

Install strace (if it doesn't already exist):

```sudo apt-get install strace```

Then run strace, replace test.py with the name of your test program:

```strace -f -o strace.log python test.py```

Paste the contents of strace.log into a gist.

boineg, 2018-01-25:
Sorry for the delay as well. I solved the issue by flashing a different image.

closed 2018-01-26 by pdp7