adafruit/adafruit-beaglebone-io-python #225: GPIO.setup() fails on docker container

maiorfi, 2018-01-20:
...because of check made in /source/common.c (https://github.com/adafruit/adafruit-beaglebone-io-python/blob/61af64e734f3f7c2ca96495cea7ea9fc78f4c877/source/common.c#L595) where library execs a grep for model string "/proc/device-tree/model". Error message is "/bin/grep: /proc/device-tree/model: No such file or directory".

I'm running beaglebone python (2) resin.io container. More info here: https://docs.resin.io/runtime/resin-base-images/#python

On container, "/proc/device-tree" is actually a symbolic link to "/sys/firmware/devicetree/base", but "/sys/firmware" doesn't exist at all.

pdp7, 2018-01-26:
@maiorfi thanks for reporting.

@RobertCNelson any thoughts about a more robust way to check the board model?

RobertCNelson, 2018-01-26:
@pdp7 easy give up!

If we can't read /proc/device-tree, the library should give up and spit out an error like. "Unable to detect board, please set an override in your python project: bbio_mode=BBB" Or something like that, but optional..

Regards,

pdp7, 2018-02-02:
> I'm running beaglebone python (2) resin.io container. More info here: https://docs.resin.io/runtime/resin-base-images/#python

@maiorfi I'm not too familiar with Resin.io.  Does this mean that the container is running on the BeagleBone?  Or is it that there is a container with a BeagleBone environment running on a host computer?

RobertCNelson, 2018-02-02:
It's yocto base, they've been building off our kernel, so feature wise it's in lock-step with us.

https://docs.resin.io/understanding/understanding-devices/2.x/

Regards,

maiorfi, 2018-02-02:
Resin.IO provided for container images that you can run using Docker (daemon) on a physical host. In this case we are talking about using BBB as host.
So, since docker allows for "exposing" host devices (such as /dev/ttyUSB0) to containers (via --device option in "docker run" command), I am not sure one can do same thing with entire device subtrees, such as /sys/class. Maybe mapping volumes between host and container in /sys/class space?

pdp7, 2018-02-04:
@maiorfi I would be open to the idea of adding a [special case in the code](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/61af64e734f3f7c2ca96495cea7ea9fc78f4c877/source/common.c#L595) to look at an alternate path if `/proc/device-tree/model` does not exist.  Could you create a file somewhere in the filesystem with the board model?

maiorfi, 2018-02-05:
...maybe parsing "uname -a" could be useful in order to have some hint about container hosting, but the real question is "is there anything that will prevent library from working as expected when running inside a container?"

RobertCNelson, 2018-02-05:
@maiorfi no uname -a is not going to tell you anything about the underline hardware.  Your container should stop playing games, and just export proc.

Regards,

maiorfi, 2018-02-05:
I guess we could ask resin.io (container images maintainer) to provide for that on their base images, but I am wondering if even skipping that check, container running python library could actually access to underlying hardware...

RobertCNelson, 2018-02-05:
Try it:

```
cd /sys/class/pwm/pwmchip*/
echo 1 > export
cd pwm*/
echo 200000 > period
echo 100000 > duty_cycle
echo 1 > enable
```

pdp7, 2018-02-08:
@maiorfi were you able to try the above instructions?

pdp7, 2018-02-14:
@maiorfi were you able to try the PWM test listed above?

pdp7, 2018-02-22:
Please reopen if this issue is still of interest

closed 2018-02-22 by pdp7