adafruit/adafruit-beaglebone-io-python #226: PWM - Problem with a syses file error

doggone1, 2018-01-21:
I saw where this has been an issue in the past but looked like everyone had resolved.  I had kernel 4.9; however, I saw that there was an issue there( config-pin -q p8.13 was showing it as pwm for me if that matters), so i followed the instructions on previous posting and moved to kernel 4.4.  I did reboot and have the same python RuntimeError. 

I have tried running as debian and with sudo and have also just tried python console.  Current kernel version I'm running is 4.4.110.

The fail occurs at PWM.start("P8_13", x, x)

Any help is much appreciated

pdp7, 2018-01-26:
Please paste the output of these commands:
```
sudo /opt/scripts/tools/version.sh
```
```
cat /boot/uEnv.txt
```

Also, it would be helpful if you could run strace so that I can better understand what is happening.

Install strace (if it doesn't already exist):

```sudo apt-get install strace```

Then run strace, replace test.py with the name of your test program:

```strace -f -o strace.log python test.py```

Paste the contents of strace.log into a gist.

Finally, I would recommend trying the latest Debian image:
**[Debian 9.3 "Stretch" iot (2018-01-21)](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Stretch_Snapshot_iot)**

doggone1, 2018-01-26:
version.sh
git:/opt/scripts/:[714e162ba98cf3d2897e7fc95e951c6df15a7d0a]
eeprom:[A335BNLT000C3114BBBK2363]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-08-31]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-rc2-00002-g7c9353]
kernel:[4.4.110-ti-r143]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180104.0-0rcnee1~stretch+20180104]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
dmesg | grep pinctrl-single
[    1.186023] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END

I tried to install strace; however, i am getting I don't have enough free space in /var/cache/apt/archives.  I tried apt-get autoclean and apt-get clean and it seems to have made zero difference in space

doggone1, 2018-01-27:
sorry, forgot to include the uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.4.110-ti-r143
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

I am loading up the new Debian 9.3 that you provided the link for right now so we shall see how that goes.  I had been running off of the flashed eMMC but will run this off of 32GB sd card.

doggone1, 2018-01-27:
Attempted with new Debian image and I still get the same Problem with sysfs file when trying PWM.start("P8_13",95.0,60)

doggone1, 2018-01-27:
think i figured out how to gist the strace.log file.. https://gist.github.com/doggone1/c15a17412c7fad141074b2a0e7998163

RobertCNelson, 2018-01-27:
```open("/sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip6/pwm1/period", O_RDWR) = -1 EACCES (Permission denied)
1779  g```

Let's double check:

```
debian@test-bbb-2:~$ cat /etc/group | grep pwm
pwm:x:994:debian
debian@test-bbb-2:~$ cat /etc/group | grep gpio
gpio:x:995:debian
```
If those aren't set, just open it up in nano as root and add yourself and reboot..

Regards,

RobertCNelson, 2018-01-27:
@pdp7 i just added a "groups" check to version.sh:

```
debian@test-bbb-2:/opt/scripts/tools$ sudo ./version.sh 
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLT0A5B2313BBBK3287]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Debian Image 2015-08-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
kernel:[4.9.78-ti-r93]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180119.0-0rcnee0~stretch+20180119]
WARNING:pkg:[bb-wl18xx-firmware]:[NOT_INSTALLED]
WARNING:pkg:[firmware-ti-connectivity]:[NOT_INSTALLED]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev i2c admin spi weston-launch xenomai gpio pwm cloud9ide]
dmesg | grep pinctrl-single
[    1.440871] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END
```

doggone1, 2018-01-27:
grep pwm comes back with pwm:x:998:debian and gpio with gpio:x:999:debian  so i think i'm good there, right?

Sorry, I'm not understanding the "groups" check in version.sh.  Is this something you think i should add?  If so, how do I do that?  

I do appreciate the help on this.

RobertCNelson, 2018-01-27:
@doggone1 yeah your good, the only problem.  For the "pwm" you need a v4.11.x+ kernel for a non-root access. (your running v4.9.x)..  So with v4.9.x you'll nee to run it as root..

Regards,

doggone1, 2018-01-27:
Ok, thank you Robert, i will try to run as root and see if that works until Beaglebone image has a 4.11 version

doggone1, 2018-01-28:
Strange. Earlier sudo didn't solve but now it does. Thank you sir

pdp7, 2018-01-30:
Thanks. Please reopen if there are still any problems.

closed 2018-01-30 by pdp7