adafruit/adafruit-beaglebone-io-python #228: GPIO value set up failure

rd-tyler, 2018-01-30:
I am currently running into an issue using the BBB version of the GPIO set up using this library. I am able to set up PWM and GPIO manually but using anything included in the Char_LCD library, including the example sets ( https://github.com/adafruit/Adafruit_Python_CharLCD ) I receive the following error:
```
debian@beaglebone:~/Adafruit_Python_CharLCD/examples$ sudo ./char_lcd_rgb_pwm.py
Traceback (most recent call last):
  File "./char_lcd_rgb_pwm.py", line 71, in <module>
    enable_pwm=True)
  File "build/bdist.linux-armv7l/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 367, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 152, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/GPIO.py", line 278, in setup
ValueError: Set gpio value failed, missing file or invalid permissions.
```

After some digging I was able to find similar issues having been resolved by upgrading the kernel version. In attempting to do so I get the following:
```
debian@beaglebone:/opt/scripts/tools$ sudo ./update_kernel.sh --lts-4_14
info: checking archive
2018-01-23 23:47:42 URL:https://rcn-ee.com/repos/latest/stretch-armhf/LATEST-ti [136/136] -> "LATEST-ti" [1]
-----------------------------
Kernel Options:
ABI:1 LTS41 4.1.30-ti-r70
ABI:1 LTS44 4.4.110-ti-r143
ABI:1 LTS49 4.9.77-ti-r92
ABI:1 LTS414 4.14.14-ti-r27
-----------------------------
Kernel version options:
-----------------------------
LTS44: --lts-4_4
LTS49: --lts-4_9
LTS414: --lts-4_14
STABLE: --stable
TESTING: --testing
-----------------------------
info: you are running: [4.4.91-ti-r133], latest is: [4.14.14-ti-r27] updating...
Ign:1 http://deb.debian.org/debian stretch InRelease                          
Hit:2 http://repos.rcn-ee.com/debian stretch InRelease                        
Hit:3 http://deb.debian.org/debian stretch-updates InRelease
Hit:4 http://deb.debian.org/debian-security stretch/updates InRelease
Hit:5 http://deb.debian.org/debian stretch Release
Reading package lists... Done
Building dependency tree       
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Checking dpkg...
Checking apt-cache...
debug: installing: [linux-image-4.14.14-ti-r27]
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
```
Trying to execute the recommended command just temporarily disconnects the beaglebone, resets it to the standard landing screen, and doesn't actually resolve the issue as it persists and repeats the message.

When executing:
```
sudo apt list --upgradable
```
I get the following:
```
debian@beaglebone:~$ sudo apt list --upgradable
[sudo] password for debian:
Listing... Done
bb-node-red-installer/unknown 0.17.5-0rcnee5~stretch+20180120 armhf [upgradable from: 0.17.5-0rcnee0~stretch+20170816]
N: There is 1 additional version. Please use the '-a' switch to see it
```

So, I tried to upgrade node red using the following approach:
```
sudo apt remove bb-node-red-installer --purge
sudo apt update
sudo apt install bb-node-red-installer
```
however I was unable to remove the old version and was stopped on the first step:
```
debian@beaglebone:~$ sudo apt remove bb-node-red-installer --purge
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem.
```

For additional reference my version.sh returns:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[b5d4d4251cc19ff84bc5ad0076cc4493caa3a7f1]
eeprom:[A335BNLT00C03917BBBK1D4C]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.4.91-ti-r133]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180119.0-0rcnee0~stretch+20180119]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
dmesg | grep pinctrl-single
[    2.131993] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END
```

and my uEnv.txt:
```
debian@beaglebone:~$ cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.4.91-ti-r133
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```

My next step is to attempt the newest release of debian as a clean flash but I have yet to attempt it. Any insights would be greatly appreciated.

pdp7, 2018-02-02:
> My next step is to attempt the newest release of debian as a clean flash but I have yet to attempt it.
Have you been able to try this yet?

Tomorrow I will try to run Adafruit_Python_CharLCD/char_lcd_rgb_pwm.py and see if I can reproduce the issue.

pdp7, 2018-02-04:
## Running example code
I do not have a LCD wired up but I am able to run `char_lcd_rgb_pwm.py` without an error message:
```
debian@beaglebone:~/tmp/Adafruit_Python_CharLCD/examples$ sudo python ./char_lcd_rgb_pwm.py 
Press Ctrl-C to quit.
```

Here is the diff of `char_lcd_rgb_pwm.py` to enable the BeagleBone settings _(the default is Raspberry Pi)_:
```diff
diff --git a/examples/char_lcd_rgb_pwm.py b/examples/char_lcd_rgb_pwm.py
index 8f6fc2b..21371dc 100755
--- a/examples/char_lcd_rgb_pwm.py
+++ b/examples/char_lcd_rgb_pwm.py
@@ -36,26 +36,26 @@ def hsv_to_rgb(hsv):
         return (v, p, q)
 
 # Raspberry Pi configuration:
-lcd_rs = 27  # Change this to pin 21 on older revision Raspberry Pi's
-lcd_en = 22
-lcd_d4 = 25
-lcd_d5 = 24
-lcd_d6 = 23
-lcd_d7 = 18
-lcd_red   = 4
-lcd_green = 17
-lcd_blue  = 7  # Pin 7 is CE1
+#lcd_rs = 27  # Change this to pin 21 on older revision Raspberry Pi's
+#lcd_en = 22
+#lcd_d4 = 25
+#lcd_d5 = 24
+#lcd_d6 = 23
+#lcd_d7 = 18
+#lcd_red   = 4
+#lcd_green = 17
+#lcd_blue  = 7  # Pin 7 is CE1
 
 # BeagleBone Black configuration:
-# lcd_rs = 'P8_8'
-# lcd_en = 'P8_10'
-# lcd_d4 = 'P8_18'
-# lcd_d5 = 'P8_16'
-# lcd_d6 = 'P8_14'
-# lcd_d7 = 'P8_12'
-# lcd_red   = 'P9_16'
-# lcd_green = 'P9_14'
-# lcd_blue  = 'P8_13'
+lcd_rs = 'P8_8'
+lcd_en = 'P8_10'
+lcd_d4 = 'P8_18'
+lcd_d5 = 'P8_16'
+lcd_d6 = 'P8_14'
+lcd_d7 = 'P8_12'
+lcd_red   = 'P9_16'
+lcd_green = 'P9_14'
+lcd_blue  = 'P8_13'
 
 # Define LCD column and row size for 16x2 LCD.
 lcd_columns = 16
```

## Pin Modes
Here is the pin modes for the pins listed in the example code:
```
debian@beaglebone:~$ config-pin -q P8_8
P8_08 Mode: default Direction: out Value: 1
debian@beaglebone:~$ config-pin -q P8_10
P8_10 Mode: default Direction: out Value: 0
debian@beaglebone:~$ config-pin -q P8_18
P8_18 Mode: default Direction: out Value: 0
debian@beaglebone:~$ config-pin -q P8_16
P8_16 Mode: default Direction: out Value: 1
debian@beaglebone:~$ config-pin -q P8_14
P8_14 Mode: default Direction: out Value: 0
debian@beaglebone:~$ config-pin -q P8_12
P8_12 Mode: default Direction: out Value: 0
debian@beaglebone:~$ config-pin -q P9_16
P9_16 Mode: pwm
debian@beaglebone:~$ config-pin -q P9_14
P9_14 Mode: pwm
debian@beaglebone:~$ config-pin -q P8_13
P8_13 Mode: pwm
```

## System info
```
debian@beaglebone:~$ cat /boot/uEnv.txt |grep -v ^#

uname_r=4.14.2-ti-r13

enable_uboot_overlays=1
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
enable_uboot_cape_universal=1

cmdline=coherent_pool=1M net.ifnames=0 quiet
```
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[e5c23dc2a45f5c9ace291dd32e7c3d79edc1932d]
eeprom:[A335BNLT000C3014BBBK1316]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.14.2-ti-r13]
nodejs:[v6.12.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171126.0-0rcnee1~stretch+20171126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
```

pdp7, 2018-02-04:
@rd-tyler please try the **Debian 9.3 2018-01-28 4GB SD IoT image** on [BeagleBoard.org Latest Firmware Images page](http://beagleboard.org/latest-images)

rd-tyler, 2018-02-05:
@pdp7 Sorry for the delay - the last few days have been utter madness. 

Upon installing the above image you noted  I was able to run the basic demo without errors by executing:
```
./ char_lcd.py
```
This execution doesn't yield any warnings or errors but it does not seem to correctly interface with the LCD I have wired up (could easily be a hasty wiring issue). Things get a little stranger whenever i try to run either ```./char_lcd_rgb.py``` or ```./char_lcd_rgb_pwm.py```
I get the following:
```
debian@beaglebone:~/Adafruit_Python_CharLCD/examples$ ./char_lcd_rgb.py
Traceback (most recent call last):
  File "./char_lcd_rgb.py", line 40, in <module>
    lcd_columns, lcd_rows, lcd_red, lcd_green, lcd_blue)
  File "build/bdist.linux-armv7l/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 367, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 152, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/GPIO.py", line 278, in setup
TypeError: argument 1 must be string, not int
```
and
```
debian@beaglebone:~/Adafruit_Python_CharLCD/examples$ sudo python ./char_lcd_rgb_pwm.py
Traceback (most recent call last):
  File "./char_lcd_rgb_pwm.py", line 71, in <module>
    enable_pwm=True)
  File "build/bdist.linux-armv7l/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 367, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 152, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/GPIO.py", line 278, in setup
TypeError: argument 1 must be string, not int
```
This error seems related as they are the same line numbers.

If it is of use, I am running this on a fresh install of the image running on the microSD where the only changes made are the IP address being used by the BBb (changed in ```/etc/network/interfaces``` ) and installation of the char_lcd package. I attempted the installation and execution after changing the IP address and rebooting the BBb. Everything seemed to be working fine until this point. 

Am I missing an overlay or another critical component?

pdp7, 2018-02-06:
@rd-tyler could you try running those example programs again with `sudo`?

could you please post a photo of the wiring?

rd-tyler, 2018-02-06:
@pdp7 Looks like that may have done the trick, at least for the software side. 
```
debian@beaglebone:~/Adafruit_Python_CharLCD/examples$ sudo ./char_lcd_rgb_pwm.py
[sudo] password for debian:
Press Ctrl-C to quit.
```
and
``` 
debian@beaglebone:~/Adafruit_Python_CharLCD/examples$ sudo python ./char_lcd_rgb_pwm.py
Press Ctrl-C to quit.
```

In terms of hardware I have two configurations based on whether or not it's running for PWM or in standard mode. 

Standard mode:
Fritzing Diagram
[image]
Broad View
[image]
Detail View 1
[image]
Detail View 2
[image]

PWM Configuration
Fritzing Diagram
[image]
Only the RGB hookup location changed
[image]

Testing each respective example with its wiring configuration yielded only a non-colored LCD with the following output, regardless of contrast resistance on the potentiometer (until it was low enough to return the screen to showing nothing). 
[image]

I do have a second LCD I can try tomorrow once I get the headers added.  According to the documentation this is a Positive LCD with the following pinout. (https://cdn.sparkfun.com/datasheets/LCD/Monochrome/ADM1602K-FS-RGB-FBW.pdf).
The second LCD appears to be Negative but I will just adjust the wiring accordingly. 

 Hopefully this is of help, if there's anything else I can provide to help solve this I would be more than happy to do so.

pdp7, 2018-02-07:
@rd-tyler thanks for posting the diagrams and photos.  I have similar LCDs so I will try wiring up to a BeagleBone Black to what results I get.

pdp7, 2018-02-07:
@rd-tyler The only char LCD I have right now is a [20x4 model from SparkFun](https://www.sparkfun.com/products/256).  Unfortunately, it is not RGB.

For a simple test, I wired up according to the Adafruit Learning System tutorial.  I did not connect the backlight.  I was able to run these examples OK and see the output on the LCD: `char_lcd.py`, `char_lcd_rgb.py` and `char_lcd_rgb_pwm.py`

Here are the photos of my wiring:
https://photos.app.goo.gl/LzqQrPsl15qctK0E3

Could you try wiring up the LCD without the blacklight pins connected? This might make the wiring simpler to verify for a basic test.

rd-tyler, 2018-02-07:
@pdp7 No worries. Thank you so much for your help. Must have been a faulty LCD. I swapped it for my White on Black LCD (new part) and it came right up when using ```sudo ./char_lcd.py```

[image]

I am confident that had the LCD been working properly we would have been all set. Sorry for all of the hassle. I really appreciate all of your assitance on this one.

pdp7, 2018-02-08:
@rd-tyler good to see it is working.

OK to close this now?

rd-tyler, 2018-02-08:
@pdp7 I believe so. Thanks again!

closed 2018-02-08 by rd-tyler

estudoaluno, 2019-10-28:
Eu gostaria de uma ajuda ,Quando tento faz  instalacao da erro.
https://github.com/adafruit/Adafruit_Python_CharLCD.git