adafruit/adafruit-beaglebone-io-python #229: PWM enablement: P8_07, P8_08, P8_09 and P8_10

RobertCNelson, 2018-02-02:
Yes, @pdp7 you've read correctly, these Pin's have PWM functionality!!!

This stems from that timer/mmap thread we had last month.  While looking at their code, they were mmap'ing into the timer to create a pwm output.  Turns out this module/block got merged into the kernel a few years ago and we missed it.

Kernel config:
```
CONFIG_PWM_OMAP_DMTIMER
```

U-Boot Overlays:
https://github.com/beagleboard/bb.org-overlays/commit/563af291b72c97a5f93d920ff599e99e74b0c88d

Now for the fun: /sys differences.. (not under ocp like all the pwm's..) Looking at common.c i see a big 'rewrite' to support this. ;)
```
/sys/devices/platform/dmtimer-pwm-*/
```

The platform name comes from the node, so the user can easily customization it.. For now let's assume everyone uses the "dmtimer-pwm-<timer #>" name...

```
fragment@2 {
	target-path="/";
	__overlay__ {
		dmtimer-pwm-4 {
			pinctrl-names = "default";
			pinctrl-0 = <&timer4_pin>;

			compatible = "ti,omap-dmtimer-pwm";
			#pwm-cells = <3>;
			ti,timers = <&timer4>;
			//ti,prescaler = <0>;		/* 0 thru 7 */
			ti,clock-source = <0x00>;	/* timer_sys_ck */
			//ti,clock-source = <0x01>;	/* timer_32k_ck */
		};
	};
};
```

Currently: these kernels support it:
```
4.15.0-bone3
4.14.16-ti-r30/4.14.16-ti-rt-r30
```
These are building and have been tested:
```
4.9.79-ti-r96/4.9.79-ti-rt-r96
4.4.113-ti-r146/4.4.113-ti-rt-r146
```
With this overlay setup: 
```
uboot_overlay_addr0=/lib/firmware/BB-PWM-TIMER-P8.07.dtbo
uboot_overlay_addr1=/lib/firmware/BB-PWM-TIMER-P8.08.dtbo
uboot_overlay_addr2=/lib/firmware/BB-PWM-TIMER-P8.09.dtbo
uboot_overlay_addr3=/lib/firmware/BB-PWM-TIMER-P8.10.dtbo
```

I haven't pushed an overlay update with these yet so you will need to manually:
```
cd /opt/source/bb.org-overlays/
git pull
make
sudo make install
```

Regards,

pdp7, 2018-02-04:
@RobertCNelson very interesting, thanks for letting me know

Savanth123, 2019-08-14:
I am unable to run the servos from PWM pins in beagle bone black what is the error.Please anyone help

pdp7, 2019-12-17:
@Savanth123 please open a new issue if still having problem.

pdp7, 2019-12-17:
@MarkAYoder / @jadonk - another issue that might be a good for a student.  This would enable use of additional PWM pins.

pdp7, 2020-08-01:
@daschachter thanks!

closed 2020-08-01 by pdp7