adafruit/adafruit-beaglebone-io-python #230: PWM Runtime Error

gr4mp, 2018-02-04:
Hello ,
I know this issue is reopened like a million times yet I couldn't solve it
After a fresh flash of ubuntu(version of kernel and os specified below) I install bb-cape-overlays and then Adafruit_BBIO . And of course i get runtime error so I modified /boot/uEnv.txt ,

```
ubuntu@arm:/home$ cat /boot/uEnv.txt 
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.16-ti-r30
#uuid=
#dtb=

##BeagleBone Black/Green dtb's for v4.1.x (BeagleBone White just works..)

##BeagleBone Black: HDMI (Audio/Video) disabled:
#dtb=am335x-boneblack-emmc-overlay.dtb

##BeagleBone Black: eMMC disabled:
#dtb=am335x-boneblack-hdmi-overlay.dtb

##BeagleBone Black: HDMI Audio/eMMC disabled:
#dtb=am335x-boneblack-nhdmi-overlay.dtb

##BeagleBone Black: HDMI (Audio/Video)/eMMC disabled:
#dtb=am335x-boneblack-overlay.dtb

##BeagleBone Black: wl1835
#dtb=am335x-boneblack-wl1835mod.dtb

##BeagleBone Green: eMMC disabled
#dtb=am335x-bonegreen-overlay.dtb

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/cape-universaln-00A0.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M quiet cape_universal=enable
#cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=cape-universaln-00A0

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

```
And then loaded the cape-universaln

`echo cape-universaln > /sys/devices/platform/bone_capemgr/slots`
But ,
```
ubuntu@armcat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1 
 1: PF----  -1 
 2: PF----  -1 
 3: PF----  -1 
 4: P-O---  -1 Override Board Name,00A0,Override Manuf,cape-universaln

```
It doesn't seem to be correctly loaded and also I still get the runtime errors from basic test script.
```
ubuntu@arm:/home$ cat test.py 
import Adafruit_BBIO.PWM as PWM 
#PWM.start(channel, duty, freq=2000, polarity=0) 
#duty values are valid 0 (off) to 100 (on) 
PWM.start("P9_14", 50)
PWM.set_duty_cycle("P9_14", 25.5) 
PWM.set_frequency("P9_14", 10)

PWM.stop("P9_14")
PWM.cleanup()

#set polarity to 1 on start:
PWM.start("P9_14", 50, 2000, 1)
```

```
ubuntu@arm:/home$ python test.py 
Traceback (most recent call last):
  File "test.py", line 4, in <module>
    PWM.start("P9_14", 50)
RuntimeError: Unknown error
```
Lastly, the reason why I am using ubuntu images is I need to use ROS for my project and I really did not want to cross compile ROS for debian image.

- Ubuntu 16.04.3 LTS (GNU/Linux 4.14.16-ti-r30 armv7l)
-Beaglebone Black

pdp7, 2018-02-04:
The Cape Manager has been deprecated by [U-Boot Overlays](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays).  Since `/sys/devices/platform/bone_capemgr/slots` exists on your system, I believe it is not using U-Boot Overlays. 

Please paste the output of these commands in that issue:
```
sudo /opt/scripts/tools/version.sh
```

I normally recommend people use the official BeagleBoard.org Debian image.  This is the latest snapshot image: **[Debian 9.3 "Stretch" iot](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Stretch_Snapshot_iot)**

@RobertCNelson can ROS now be used with Debian Stretch?

RobertCNelson, 2018-02-04:
I talked with the ROS guys last week.. They are only doing "stretch" builds for arm64, and amd64.

Luckily their build infrastructure is on github, so there is another option for stretch, it'll probably take a few x15's, and some time...

gr4mp, 2018-02-04:
Output of

`sudo /opt/scripts/tools/version.sh`

is

```
root@arm:/home/ubuntu# /opt/scripts/tools/version.sh 
git:/opt/scripts/:[ad016da40de5805f1a6f981cbb6c454b1a7f244b]
eeprom:[A335BNLT00C05315BBBK1009]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Ubuntu Image 2018-01-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]
kernel:[4.14.16-ti-r30]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~xenial+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~xenial+20180104]
WARNING:pkg:[firmware-ti-connectivity]:[NOT_INSTALLED]
groups: ‘debian’: no such user
groups:[]
dmesg | grep pinctrl-single
[    0.931178] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.932747] gpio-of-helper ocp:cape-universal: ready
END
```

Also another issue is, I have to load the cape-universal manually on every boot. How can it be loaded at the boot ?

RobertCNelson, 2018-02-04:
```
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]
```
there's your bug..

This might work:
```
sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k
```
But i'm guessing MLO is in a fat partition in /dev/mmcblk1p1

so just mount it and delete that file..

Regards,

gr4mp, 2018-02-04:
I believe I 'formatted' eMMC with your suggestion

`sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k`

So MLO is not there,

```
root@arm:~# mount /dev/mmcblk1p1 /media/emmc
root@arm:~# cd /media/emmc/
root@arm:/media/emmc# ls -a
.  ..  System Volume Information
root@arm:/media/emmc# 
```
But it look like the same, except I commented out 

```
enable_uboot_overlays=1
enable_uboot_cape_universal=1
```
@pdp7  I suppose commenting those out was what you meant ?

```
root@arm:/opt/scripts/tools# ./version.sh 
git:/opt/scripts/:[ad016da40de5805f1a6f981cbb6c454b1a7f244b]
eeprom:[A335BNLT00C05315BBBK1009]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Ubuntu Image 2018-01-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]
kernel:[4.14.16-ti-r30]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~xenial+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~xenial+20180104]
WARNING:pkg:[firmware-ti-connectivity]:[NOT_INSTALLED]
groups: ‘debian’: no such user
groups:[]
dmesg | grep pinctrl-single
[    0.919568] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.920870] gpio-of-helper ocp:cape-universal: ready
END

```

pdp7, 2018-02-04:
I see the 2015 U-Boot is still on eMMC:
```
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]
```
@RobertCNelson does that mean that @gr4mp needs to completely erase the eMMC?

RobertCNelson, 2018-02-04:
Yeah, just erase the eMMC, it's getting in the way..

gr4mp, 2018-02-04:
I deleted all partitions on eMMC but I think BBB recreates partitions and installs U-boot(?) on boot on eMMC. But there is nothing in eMMC.
I've been wondering around and came to this,

> BBB uses MLO and u-boot from eMMC and then loads uEnv.txt file from the SDCard and then boots the kernel from the SDCard. If you press the boot button, then MLO and u-boot on the SDCard are used. 

Weird thing is when I mount FAT partition of the eMMC 

```
root@arm:~# mount /dev/mmcblk1p1 /media/emmc
root@arm:~# cd /media/emmc/
root@arm:/media/emmc# ls -a
.  ..  System Volume Information
root@arm:/media/emmc# 
```

pdp7, 2018-02-06:
@gr4mp what does `sudo /opt/scripts/tools/version.sh` show now?

gr4mp, 2018-02-06:
```
Command (m for help): d
Partition number (1,2, default 2): 1

Partition 1 has been deleted.

Command (m for help): d
Selected partition 2
Partition 2 has been deleted.

Command (m for help): p
Disk /dev/mmcblk1: 3,6 GiB, 3825205248 bytes, 7471104 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xe8820a06
```
`reboot`

```
Command (m for help): p
Disk /dev/mmcblk1: 3,6 GiB, 3825205248 bytes, 7471104 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xe8820a06

Device         Boot Start     End Sectors  Size Id Type
/dev/mmcblk1p1       2048   34815   32768   16M  e W95 FAT16 (LBA)
/dev/mmcblk1p2      34816 7471103 7436288  3,6G 83 Linux
```
```
root@arm:~# mkdir /media/emmc
root@arm:~# mount /dev/mmcblk1p2 /media/emmc
root@arm:~# cd /media/emmc/
root@arm:/media/emmc# ls
lost+found
root@arm:/media/emmc# 
```

```
root@arm:~# /opt/scripts/tools/version.sh 
git:/opt/scripts/:[320aacb1756a3cc87d7465849c9ab9b63421726b]
eeprom:[A335BNLT00C05315BBBK1009]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Ubuntu Image 2018-01-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]
kernel:[4.14.16-ti-r30]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~xenial+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~xenial+20180104]
WARNING:pkg:[firmware-ti-connectivity]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev i2c gpio pwm eqep admin spi tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.919418] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.920737] gpio-of-helper ocp:cape-universal: ready
END
```

pdp7, 2018-02-06:
@gr4mp thanks.

@RobertCNelson is this expected?
```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]
```

RobertCNelson, 2018-02-06:
@gr4mp please re-run it, i added more details to the bootloader..

```
cd /opt/scripts/tools/
git pull
sudo ./version.sh
```
Regards,

gr4mp, 2018-02-06:
@RobertCNelson  After updating 'version.sh'

```
root@arm:/opt/scripts/tools# ./version.sh 
git:/opt/scripts/:[aa257709957bdaab058ef4428469b87bb2c2f19e]
eeprom:[A335BNLT00C05315BBBK1009]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Ubuntu Image 2018-01-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]:[location: dd MBR]
kernel:[4.14.16-ti-r30]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~xenial+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~xenial+20180104]
WARNING:pkg:[firmware-ti-connectivity]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev i2c gpio pwm eqep admin spi tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.920923] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.921967] gpio-of-helper ocp:cape-universal: ready
END

```

RobertCNelson, 2018-02-06:
thanks @gr4mp it's in the dd MBR 
```
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.01-00001-gb2412df]:[location: dd MBR]
```
Let's try harder to remove it: (this will clear the first 10Mb's)
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

gr4mp, 2018-02-06:
```
root@arm:/home/ubuntu# dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
10+0 records in
10+0 records out
10485760 bytes (10 MB, 10 MiB) copied, 1,24628 s, 8,4 MB/s
```

`reboot`

```
git:/opt/scripts/:[aa257709957bdaab058ef4428469b87bb2c2f19e]
eeprom:[A335BNLT00C05315BBBK1009]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Ubuntu Image 2018-01-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]:[location: dd MBR]
kernel:[4.14.16-ti-r30]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~xenial+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~xenial+20180104]
WARNING:pkg:[firmware-ti-connectivity]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev i2c gpio pwm eqep admin spi tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.920904] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.921958] gpio-of-helper ocp:cape-universal: ready
END
```
It's finally gone!

I edited /boot/uEnv.txt as below

r```
oot@arm:/home/ubuntu# cat /boot/uEnv.txt 
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.16-ti-r30
#uuid=
#dtb=

##BeagleBone Black/Green dtb's for v4.1.x (BeagleBone White just works..)

##BeagleBone Black: HDMI (Audio/Video) disabled:
#dtb=am335x-boneblack-emmc-overlay.dtb

##BeagleBone Black: eMMC disabled:
#dtb=am335x-boneblack-hdmi-overlay.dtb

##BeagleBone Black: HDMI Audio/eMMC disabled:
#dtb=am335x-boneblack-nhdmi-overlay.dtb

##BeagleBone Black: HDMI (Audio/Video)/eMMC disabled:
#dtb=am335x-boneblack-overlay.dtb

##BeagleBone Black: wl1835
#dtb=am335x-boneblack-wl1835mod.dtb

##BeagleBone Green: eMMC disabled
#dtb=am335x-bonegreen-overlay.dtb

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```
And the following python script runs without any issue

```
ubuntu@arm:/home$ cat test.py 
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_14",50);
pwm.stop("P9_14");

```
Gotta test it with an oscilloscope though. If BBB can generate those square waves without an issue i will close the issue.

gr4mp, 2018-02-07:
Okay I've tested square waves there is usually no problem but sometimes i get this error below. Have any ideas ?

```
>>> pwm.start("P9_14",50)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Problem with a sysfs file
```

pdp7, 2018-02-07:
@gr4mp is python running as root (`sudo python`)?

It would be helpful if you could run strace so that I can better understand what is happening.

Install strace (if it doesn't already exist):
```
sudo apt-get install strace
```
Then run strace, replace test.py with the name of your test program:
```
strace -f -o strace.log python test.py
```
Paste the contents of strace.log into a GitHub Gist and post the link in a reply here.

gr4mp, 2018-02-07:
Interestingly there was not any errors this time, I both tried running as root and 'user' .

Here is the [strace.log](https://gist.github.com/gr4mp/a7fa3b346f890db9b906bd42f26b1166)
@pdp7 

Thanks.

pdp7, 2018-02-08:
> Interestingly there was not any errors this time, I both tried running as root and 'user' .

@gr4mp: does this mean it working OK for you as there is no error message?

gr4mp, 2018-02-08:
@pdp7  Yes, I did some other tests and it seems OK thanks for helping me out.

closed 2018-02-08 by gr4mp