adafruit/adafruit-beaglebone-io-python #231: config-pin not working for eqep0, 2, 2b

czyskows, 2018-02-08:
beaglebone black, 4.4.87-ti-xenomai-r121

python 2

After: echo cape-universala > /sys/devices/platform/bone_capemgr/slots

root@bela:~# config-pin P8.11 qep
bash: /sys/devices/platform/ocp/ocp*P8_11_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P8_11_pinmux/state
root@bela:~# config-pin P8.12 qep
bash: /sys/devices/platform/ocp/ocp*P8_12_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P8_12_pinmux/state
root@bela:~# config-pin P9.27 qep
bash: /sys/devices/platform/ocp/ocp*P9_27_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P9_27_pinmux/state
root@bela:~# config-pin P8.15 qep
bash: /sys/devices/platform/ocp/ocp*P8_15_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P8_15_pinmux/state
root@bela:~# config-pin P8.16 qep
bash: /sys/devices/platform/ocp/ocp*P8_16_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P8_16_pinmux/state

However,
root@bela:~# config-pin P8.33 qep
root@bela:~# config-pin P8.35 qep # for eqep1 are both fine

I've looked at the cape-uiniversala overlay here: https://github.com/cdsteinkuehler/beaglebone-universal-io/blob/master/cape-universala-00A0.dts
I should be able to do config-pin on the pins listed, but I just get the above errors.

pdp7, 2018-02-08:
**Please paste the output of these commands:**
```
sudo /opt/scripts/tools/version.sh
```
```
cat /boot/uEnv.txt
```

The Cape Manager has been deprecated by [U-Boot Overlays](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays).  Since `/sys/devices/platform/bone_capemgr/slots` exists on your system, I believe it is not using U-Boot Overlays. 

I would recommend trying this image:
[Debian 9.3 "Stretch" iot](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Stretch_Snapshot_iot)

cape-universal and config-pin are now maintained in the [bb-overlays repo](https://github.com/beagleboard/bb.org-overlays/tree/master/tools/beaglebone-universal-io).

czyskows, 2018-02-08:
root@bela:~# sudo /opt/scripts/tools/version.sh
sudo: /opt/scripts/tools/version.sh: command not found
root@bela:~# /opt/scripts/tools/version.sh
-bash: /opt/scripts/tools/version.sh: No such file or directory
root@bela:~# cat /boot/uEnv.txt
cat: /boot/uEnv.txt: No such file or directory
root@bela:~# cat /mnt/boot/uEnv.txt
uenvcmd=load mmc ${mmcid}:1 ${fdtaddr} ${dtb};load mmc ${mmcid}:2 ${loadaddr} /boot/vmlinuz-${uname_r};setenv bootargs console=${console} root=/dev/mmcblk${mmcid}p2 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet;bootz ${loadaddr} - ${fdtaddr};

console=ttyS0,115200n8

uname_r=4.4.87-ti-xenomai-r121
dtb=am335x-bone-bela.dtb
#dtb=am335x-bone-bela-black-wireless.dtb
mmcid=0
enable_uboot_overlays=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
root@bela:~#

The image that I'm using is for the Bela Board audio cape (bela.io). It's a real-time image for low-latency audio processing and necessary for the work I'm doing with the Beaglebone, so I can't use a regular distribution of Debian. Is there a way to work around? Thank you for the help!

pdp7, 2018-02-08:
@czyskows please paste the output of:
```
cat /etc/debian_version
cat /etc/dogtag
```
I suspect this quite an old image given that `/opt/scripts/tools/version.sh` does not exist.

@RobertCNelson @giuliomoro are you familiar with what current image people should use for Bela?

giuliomoro, 2018-02-08:
This is our Bela image with Debian Stretch. We don't have the `/opt/scripts` in there, maybe we should? We are still on kernel 4.4.87, because we need to catch up with maintaining the Xenomai-patched kernel.
We are admittedly a bit confused by the death of capemgr and u-boot overlays, so we will follow up in the email thread with Robert about this very problem. Starting in March, we will have someone available to take care of all of this.

pdp7, 2018-02-08:
@giuliomoro here is some information that @RobertCNelson has added to the eLinux wiki:
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

RobertCNelson, 2018-02-08:
Yeah, we've been talking with Bela, i need bring back the v4.4.x/v4.9.x xenomai options.. ;)

pdp7, 2018-02-08:
@RobertCNelson any advice for @czyskows who wants to use both the Bela image and the eQEP peripheral?

czyskows, 2018-02-08:
root@bela:~# cat /etc/debian_version
9.1

root@bela:~# cat /etc/dogtag
cat: /etc/dogtag: No such file or directory

RobertCNelson, 2018-02-08:
right now a custom overlay, or tweak the bela base dtb..

RobertCNelson, 2018-02-08:
ps, where is bela's am335x-bela.dtb stored?

giuliomoro, 2018-02-08:
> ps, where is bela's am335x-bela.dtb stored?

`/mnt/boot/`

same as `uEnv.txt`

RobertCNelson, 2018-02-08:
Sorry, i meant in github/etc? then i can steal it and migrate it to an overlay. ;)

giuliomoro, 2018-02-08:
> right now a custom overlay, or tweak the bela base dtb..

e.g. : generated with http://www.kilobaser.com/blog/2014-07-28-beaglebone-black-devicetreeoverlay-generator ?

giuliomoro, 2018-02-08:
> Sorry, i meant in github/etc? then i can steal it and migrate it to an overlay. ;)

https://github.com/BelaPlatform/dtb-rebuilder/blob/4.4-ti/src/arm/am335x-bone-bela.dts
https://github.com/BelaPlatform/dtb-rebuilder/blob/4.4-ti/src/arm/am335x-bone-bela-common.dtsi

I think we decided to integrate our stuff in the `dts` because we were not sure about when the overlays would be loaded during boot, but happy to switch back to overlays if they are guaranteed to be loaded _early_

RobertCNelson, 2018-02-08:
Yeah, we do it via u-boot now, so there's no kernel delays or races to deal with.

RobertCNelson, 2018-02-08:
pushed a resync, now building so i can run test:
https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/3adae43c02c8ba477d2a3e882329847af82e2b31

Regards,

giuliomoro, 2018-02-08:
great! I am rebuilding too

RobertCNelson, 2018-02-08:
Just one config problem, pushed a fix, and got it booting now tagging for the build farm..

RobertCNelson, 2018-02-08:
@giuliomoro please pull, test, and push to this repo::

(please make sure to add your copyright info at the top: https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-BELA-B2.dts#L2 )

```
git clone https://github.com/beagleboard/bb.org-overlays
cd ./bb.org-overlays/
make
sudo make install
```

Then in /boot/uEnv.txt just have: (with the newer kernel)

```
uboot_overlay_addr0=/lib/firmware/BB-BELA-B2.dtbo

enable_uboot_overlays=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
``
and let's see what happens. ;)

Regards,

giuliomoro, 2018-02-09:
@RobertCNelson, actually, no it does not seem to apply the overlay:

I thought the uboot overlay was working in the first place, but it was not (settings were being applied by our dtb). Other issue is that we are running a very old u-boot (from 2017.03), so that's probably the issue now.

Anyhow, this is the full uEnv.txt:
```
uenvcmd=load mmc ${mmcid}:1 ${fdtaddr} ${dtb};load mmc ${mmcid}:2 ${loadaddr} /boot/vmlinuz-${uname_r};setenv bootargs console=${console} root=/dev/mmcblk${mmcid}p2 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet;bootz ${loadaddr} - ${fdtaddr};

console=ttyS0,115200n8

mmcid=0
enable_uboot_overlays=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1

uboot_overlay_addr0=/lib/firmware/BB-BELA-B2.dtbo
dtb=am335x-boneblack.dtb
uname_r=4.4.113-ti-xenomai-r146
```

Also, I can still see:
```
root@bela:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
```
Is it expected?

RobertCNelson, 2018-02-09:
Yeah, you'll need the latest u-boot:

```
https://rcn-ee.com/repos/bootloader/am335x_evm/MLO-am335x_evm-v2018.01-r12
https://rcn-ee.com/repos/bootloader/am335x_evm/u-boot-am335x_evm-v2018.01-r12.img
```
Drop the "dtb=am335x-boneblack.dtb", enable_uboot_overlays will force a base *.dtb that every boards builds off of.

It also doesn't like /uEnv.txt or "uenvcmd" as that goes down the legacy branch (non-u-boot overlays).

Serial log is best to double check what it's doing..

Regards,

giuliomoro, 2018-02-09:
Not lucky:

```
U-Boot SPL 2014.04-00014-g47880f5 (Apr 22 2014 - 13:23:54)
reading args
spl_load_image_fat_os: error reading image args, err - -1
reading u-boot.img
reading u-boot.img

U-Boot 2014.04-00014-g47880f5 (Apr 22 2014 - 13:23:54)

I2C:   ready
DRAM:  512 MiB
NAND:  0 MiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
*** Warning - readenv() failed, using default environment

Net:   <ethaddr> not set. Validating first E-fuse MAC
cpsw, usb_ether
Hit any key to stop autoboot:  0
gpio: pin 53 (gpio 53) value is 1
mmc0 is current device
gpio: pin 54 (gpio 54) value is 1
SD/MMC found on device 0
reading uEnv.txt
462 bytes read in 4 ms (112.3 KiB/s)
gpio: pin 55 (gpio 55) value is 1
Loaded environment from uEnv.txt
Importing environment from mmc ...
Checking if uenvcmd is set ...
gpio: pin 56 (gpio 56) value is 1
Running uenvcmd ...
reading zImage
** Unable to read file zImage **
8393568 bytes read in 475 ms (16.9 MiB/s)
Kernel image @ 0x82000000 [ 0x000000 - 0x801360 ]
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree

uenvcmd was not defined in uEnv.txt ...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
mmc1(part 0) is current device
gpio: pin 54 (gpio 54) value is 1
SD/MMC found on device 1
reading uEnv.txt
1430 bytes read in 4 ms (348.6 KiB/s)
gpio: pin 55 (gpio 55) value is 1
Loaded environment from uEnv.txt
Importing environment from mmc ...
Checking if uenvcmd is set ...
gpio: pin 56 (gpio 56) value is 1
Running uenvcmd ...
reading zImage
3717760 bytes read in 206 ms (17.2 MiB/s)
reading initrd.img
2868972 bytes read in 160 ms (17.1 MiB/s)
reading /dtbs/am335x-boneblack.dtb
25080 bytes read in 9 ms (2.7 MiB/s)
Kernel image @ 0x82000000 [ 0x000000 - 0x38ba80 ]
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Using Device Tree in place at 88000000, end 880091f7
```

giuliomoro, 2018-02-09:
ok, after removing the `MLO` file from  the eMMC, it tries to use the correct u-boot, which is on the SD card, however it still fails `SD/MMC found on device 0 ** Bad device 0:2 0x82000000 ** ** Bad device 0:2 0x82000000 **`. Maybe this has something to do with some customizations @lbdonovan knows about?

```
U-Boot SPL 2018.01-00002-g9aa111a004 (Jan 20 2018 - 12:45:29)
Trying to boot from MMC1
reading u-boot.img
reading u-boot.img

U-Boot 2018.01-00002-g9aa111a004 (Jan 20 2018 - 12:45:29 -0600), Build: jenkins-github_Bootloader-Builder-32

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Power-on reset has occurred.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Using default environment

Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   eth0: MII MODE
cpsw, usb_ether
Press SPACE to abort autoboot in 2 seconds
board_name=[A335BNLT] ...
board_rev=[00C0] ...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
** Bad device 0:2 0x82000000 **
** Bad device 0:2 0x82000000 **
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc0 is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
reading uEnv.txt
465 bytes read in 4 ms (113.3 KiB/s)
gpio: pin 55 (gpio 55) value is 1
Loaded environment from /uEnv.txt
Importing environment from mmc ...
Checking if uenvcmd is set ...
Checking if client_ip is set ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
** Invalid partition 3 **
** Invalid partition 4 **
** Invalid partition 5 **
** Invalid partition 6 **
** Invalid partition 7 **
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc1(part 0) is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
reading uEnv.txt
1430 bytes read in 4 ms (348.6 KiB/s)
gpio: pin 55 (gpio 55) value is 1
Loaded environment from /uEnv.txt
Importing environment from mmc ...
Checking if uenvcmd is set ...
gpio: pin 56 (gpio 56) value is 1
Running uenvcmd ...
** Invalid partition 7 **
** Invalid partition 7 **
** Invalid partition 7 **
Checking if client_ip is set ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
** Invalid partition 3 **
** Invalid partition 4 **
** Invalid partition 5 **
** Invalid partition 6 **
** Invalid partition 7 **
```

RobertCNelson, 2018-02-09:
I left this as that's the u-boot default in mainline, thus it's easier to test pure mainline, if you have the rootfs setup for mainline..
```
mmc0 is current device
SD/MMC found on device 0
** Bad device 0:2 0x82000000 **
** Bad device 0:2 0x82000000 **
switch to partitions #0, OK
```

For U-Boot overlays, you need /boot/uEnv.txt:

```
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
reading uEnv.txt
465 bytes read in 4 ms (113.3 KiB/s)
gpio: pin 55 (gpio 55) value is 1
Loaded environment from /uEnv.txt
Importing environment from mmc ...
Checking if uenvcmd is set ...
Checking if client_ip is set ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
```
Looks like it grabbed /uEnv.txt, remove that file and make sure you only have /boot/uEnv.txt

Since U-Boot overlays relies on the latest U-Boot i had to separate it from the /uEnv.txt method.

Regards,

czyskows, 2018-02-14:
> @czyskows <https://github.com/czyskows> There was a good discussion here
> between @giuliomoro <https://github.com/giuliomoro> and @RobertCNelson
> <https://github.com/robertcnelson>, but I'd like to sync up on the
> original issue.
>
> *Is eqep0/2/2b still an issue for you?*
>
> Also, for config-pin issues in general, this is the official repo for it:
> https://github.com/beagleboard/bb.org-overlays/
> tree/master/tools/beaglebone-universal-io

Hi Drew,

I was following the discussion, but it seems that nothing ever got sorted.
I still haven't been able to get the eqeps to work as a result. I'll follow
up again with Giulio. Thanks for checking on this.

Cheers,
Colin

czyskows, 2018-02-14:
Is there a working fix for the eqeps on the Bela, then? @giuliomoro Have you been able to successfully configure the pins for eqep at any point?

Cheers,
Colin

pdp7, 2018-02-14:
@czyskows I'm not too familiar with the composition of the Bela image.  Hopefully @giuliomoro can make a suggestion.

One thing to try though, you could clone bb.org-overlays and install it:
https://github.com/beagleboard/bb.org-overlays
> Step 1: Clone this repo:
> ```git clone https://github.com/beagleboard/bb.org-overlays```
> ```cd ./bb.org-overlays```
> Step 2: Install *.dtbo:
> ```./install.sh```

czyskows, 2018-02-14:
Ok. Thanks. I'll give that a shot and let you know if I have any success.

giuliomoro, 2018-02-14:
Sorry if I left the conversation there, I can try again later today. Point is we have a filesystem with two partitions, "old" style, with a small boot FAT32 partition  and a larger ext4 partition, so I am not sure how the stock uboot will play with itl.

giuliomoro, 2018-02-14:
@czyskows reading back your original post, it seems that the problem may be that we are setting pins P8.11, P8.12, P8.27, P9.27, P8.15, P8.16 in our base  device tree. You may have to edit those muxer settings out of https://github.com/BelaPlatform/dtb-rebuilder/blob/4.4-ti/src/arm/am335x-bone-bela-common.dtsi and recompile it.

The file should be on the board in `/opt/dtb-rebuilder` and you should be able to re-build (after edit) with `make src/arm/am335x-bone-bela.dtb` and then you need to copy the generated file to `/mnt/boot`. However, be careful when copying stuff to `/mnt/boot` while it is mounted on the host, as it may easily get corrupted. In fact, I'd suggest you run this from the host:
```
scp root@192.168.7.2:/opt/dtb-rebuilder/src/arm/am335x-bone-bela.dtb /Volumes/BELABOOT/
```
if the volume is mounted on `/Volumes/BELABOOT`.

czyskows, 2018-02-14:
> blob/4.4-ti/src/arm/am335x-bone-bela-common.dtsi and recompile it.

Cool. I'll give that a shot. Thank you!

czyskows, 2018-02-15:
@giuliomoro That seems to have done it!  I just commented out the lines for those pins in the am335x-bone-bela-common.dtsi file, recompiled, copied, rebooted and was able to do the pin config:
root@bela:~# config-pin P8.11 qep
root@bela:~# config-pin P8.12 qep
root@bela:~# config-pin P8.33 qep
root@bela:~# config-pin P8.35 qep

Thank you so much for all of the help with this and sorry for being such a pain (as always)!

Cheers,
Colin

pdp7, 2018-02-15:
@czyskows good to hear. ok to close this now?

czyskows, 2018-02-15:
Sure thing.  Thanks!

closed 2018-02-15 by pdp7