adafruit/adafruit-beaglebone-io-python #232: GPIO.setup not working on 4.9-bone

Thalaivar, 2018-02-09:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **BeagleBone Blue Debian 9.3 IOT 4.9-bone**

- Python version (run `python -version` or `python3 -version`):  **Python 2.7.13**

- Error message you are receiving, including any Python exception traces:  **Set gpio value failed, missing file or invalid permissions***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **`import Adafruit_BBIO.GPIO as GPIO`
                   ` GPIO.setup("RED_LED", GPIO.OUT)`**

pdp7, 2018-02-10:
@Thalaivar It would be helpful if you could run strace so that I can better understand what is happening.

Install strace (if it doesn't already exist):

```sudo apt-get install strace```

Then run strace, replace test.py with the name of your test program:

```strace -f -o strace.log python test.py```

Paste the contents of strace.log into a gist.

Thalaivar, 2018-02-10:
Heres the strace.log: https://gist.github.com/Thalaivar/4dba9f39761810616ddff77fdb6018bd

pdp7, 2018-02-11:
@Thalaivar thanks.  I see that Python fails to open this file:
```
open("/sys/class/leds/red/brightness", O_WRONLY) = -1 ENOENT (No such file or directory)
```
Please paste the output of these commands:
```
cat /proc/device-tree/model 
find /sys/class/leds/
```

Thalaivar, 2018-02-11:
Here's the output of `/sys/class/leds/`:
``` 
    /sys/class/leds/
    /sys/class/leds/beaglebone:green:usr3
    /sys/class/leds/beaglebone:green:usr1
    /sys/class/leds/wl18xx_bt_en
    /sys/class/leds/beaglebone:green:usr2
    /sys/class/leds/beaglebone:green:usr0
```

and of `cat /proc/device-tree/model`:
```
TI AM335x BeagleBone Blue
```

I think its an issue with the 4.9-bone dt

pdp7, 2018-02-12:
@RobertCNelson this is odd to me.  A BeagleBone Blue is missing `/sys/class/leds` entries for the battery LEDs and the red LED.  Any ideas?

RobertCNelson, 2018-02-12:
i don't think i spent much time backporting much of the blue to v4.9.x-bone kernel, while the v4.14.x-bone kernel will have all the settings i set to mainline.  v4.14.x-bone should also easily allow uio-pruss.

Thalaivar, 2018-02-13:
@RobertCNelson I spent a considerable time screwing around trying to get uio-pruss to work on the 4.9-ti kernel, until I found out it was possible only on 4.9bone, due to some u-boot overlays being disabled or something, and I'm not too clear about all this stuff 😢 , so I have a few questions:

1) Can any work be done to improve the bone kernel so that it is more in sync with the other stuff like 4.9-ti, because a lot of stuff like the LEDs, like the battery LEDs and Wi-Fi LED don't work at all. Also the pinmux is really weird.

2) Will the 4.14-bone allow me to use the uio-pruss interface to use the PRUs? Also, the pinmux on the 4.9-bone is not setup well, will the same problems be there in 4.14x?

pdp7, 2018-02-13:
4.14 bone

pdp7, 2018-02-14:
@Thalaivar I was talking to @RobertCNelson today and we believe uio-pruss should work OK with 4.14 bone kernel.  Please try installing with this command:
```
sudo /opt/scripts/tools/update_kernel.sh --lts-4_14
```
@RobertCNelson Would @Thalaivar need to use [dtb-rebuilder](https://github.com/RobertCNelson/dtb-rebuilder) to make device tree changes?

Thalaivar, 2018-02-14:
the uio_pruss is not being enabled on 4.14 bone kernel even after running modprobe uio_pruss and also after uncommenting the u_boot overlay for uio pruss in the uEnv.txt... please advise

pdp7, 2018-02-14:
@RobertCNelson any ideas?

Thalaivar, 2018-02-14:
@RobertCNelson  Please advise on how I can enable uio_pruss on kernel 4.9-ti-rt or 4.14-bone-rt or 4.14-ti-rt

RobertCNelson, 2018-02-14:
So on the Black:

```
sudo ./update_kernel.sh --bone-kernel --lts-4_14
```
/boot/uEnv.txt:
```
uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
```
```
debian@beaglebone:~$ dmesg | grep pruss
[   40.184421] omap_hwmod: pruss: _wait_target_ready failed: -16
[   40.190294] pruss_uio 4a300000.pruss: use pm_runtime_put_sync_suspend() in driver?
[   40.197914] pruss_uio 4a300000.pruss: pm_runtime_get_sync() failed
[   40.456897] pruss_uio: probe of 4a300000.pruss failed with error -16
debian@beaglebone:~$ lsmod | grep pruss
uio_pruss              16384  0
uio                    16384  2 uio_pruss,uio_pdrv_genirq
```
When i move it to the Blue, the same image shows:

```
Loaded environment from /boot/uEnv.txt
Checking if uname_r is set in /boot/uEnv.txt...
gpio: pin 56 (gpio 56) value is 1
Running uname_boot ...
loading /boot/vmlinuz-4.14.19-bone12 ...
9302528 bytes read in 628 ms (14.1 MiB/s)
uboot_overlays: [uboot_base_dtb=am335x-boneblue.dtb] ...
uboot_overlays: Switching too: dtb=am335x-boneblue.dtb ...
loading /boot/dtbs/4.14.19-bone12/am335x-boneblue.dtb ...
54393 bytes read in 62 ms (856.4 KiB/s)
uboot_overlays: add [enable_uboot_overlays=1] to /boot/uEnv.txt to enable...
loading /boot/initrd.img-4.14.19-bone12 ...
4455918 bytes read in 326 ms (13 MiB/s)
```
Looks like i broke something in U-Boot, as "enable_uboot_overlays=1" is enabled properly (same SDCARD just moved from the Black -> Blue)..

Heading to a meeting, then lunch, so i won't have to time to debug it till this afternoon.

Regards,

mvduin, 2018-02-14:
@RobertCNelson To me it looks like u-boot overlays aren't supported at all for the beaglebone blue right now. `board/ti/am335x/board.c` sets `uboot_base_dtb=am335x-boneblue.dtb`, which is not a "base" dtb at all, and in particular enables remoteproc-pru.

RobertCNelson, 2018-02-14:
Fudge:

https://github.com/RobertCNelson/Bootloader-Builder/blob/master/patches/v2018.01/0002-U-Boot-BeagleBone-Cape-Manager.patch#L684-L685

```
+			"if test $board_name = BBBL; then " \
+				"env delete -f enable_uboot_overlays; fi; " \
```
i was protecting something when i did that, i think replacing that with "env delete -f uboot_try_cape_universal" just needs to be cleared..

be back in an hour, i'll replace that line..

Regards,

RobertCNelson, 2018-02-14:
Okay fix incoming:

```
Running uname_boot ...
loading /boot/vmlinuz-4.14.19-bone12 ...
9302528 bytes read in 628 ms (14.1 MiB/s)
uboot_overlays: [uboot_base_dtb=am335x-boneblue.dtb] ...
uboot_overlays: Switching too: dtb=am335x-boneblue.dtb ...
loading /boot/dtbs/4.14.19-bone12/am335x-boneblue.dtb ...
54393 bytes read in 62 ms (856.4 KiB/s)
uboot_overlays: [fdt_buffer=0x60000] ...
uboot_overlays: loading /lib/firmware/AM335X-PRU-UIO-00A0.dtbo ...
853 bytes read in 91 ms (8.8 KiB/s)
loading /boot/initrd.img-4.14.19-bone12 ...
4455918 bytes read in 325 ms (13.1 MiB/s)

debian@beaglebone:/sys$ lsmod | grep uio
uio_pruss              16384  0
uio_pdrv_genirq        16384  0
uio                    16384  2 uio_pruss,uio_pdrv_genirq

debian@beaglebone:/sys$ uname -r
4.14.19-bone12
```
https://github.com/RobertCNelson/Bootloader-Builder/commit/a1d090f01ef0d291817dac12fb220507226349fc

```
 +			"if test $board_name = BBBL; then " \
-+				"env delete -f enable_uboot_overlays; fi; " \
++				"if test -n ${uboot_try_cape_universal}; then " \
++					"env delete -f uboot_try_cape_universal; " \
++				"fi;" \
++			"fi;" \
```

RobertCNelson, 2018-02-14:
Okay, now pushed out, just run: (with a network connection)

```
sudo /opt/scripts/tools/developers/update_bootloader.sh
```

It'll dl/install 'r13'
```
Warning: this script will flash your bootloader with:
u-boot-mmc-spl.bin: [MLO-am335x_evm-v2018.01-r13]
u-boot.bin: [u-boot-am335x_evm-v2018.01-r13.img]
for: [am335x_evm]
-----------------------------
log: dd if=/tmp/tmp.r2aVuNIkrZ/dl/MLO-am335x_evm-v2018.01-r13 of=/dev/mmcblk0 seek=1 bs=128k
log: dd if=/tmp/tmp.r2aVuNIkrZ/dl/u-boot-am335x_evm-v2018.01-r13.img of=/dev/mmcblk0 seek=1 bs=384k

Are you 100% sure, on selecting [am335x_evm] (y/n)? 
```
/dev/uio* should show up:

```
debian@beaglebone:~$ ls /dev/ui*
/dev/uinput  /dev/uio1  /dev/uio3  /dev/uio5  /dev/uio7
/dev/uio0    /dev/uio2  /dev/uio4  /dev/uio6
```

if it still fails let's last check:

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh | grep bootloader
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g60ae27e116]:[location: dd MBR]
```

Regards,

pdp7, 2018-02-15:
@Thalaivar were you able to try `sudo /opt/scripts/tools/developers/update_bootloader.sh`?

Thalaivar, 2018-02-15:
I ran `sudo /opt/scripts/tools/developers/update_bootloader.sh` and went ahead with `Are you 100% sure, on selecting [am335x_evm] (y/n)? ` but on reboot `ls /dev/ui*` does not show up.
```
debian@beaglebone:/sys$  sudo /opt/scripts/tools/version.sh | grep bootloader
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g60ae27e116]:[location: dd MBR]
```
```
debian@beaglebone:/sys$ uname -r
4.14.19-bone12
```

Thalaivar, 2018-02-15:
I ran `sudo /opt/scripts/tools/developers/update_bootloader.sh` and went ahead with `Are you 100% sure, on selecting [am335x_evm] (y/n)? ` but on reboot `ls /dev/ui*` does not show up.
```
debian@beaglebone:/sys$  sudo /opt/scripts/tools/version.sh | grep bootloader
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g60ae27e116]:[location: dd MBR]
```
```
debian@beaglebone:/sys$ uname -r
4.14.19-bone12
```
```
debian@beaglebone:/sys$ ls /dev/ui*
/dev/uinput
```

RobertCNelson, 2018-02-15:
@Thalaivar did you specify the pru in /boot/uEnv.txt ?

```
uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
```
Yes the kernel is in control of the imu in v4.14.x

You can get it back via for userspace control by either doing:

```
sudo rmmod inv_mpu6050_i2c
sudo rmmod inv_mpu6050
sudo rmmod ak8975
```
Or create a blacklist:
```
debian@beaglebone:~$ cat /etc/modprobe.d/mpu.conf 
blacklist inv_mpu6050_i2c
blacklist inv_mpu6050
blacklist ak8975
```

```
debian@beaglebone:~$ i2cdetect -r 2
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-2 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- UU -- 
```
Regards,

Thalaivar, 2018-02-16:
Thanks a lot, everything is working well now. Just as one last question, how can I enable the BMP280 in the same manner?

closed 2018-02-16 by Thalaivar

RobertCNelson, 2018-02-16:
i was thinking about those too.. just type lsmod, they'll be one or two, just add them to the blacklist..

Regards,

Thalaivar, 2018-02-24:
Hi, I ran lsmod and here's the output, its only showing bmp280:
```
Module                  Size  Used by
xt_conntrack           16384  1
ipt_MASQUERADE         16384  1
nf_nat_masquerade_ipv4    16384  1 ipt_MASQUERADE
wl18xx                110592  0
wlcore                233472  1 wl18xx
pruss_soc_bus          16384  0
uio_pruss              16384  0
bmp280_i2c             16384  0
bmp280                 20480  1 bmp280_i2c
wlcore_sdio            16384  0
evdev                  24576  1
uio_pdrv_genirq        16384  0
uio                    20480  2 uio_pruss,uio_pdrv_genirq
usb_f_mass_storage     53248  2
usb_f_acm              16384  2
u_serial               20480  3 usb_f_acm
usb_f_ecm              20480  2
usb_f_rndis            32768  2
u_ether                24576  2 usb_f_ecm,usb_f_rndis
libcomposite           61440  16 usb_f_ecm,usb_f_acm,usb_f_mass_storage,usb_f_rndis
iptable_nat            16384  1
nf_conntrack_ipv4      20480  3
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
nf_nat_ipv4            16384  1 iptable_nat
nf_nat                 53248  2 nf_nat_masquerade_ipv4,nf_nat_ipv4
nf_conntrack          167936  6 nf_conntrack_ipv4,ipt_MASQUERADE,nf_nat_masquerade_ipv4,xt_conntrack,nf_nat_ipv4,nf_nat
iptable_mangle         16384  0
iptable_filter         16384  1
binfmt_misc            20480  1
ip_tables              24576  3 iptable_mangle,iptable_filter,iptable_nat
x_tables               32768  5 iptable_mangle,ip_tables,iptable_filter,ipt_MASQUERADE,xt_conntrack
```