adafruit/adafruit-beaglebone-io-python #234: GPIO input don't work when UARTs uboot_overlay cape enable

DavidCambrotta, 2018-02-20:
Hi,
I have read some other issues with the same problem, but nothing exactly like mine, and i'm current using a BeagleBone Black Rev C.
I start using Debian 9.3 2018-01-28 4GB SD IoT image (Kernel 4.9) **with no modify done in /boot/uEnv.txt file, the I/Os pins works correctly for both modes: GPIO.OUT and GPIO.IN.**

Then when i enable UARTs (1,4 & 5) and SPI 0 in the /boot/uEnv.txt:

uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-UART4-00A0.dtbo
uboot_overlay_addr2=/lib/firmware/BB-UART5-00A0.dtbo
uboot_overlay_addr3=/lib/firmware/BB-SPIDEV0-00A0.dtbo 

cape_enable=bone_capemgr.enable_partno=BB-UART1,BB-UART4,BB-UART5,BB-SPIDEV0

And disable video, audio, wireless, adc:

disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1

cape_disable=bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN

After this enables and disables, the GPIO pins only works in GPIO.OUT mode, if i use pins in the GPIO.IN, gives the next error:

>>> import Adafruit_BBIO.GPIO as GPIO
>>> GPIO.setup("P8_18",GPIO.OUT)
>>> GPIO.setup("P8_18",GPIO.IN)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Set gpio mode failed, missing file or invalid permissions.

In other issues these 2 commands are always asked, where they are:

root@beaglebone:~# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLT00C03317BBBK13B9]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]
kernel:[4.9.78-ti-r94]
nodejs:[v6.13.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/BB-UART5-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr3=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    1.368338] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END

root@beaglebone:~# dmesg | grep bone
[    0.000000] Kernel command line: console=ttyO0,115200n8 bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN bone_capemgr.enable_partno=BB-UART1,BB-UART4,BB-UART5,BB-SPIDEV0 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
[    5.390948] systemd[1]: Set hostname to <beaglebone>.

RobertCNelson, 2018-02-20:
@DavidCambrotta okay give this a shot, a lot of this still in flight, but a good chuck is ready for users to test, and just pushed this for you: https://github.com/beagleboard/bb.org-overlays/commit/d65f62eeef5410f8a28d9ded9a673164383895fb

Step 1, install this version of u-boot
```
sudo /opt/scripts/tools/developers/update_bootloader.sh --beta
```
Step 2: install this kernel:
```
sudo /opt/scripts/tools/update_kernel.sh --bone-channel --exp
```
Step 3: Update overlays
```
cd /opt/source/bb.org-overlays/
git pull
make
sudo make install
```
Step 4: Reboot
```
sudo reboot
```
Step 5: Verify:
```
sudo /opt/scripts/tools/version.sh | grep bootloader
ls /sys/devices/platform/ocp/ | grep pinmux []?
```

DavidCambrotta, 2018-02-20:
Done.

root@beaglebone:~# sudo /opt/scripts/tools/version.sh | grep bootloader
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-rc2-00002-ged1e50]

root@beaglebone:~# ls /sys/devices/platform/ocp/ | grep pinmux*
ocp:P8_07_pinmux
ocp:P8_08_pinmux
ocp:P8_09_pinmux
ocp:P8_10_pinmux
ocp:P8_11_pinmux
ocp:P8_12_pinmux
ocp:P8_13_pinmux
ocp:P8_14_pinmux
ocp:P8_15_pinmux
ocp:P8_16_pinmux
ocp:P8_19_pinmux
ocp:P8_26_pinmux
ocp:P8_27_pinmux
ocp:P8_28_pinmux
ocp:P8_29_pinmux
ocp:P8_30_pinmux
ocp:P8_31_pinmux
ocp:P8_32_pinmux
ocp:P8_33_pinmux
ocp:P8_34_pinmux
ocp:P8_35_pinmux
ocp:P8_36_pinmux
ocp:P8_39_pinmux
ocp:P8_40_pinmux
ocp:P8_41_pinmux
ocp:P8_42_pinmux
ocp:P8_43_pinmux
ocp:P8_44_pinmux
ocp:P8_45_pinmux
ocp:P8_46_pinmux
ocp:P9_12_pinmux
ocp:P9_14_pinmux
ocp:P9_15_pinmux
ocp:P9_16_pinmux
ocp:P9_17_pinmux
ocp:P9_18_pinmux
ocp:P9_19_pinmux
ocp:P9_20_pinmux
ocp:P9_23_pinmux
ocp:P9_25_pinmux
ocp:P9_27_pinmux
ocp:P9_28_pinmux
ocp:P9_29_pinmux
ocp:P9_30_pinmux
ocp:P9_31_pinmux
ocp:P9_41_pinmux
ocp:P9_42_pinmux
ocp:P9_91_pinmux
ocp:P9_92_pinmux

After the verify, i did this:
>>> import Adafruit_BBIO.GPIO as GPIO
>>> GPIO.setup("P8_18", GPIO.IN)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Set gpio mode failed, missing file or invalid permissions.
>>> GPIO.setup("P8_16", GPIO.IN)
>>> GPIO.setup("P8_19", GPIO.IN)

So, the pins P8_19 and P8_16 works fine (choose the pins from the /sys/devices/platform/ocp/  list), but P8_18 don't, and that pin isn't being used in nothing ( I can use other pin for my porpuse with no problem, now I just don't understand why I can't configure that pin in IN mode, but works in OUT mode).

Thanks for the help @RobertCNelson 

Best Regards,
David Caniceiro

RobertCNelson, 2018-02-20:
@DavidCambrotta sorry, that was mistake on my side:

https://github.com/beagleboard/bb.org-overlays/commit/ea037d511a005025b01437430c5ee2e01e81cde9

Step 1: Update overlays
```
cd /opt/source/bb.org-overlays/
git pull
make
sudo make install
```
Step 2: Reboot
```
sudo reboot
```

DavidCambrotta, 2018-02-20:
No problem, working fine now.

Really thanks for the support @RobertCNelson

pdp7, 2018-02-21:
thanks @RobertCNelson 

@DavidCambrotta ok to close?

DavidCambrotta, 2018-02-21:
Yes @pdp7, you can close the issue.

closed 2018-02-22 by pdp7