adafruit/adafruit-beaglebone-io-python #236: Controlling servo motors in beaglebone blue

DarioushK1, 2018-02-28:
Hi, I'm kinda new to beaglebones. I've used Adafriut_BBIO.pwm for controlling servo motors in beaglebone black, now I'm looking for something similar so I can control servo motors in beaglebone blue. Does Adafriut_BBIO work for the blue?
I use subprocess.call to run rc_test_servos in my python code to control servo motors. but the problem is that I can not exit the code until I press ctrl+c. I want rc_test_servos to run multiple times in my code but nothing can exit the rc_test_servos in python (even sys.exit(), subprocess.kill(), subprocess.terminate() and so on). I will appreciate if anyone can help me.

Thank you.
Darioush

pdp7, 2018-02-28:
Please post the code so that I can attempt to reproduce the issue.

Also, post the output of:
`sudo /opt/scripts/tools/version`

DarioushK1, 2018-02-28:
Thank you for your prompt response.
The output of sudo /opt/scripts/tools/version.sh is :

git:/opt/scripts/:[2c55e41af686d7e901dc67118397d3b07f7bc039]
eeprom:[A335BNLTBLA21717EL004759]
dogtag:[BeagleBoard.org Debian Image 2017-04-09]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.05-rc1-00002-g35aecb]
kernel:[4.4.59-ti-r96]
nodejs:[v4.8.7]

Honestly, I am receiving some joystick input from my PC and sending it with a client/server python application(LAN) to the board. The goal is to control the servo position with the joystick for steering purposes. But the problem is that after running the codes, the servo goes to a certain position and when I hit another Joystick button the codes doesn't work anymore and I have to do cntl+c to exit out of the program.

**The server code on my pc is:**

import pygame
import PodSixNet.Channel
import PodSixNet.Server
from pygame import *
from time import sleep
init()
from time import sleep

class ClientChannel(PodSixNet.Channel.Channel):
    def Network(self, data):
        print data
    def Close(self):
        self._server.close(self.gameid)

class BoxesServer(PodSixNet.Server.Server):

    channelClass = ClientChannel

    def __init__(self, *args, **kwargs):
        PodSixNet.Server.Server.__init__(self, *args, **kwargs)
        self.games = []
        self.queue = None
        self.currentIndex=0
    def Connected(self, channel, addr):
        print 'new connection:', channel
        if self.queue==None:
            self.currentIndex+=1
            channel.gameid=self.currentIndex
            self.queue=Game(channel, self.currentIndex)
    def close(self, gameid):
        try:
            game = [a for a in self.games if a.gameid==gameid][0]
            game.player0.Send({"action":"close"})
        except:
            pass
    def tick(self):
        if self.queue != None:
            sleep(.05)
            for e in event.get():
                self.queue.player0.Send({"action":"gamepad","type":e.type, "info":e.dict})
        self.Pump()

class Game:
    def __init__(self, player0, currentIndex):
        #initialize the players including the one who started the game
        self.player0=player0
#Setup and init joystick
j=joystick.Joystick(0)
j.init()
#Check init status
if j.get_init() == 1: print "Joystick is initialized"
#Get and print joystick ID
print "Joystick ID: ", j.get_id()
#Get and print joystick name
print "Joystick Name: ", j.get_name()
#Get and print number of axes
print "No. of axes: ", j.get_numaxes()
#Get and print number of trackballs
print "No. of trackballs: ", j.get_numballs()
#Get and print number of buttons
print "No. of buttons: ", j.get_numbuttons()
#Get and print number of hat controls
print "No. of hat controls: ", j.get_numhats()

print "STARTING SERVER ON LOCALHOST"
# try:
address=raw_input("Host:Port (localhost:8000): ")
if not address:
    host, port="localhost", 8000
else:
    host,port=address.split(":")
boxesServe = BoxesServer(localaddr=(host, int(port)))
while True:
    boxesServe.tick()
    sleep(0.1)

**And the Client code on Beaglebone Blue is:**

import socket
import pygame
import math
from PodSixNet.Connection import ConnectionListener, connection
from time import sleep
import serial
import os
import sys
import subprocess

class QuadGame(ConnectionListener):
    def Network_close(self, data):
        exit()
    def Network_gamepad(self, data):
        #print (data)
        if data["type"] == 10:
            #print (data)
            print " Pressed button "
            #print (data)
            if data["info"]["button"] == 0:
                subprocess.call('rc_test_servos -v  -c 1 -p -1 -f 500', shell=True)

            if data["info"]["button"] == 5:
                subprocess.call('rc_test_servos -v -c 1 -p 1 -f 500', shell=True)

    def __init__(self):
        hostname = 'maps.google.com'
        addr = socket.gethostbyname(hostname)
        print 'The address of ', hostname, 'is', addr
        address=raw_input("Host:Port (localhost:8000): ")
        try:
            if not address:
                host, port="localhost", 8000
            else:
                host,port=address.split(":")
            self.Connect((host, int(port)))
        except:
            print "Error Connecting to Server"
            print "Usage:", "host:port"
            print "e.g.", "localhost:31425"
            exit()
        print "Quad client started"
        self.running=False
        while not self.running:
            self.Pump()
            connection.Pump()
            sleep(0.01)
bg=QuadGame()
while 1:
    if bg.update()==1:
        break
bg.finished()

pdp7, 2018-03-04:
Adafruit_BBIO does work with the BeagleBone Blue after @MarkAYoder added support.

You may also want to take a look at:
https://github.com/mcdeoliveira/rcpy
> Python Library for Robotics Cape on Beaglebone Black and Beaglebone Blue

pdp7, 2018-03-15:
@DarioushK1 is it ok to close this issue?

closed 2018-03-15 by DarioushK1