adafruit/adafruit-beaglebone-io-python #238: ValueError: Set gpio mode failed, missing file or invalid permissions.

nezra, 2018-03-07:
- Platform/operating system: Beaglebone Blue, Kernel 4_14(updated via update kernel script), Debian 9.3 2018-03-05 4GB SD IoT image, fresh install

- Python version (run `python -version` or `python3 -version`):  Python 3.5.3

- Error message you are receiving, including any Python exception traces:  

File "/var/lib/cloud9/nezrasand/button.py", line 16, in <module>
    GPIO.setup(button, GPIO.IN)
ValueError: Set gpio mode failed, missing file or invalid permissions.

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): running example code located here:

https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/examples/python/button.py

strace:

Traceback (most recent call last):
  File "button.py", line 16, in <module>
    GPIO.setup(button, GPIO.IN)
ValueError: Set gpio mode failed, missing file or invalid permissions.

config-pin compains
P8_09 pinmux file not found!
Pin has no cape: P8_09

pdp7, 2018-03-07:
Please paste the output of:
```
sudo /opt/scripts/tools/version.sh 

cat /boot/uEnv.txt 
```

pdp7, 2018-03-07:
@nezra I would suggest changing the example to:
```button= "P8_9"```

pdp7, 2018-03-07:
@nezra 
Please paste the output of:
```
sudo /opt/scripts/tools/version.sh 
```
and
```
cat /boot/uEnv.txt 
```
You may also want to try running python as root to determine if it is a permissions issue:
```
sudo python /var/lib/cloud9/nezrasand/button.py
```

nezra, 2018-03-07:
`sudo /opt/scripts/tools/version.sh`
`git:/opt/scripts/:[e307a944e0be0610ff5296e0abe4ad31a6e70daa]
eeprom:[A335BNLTBLA21722EL001255]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
kernel:[4.14.20-ti-r36]
nodejs:[v6.13.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180306.0-0rcnee0~stretch+20180306]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.927338] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper`

`cat /boot/uEnv.txt `
```
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.20-ti-r36
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

```
I tried running it as root with sudo, same result.

I changed the button number to directly match the pin tables as you indicated as "P8_9"; as well as trying "P8_10" and "MODE", and it all results in the same thing. it appears any attempt to use "GPIO.IN" on any pin results in the same issue.

The pin definitions for Blue are located in this file: 

https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c

starting on line 172, so the translation is there.

MarkAYoder, 2018-03-07:
@pdp7, the Blue has a PAUSE and a MODE button and in common.c I set:
    { "PAUSE", "P8_9",  69, 1, -1},
    { "MODE",  "P8_10", 68, 1, -1},

So you should be able to refer to them by name.

--Mark

nezra, 2018-03-07:
@MarkAYoder slightly off topic, and well beyond me. Would it be possible to add the pin in common.c for the placeholders for BAT100? BAT100 appears to be LED4 from the pin list based on the schematics, P8_14-> GPIO0_26. Robotics cape installer's RC_defs.h also lists this on line 75 confirming it.

nezra, 2018-03-07:
@pdp7 Is there a chance this is all caused because the universal cape overlay isn't loading? none of the ones at beaglebone-universal-io indicate they are "compatible" with Beaglebone Blue.

Querying the status of the pin with
`debian@beaglebone:/var/lib/cloud9$ config-pin -q P8.07
P8_07 pinmux file not found!
Pin has no cape: P8_07`

makes it sound like there's a fundamental issue with the cape, and the cause of this entire issue inside the software is that it can't configure the pin because the cape overlay isn't loading.

pdp7, 2018-03-07:
@robertnelson do you think the issue that @nezra mentioned above could be result of the bootloader?

```
debian@beaglebone:/var/lib/cloud9$ config-pin -q P8.07 P8_07 pinmux file not found! Pin has no cape: P8_07
```

pdp7, 2018-03-07:
@RobertCNelson  do you think the issue that @nezra mentioned above could be result of the bootloader?

```
debian@beaglebone:/var/lib/cloud9$ config-pin -q P8.07 P8_07 pinmux file not found! Pin has no cape: P8_07
```

nezra, 2018-03-07:
@RobertCNelson @pdp7 To be safe, I followed the instructions here: https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

and am on this:
`debian@beaglebone:/opt/scripts/tools$ sudo ./version.sh | grep bootloader

bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]`

still getting the same errors with both config-pin and beaglebone_bbio.

RobertCNelson, 2018-03-07:
@pdp7 Sorry, just noticed, TI_AM335x_BeagleBone_Blue and 4.14.20-ti-r36...

This currently doesn't have all the magic enabled, like we did in v4.4.x-ti/v4.9.x-ti.  It's pretty much what i submitted to mainline in v4.13.x and all the fixes since then. So none of the other pins are exported...

nezra, 2018-03-08:
So, is it safe to assume then i need to reflash to an older image, and not follow the instructions to update anything in order to get it to work?

RobertCNelson, 2018-03-08:
ah just downgrade via:

```
sudo /opt/scripts/tools/update_kernel.sh --lts-4_9
```

nezra, 2018-03-08:
I ran the script @RobertCNelson suggested.

confirmation:

`sudo /opt/scripts/tools/version.sh`
`git:/opt/scripts/:[e307a944e0be0610ff5296e0abe4ad31a6e70daa]
eeprom:[A335BNLTBLA21722EL001255]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
kernel:[4.9.83-ti-r103]
nodejs:[v6.13.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180306.0-0rcnee0~stretch+20180306]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.434571] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper`

Same error in python3 with and without sudo, and same config-pin issue.

I am afraid downgrading the kernel didn't fix it either.

nezra, 2018-03-09:
Bit of my own investigation. config-pin will always have errors as it checks against it's database of capes, which is currently blank. it also checks against pinmux files located in /sys/devices/platform/ocp/, which only contains pinmux files for p8.15, P9.11, P9.23, and P9.28.

Since P8.8 is the green led, and config-pin says it's not there, and the python example for blinking it works fine, the issue isn't there.

nezra, 2018-03-09:
debian@beaglebone:/var/lib/cloud9$ ls /sys/devices/platform/ocp/ | grep pinmux*
ocp:D13_pinmux
ocp:H17_pinmux
ocp:J15_pinmux
ocp:P8_15_pinmux
ocp:P9_11_pinmux
ocp:P9_23_pinmux
ocp:P9_28_pinmux
ocp:U16_pinmux

I'm assuming there's something wrong with the overlay that's being loaded at boot that's causing this?

RobertCNelson, 2018-03-09:
4.9.83-ti-r103
```
debian@beaglebone:/sys/devices/platform/ocp$ ls | grep pinmux
ocp:D13_pinmux
ocp:H17_pinmux
ocp:J15_pinmux
ocp:P8_15_pinmux
ocp:P9_11_pinmux
ocp:P9_23_pinmux
ocp:P9_28_pinmux
ocp:U16_pinmux
```

RobertCNelson, 2018-03-09:
Tagging: v4.14.25-ti-r38, boot tested, and should be 100% compatible with v4.4.x/v4.9.x..

nezra, 2018-03-09:
I may have misunderstood. I thought you indicated that if I reverted to the 4.9.x kernel the other GPIOs would be exported. The BBIO checks in that specific location to use & set them from what i can tell. Since the other pinmux "folders" aren't there, BBIO errors out.

If I misunderstood, then am I correct in assuming that the comment of "This currently doesn't have all the magic enabled, like we did in v4.4.x-ti/v4.9.x-ti. It's pretty much what i submitted to mainline in v4.13.x and all the fixes since then. So none of the other pins are exported..." means that those pins just aren't exported anywhere until such time as mainline does something with it?

RobertCNelson, 2018-03-09:
So there was two problems.

v4.14.x-ti was out of sync with v4.4.x-ti/v4.9.x-ti, that's now fixed.

So back to your actual problem:

Buttons: Pause & Mode:  Looking back at v4.4.x-ti/v4.9.x-ti they were "NEVER" exported to the pinmux folder (for config-pin and friends), instead they were hardcoded:

```
			/* GPIO Inputs */
			0x09c 0x37	/*P8.9  Pause BUTTON, input pullup*/
			0x098 0x37	/*P8.10 MODE BUTTON input pullup*/
```

and never exported to /sys/..

They were tied up by libroboticscape, for button usage.  Instead they should be routed thru gpio-keys:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/devicetree/bindings/input/gpio-keys.txt

Regards,

nezra, 2018-03-09:
In essence, does that mean that all of the pins i can't access are all hard coded into and by librobotiscape and that i am barking up the wrong tree(all puns intended) here because the adafruit_bbio won't ever be able to address them? i ask because 

I was looking for an alternative to librobotiscape given it appears to be largely ignored over the last year from the looks of it, and the additions that were committed to it only made available a portion of the functions needed via a fork that also appears to have died.

RobertCNelson, 2018-03-09:
it's author actually teaches a class with it, there is a newer branch, but i have not packaged it for the repo yet.  as root you should be able to export those pins manually under /sys/class/gpio/

nezra, 2018-03-09:
You've been most patient and helpful. I appreciate it. Since that no longer applies to the problem i'm closing this.

closed 2018-03-09 by nezra

nezra, 2018-03-10:
@RobertCNelson just an fyi. after updating to v4.14.25-ti-r38, ALL of the /ocp pinmux files are now gone. the robotics cape library now no longer works at all. since those folders are now missing, adafruit_bbio and rcpy also no longer work either.

nezra, 2018-03-11:
@RobertCNelson quick question, as i think i understand now based on some of these recent changes what's going wrong. feel free to correct me if i am wrong, and if i should be posting this elsewhere.

way back in the olden days of the kernel changes, Beaglebone Blue had it's own overlay definitions, and added them those in am335x-bone-common-universal.dtsi. if i understand correctly now after much reading how this works, the am335x-bone-common-universal.dtsi file broke out ALL of the pins, which, config-pin needed and as such adafruit_bbio & libroboticscape/rcpy. it also used it's own ADDITIONAL overlay from the robotics cape installer to make available the additional pins. 

Fast forward to the 4.4 and up kernels. the iterations from this point forward ONLY defined the pins in ocp/ with pinmux in the lists you posted above, and stopped including ALL of the other pins exported from am335x-bone-common-universal.dtsi. as a result, only these pinmux files area available as you indicated above:

```
ocp:D13_pinmux
ocp:H17_pinmux
ocp:J15_pinmux
ocp:P8_15_pinmux
ocp:P9_11_pinmux
ocp:P9_23_pinmux
ocp:P9_28_pinmux
ocp:U16_pinmux
```
now, i suppose the missing pinmux COULD be added by overlay, if i understand correctly, but so far this part is well above me.  

Basically, in order for config-pin, adafruit_bbio, rc_py, and the libroboticscape library to work correctly, we NEED all of these other pinmux files to be present. it's not optional. the code base specifically looks in the correct folder for files by the name type above. it;s why i cannot access pins via "P<header>_<pin number>" such as "P8_10", or timer, or any of the other files named in the pin table. 

So i went back to the universal capes dts files. for some reason, there's no compatible "ti,beaglebone-blue" type variant listed in "compatability". since the machine present's itself as a blue, and not a "beaglebone-black" greeen, etc, the overlay doesn't appear to be loading the rest of instructions in the overlay, and as such, not putting the pins in the /ocp/ folder under a name of "ocp:*pinmux".

If the "universal-cape" wont load the pins like expected, i suppose i could try and figure out how to fix the dts file myself, recompile it, and load it, but this seems like something that should be fixed in the base overlays itself. If there's another issue/reason why this isn't being done, let me know. as it stands something in  v4.14.25-ti-r38 went awry, and now none of even the custom pins are exported to the ocp directory as indicated above, and i have to back the kernel down to test it.

RobertCNelson, 2018-03-11:
That's incorrect for the Blue... For the Black/Green/etc it was true..

For the Blue, everything required for librobotic was pre-setup in the dts file.  (the team that came up with the blue also did the librobotic library)

Even up until a few weeks ago, we purposely disabled u-boot overlays. But we recently changed that to allow users to easily enable the ADC and change the PRU.

Regards,

RobertCNelson, 2018-03-11:
Ps it the overlay repo also get's confusing, as there is a reboticape overlay, that's actually for a cape you can plug into the Black, it was a very limited production run..

nezra, 2018-03-11:
so, in order to get the board to work correctly i have to disable u-boot somehow? because as the default image works, the /ocp files are missing that everything from that point forward requires to work.

every
time you try to do anything with libroboticscape's built in test functions(rc_blink, rc_test_adc, etc), it literally spits this out:

`can't open: /sys/devices/platform/ocp/ocp:H18_pinmux/state
Pinmux: No such file or directory
can't open: /sys/devices/platform/ocp/ocp:C18_pinmux/state
Pinmux: No such file or directory
can't open: /sys/devices/platform/ocp/ocp:P9_22_pinmux/state
Pinmux: No such file or directory
can't open: /sys/devices/platform/ocp/ocp:P9_21_pinmux/state
Pinmux: No such file or directory
can't open: /sys/devices/platform/ocp/ocp:P9_26_pinmux/state
Pinmux: No such file or directory
can't open: /sys/devices/platform/ocp/ocp:P9_24_pinmux/state
Pinmux: No such file or directory
can't open: /sys/devices/platform/ocp/ocp:P9_30_pinmux/state
Pinmux: No such file or directory
can't open: /sys/devices/platform/ocp/ocp:P9_29_pinmux/state
Pinmux: No such file or directory
can't open: /sys/devices/platform/ocp/ocp:P9_31_pinmux/state
Pinmux: No such file or directory
WARNING: missing PINMUX driver
You probbaly just need a newer kernel
ERROR: export failed for hrpwm1 channel A
ERROR: failed to initialize hrpwm1
WARNING: Failed to initialize motors
ERROR: pru-rproc driver missing`

in delving into the motor code, it's the same thing. missing pinmux folders in the aforementioned directory are the root cause of the headache. the adc & leds are built in so they work, but everything motor related and other functions simply don't work in the current iteration of the kernel and uboot.

If i can't create an overlay either to fix the omission upstream or re-add the missing ocp pinmux directories, what image should i be using to get the full functionality back?

At that point, it should be mentioned in the wiki that none of the newer kernels support the Blue and all of it's functionality.

nezra, 2018-03-11:
just an additional point.

This is from the libroboticscape.so library:

https://github.com/StrawsonDesign/Robotics_Cape_Installer/blob/ac7bcd73c2f5671b2bd83f241ee1908b0a23b767/libraries/other/rc_pinmux.c

You'll find the hard coded locations where they expect the the aforementioned directories to be broken out. as these are created by an overlay at boot, i manage to track down their creation point:

These included in the dtb-rebuilder overlay files, as you can find here:

https://github.com/RobertCNelson/dtb-rebuilder/blob/a12383ce8a761dedec8238e7fa8be5703af3eb72/src/arm/am335x-boneblue.dts

and here:

https://github.com/RobertCNelson/dtb-rebuilder/blob/adb9af2bcf01e2bc962c2f4d0d0562c22aaf9379/src/arm/am335x-boneblue-mmap.dts

but chucks of the pin setup are omitted as it appears now, as are the ocp portions. based on this, an looking back through the images, i can assume the 4-9-2017 image that contains 8.x Debian and the 4.4 kernel to be the only working iteration.

the driver only calls the functions that create/initialize the GPIO setup and destroys them when the rc_initialize function is called. it doesn't set up the pinmuxing inside the driver itself, but instead relied on the overlays of the old kernel/bootloader/capemanager to do so. 

Regarding enabling overlay creation, Is there a reason we can't disable the driver/service and be allowed to create our own complete overlays? as with any SBC, anything we do, we do at our own risk, and we accept said risk.

RobertCNelson, 2018-03-12:
Yeah, that's why i push out the dtb-rebuilder repo, so you can easily customize the blue/etc *.dtb's..

I'm in portland this week, with just a bbbw, i'll take a look at this when i get back to work on friday.

Regards,