adafruit/adafruit-beaglebone-io-python #239: EHRPWM2A is also known in some cultures as GPIO0_22

ThatWileyGuy, 2018-03-10:
The table in source/common.c currently names P8_19 as EHRPWM2A, which is correct, but this pin can also be muxed to GPIO and used as GPIO0_22. I don't think there's currently a way to represent both of these names in the table.

Attempting to use GPIO0_22 currently produces an exception with "failed: error return without exception set"

Is there a way we could get this pin to work under both names?

pdp7, 2018-03-15:
@ThatWileyGuy I will take a look at if there is a way to do this.  thanks.

filipkotoucek, 2018-03-21:
I also got errors when I try to setup certain GPIO pins at Beaglebone. For example. Both pins P9_29 and P9_31 are pinmuxed as GPIO (mode 7). I can manualy export them from linux and control them. But with BBIO lib I got error at one of these:

`(redeem) root@beaglebone:/sys/class/gpio# ls
export	gpiochip0  gpiochip32  gpiochip64  gpiochip96  unexport
(redeem) root@beaglebone:/sys/class/gpio# echo 111 > export
(redeem) root@beaglebone:/sys/class/gpio# echo 110 > export
(redeem) root@beaglebone:/sys/class/gpio# ls
export	gpio110  gpio111  gpiochip0  gpiochip32  gpiochip64  gpiochip96  unexport
(redeem) root@beaglebone:/sys/class/gpio# python 
Python 2.7.13 (default, Nov 24 2017, 17:33:09) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO.GPIO as GPIO
>>> GPIO.setup("GPIO3_15", GPIO.OUT)
>>> GPIO.setup("GPIO3_14", GPIO.OUT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SystemError: error return without exception set
`
I use
BeagleBoard.org Debian Image 2018-03-09
4.4.120-bone22
Adafruit-BBIO (1.0.10)

pdp7, 2018-03-23:
@filipkotoucek the example you pasted fails with `GPIO.setup("GPIO3_14", GPIO.OUT)` as `GPIO3_14` does not appear in `source/common.c`.

What pin were you trying to represent with that?

pdp7, 2018-03-23:
@ThatWileyGuy for a test, I added a line for `GPIO0_22` in `source/common.c`:
```diff
diff --git a/source/common.c b/source/common.c
index e8fbbec..20138e6 100644
--- a/source/common.c
+++ b/source/common.c
@@ -94,6 +94,7 @@ pins_t table[] = {
   { "GPIO0_27", "P8_17", 27, -1, -1},
   { "GPIO2_1", "P8_18", 65, -1, -1},
   { "EHRPWM2A", "P8_19", 22, 4, -1},
+  { "GPIO0_22", "P8_19", 22, 4, -1},
   { "GPIO1_31", "P8_20", 63, -1, -1},
   { "GPIO1_30", "P8_21", 62, -1, -1},
   { "GPIO1_5", "P8_22", 37, -1, -1},
```
I ran this test program:
```python
import Adafruit_BBIO.GPIO as GPIO

GPIO.setup("P8_19", GPIO.OUT)
GPIO.setup("EHRPWM2A", GPIO.OUT)
GPIO.setup("GPIO0_22", GPIO.OUT)
```
and all three calls to `setup()` completed OK:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ sudo python2 ~/test_239.py
DEBUG: py_gpio.c: channel=P8_19
DEBUG: get_gpio_number(): key=P8_19
DEBUG: lookup_gpio_by_key(): return BBIO_OK
DEBUG: py_gpio.c: channel=EHRPWM2A
DEBUG: get_gpio_number(): key=EHRPWM2A
DEBUG: lookup_gpio_by_key(): gpio=22
DEBUG: lookup_gpio_by_key(): return BBIO_OK
DEBUG: py_gpio.c: channel=GPIO0_22
DEBUG: get_gpio_number(): key=GPIO0_22
DEBUG: lookup_gpio_by_key(): gpio=22
DEBUG: lookup_gpio_by_key(): return BBIO_OK
```

pdp7, 2018-03-23:
@ThatWileyGuy I've created [branch pr239](https://github.com/adafruit/adafruit-beaglebone-io-python/tree/pr239) if you would like to try out this change.

If no issues arise, then this approach could be used to add GPIO_x names for the other PWM pins.

filipkotoucek, 2018-03-23:
@pdp7 according to Beaglebone pinout http://exploringbeaglebone.com/wp-content/uploads/resources/BBBP9Header.pdf gpio3[14] sits on P9_31. Ou I tried to setup SPI1_SCLK and it worked, so I assume that is correct. But why why the macros are inconsistent? My firmware uses the numbers in GPIOx_x to calculate some memory offsets so I need to use these macros instead of SPI1_SCLK and similar.

pdp7, 2018-03-28:
@filipkotoucek The pin mapping in [`table` in source/common.c](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L73) shows the names that are currently supported.  It is based on [bonescript/bone.js](https://raw.github.com/jadonk/bonescript/master/node_modules/bonescript/bone.js).

pdp7, 2018-07-20:
@ThatWileyGuy is this still an issue? thanks

closed 2018-08-16 by pdp7