adafruit/adafruit-beaglebone-io-python #240: ADC.setup() Prevents Beaglebone Black Wireless Program from Running at Autostart

Tomasm64, 2018-03-12:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Beaglebone Black with Debian image 2017-07-01

- Python version (run `python -version` or `python3 -version`):  python 2.7.9

- Error message you are receiving, including any Python exception traces: N/A

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

I use the following command to run a python program at startup:
`crontab -e`

On the bottom of the file, I write the following:

`@reboot python filename.py &`

When I try to run the following code automatically, it doesn't run.
`#Headers`
`import Adafruit_BBIO.GPIO as GPIO`
`import Adafruit_BBIO.ADC as ADC`

`ADC.setup()`
`GPIO.setup(PINOUT,GPIO.OUT)`

`while True:`
`	GPIO.output(PINOUT,GPIO.HIGH)`

However, when I disable the following line, it does run.

`ADC.setup()`

I need the program to be able to read analog values in the future. How can I fix this?

pdp7, 2018-03-15:
I'm at a conference this week but I will attempt to reproduce this issue once I return home.  Thanks.

silver2row, 2018-03-16:
Hello,

This may help. I found this software on Adafruit's wesbite:

    import Adafruit_BBIO.ADC as ADC
    import time

    sensor_pin = 'P9_40'

    ADC.setup()

    while True:
        reading = ADC.read(sensor_pin)
        millivolts = reading * 1800  # 1.8V reference = 1800 mV
        temp_c = (millivolts - 500) / 10
        temp_f = (temp_c * 9/5) + 32
        print('mv=%d C=%d F=%d' % (millivolts, temp_c, temp_f))
        time.sleep(1)

Also...if you get bored, you can try to use setting up a service:

    https://learn.adafruit.com/running-programs-automatically-on-your-tiny-computer?view=all

Until the other fellow gets back, this may help.

Seth

pdp7, 2018-03-23:
@Tomasm64 Please make sure you are using the latest version:
```
sudo pip install --upgrade Adafruit_BBIO
```
I agree with the above comment that the correct solution to run a program after boot is with a systemd service.  I would suggest looking this Adafruit tutorial on how to create a until file: **[systemd: Writing and Enabling a Service](https://learn.adafruit.com/running-programs-automatically-on-your-tiny-computer/systemd-writing-and-enabling-a-service)**

If you do prefer to stick with cron, then please install strace to trace what is happening:
```
sudo apt-get install strace
```

Then update your cron entry to invoke strace:
```
@reboot strace -f -o /tmp/strace-python.txt python filename.py &
```

Create a GitHub gist with the contest of `/tmp/strace-python.txt` and paste the link here.

pdp7, 2018-03-28:
@Tomasm64 is this still an issue for you?

pdp7, 2018-04-04:
@Tomasm64 is this still an issue?

pdp7, 2018-04-07:
@Tomasm64 is it ok to close this issue?

closed 2018-04-08 by pdp7