adafruit/adafruit-beaglebone-io-python #241: Getting position of rotary encoder by Adafruit_BBIO.Encoder

DarioushK1, 2018-03-13:
Hi, I'm kinda new to beaglebones. I've used Adafruit_BBIO.Encoder to read the current position of my rotary encoder. However, the output is always 0 or -1. p8.11 and p8.12 are the pins that I've used.
Here is my code:

from Adafruit_BBIO.Encoder import RotaryEncoder, eQEP2
# Instantiate the class to access channel eQEP2, and initialize
# that channel
myEncoder = RotaryEncoder(eQEP2)
# Get the current position
cur_position = myEncoder.position
while 1:
    print cur_position

I will appreciate if anyone can help me.

Thank you.
Darioush

pdp7, 2018-03-15:
I'm at a conference this week but I will attempt to reproduce this issue once I return home.  Thanks.

DarioushK1, 2018-03-15:
I solved it. Thanks.

closed 2018-03-15 by DarioushK1

Shadi4772, 2018-07-19:
Hello, I'm trying to do the same and I always get 0. can you help me with your solution? Thank you in advance