adafruit/adafruit-beaglebone-io-python #243: UART setup not working for pocket beagle pins

tesorrells, 2018-03-21:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  pocket beagle / Linux beaglebone 4.4.113-ti-r145

- Python version (run `python -version` or `python3 -version`):  python 2.7.9

- Error message you are receiving, including any Python exception traces:  py_setup_uart(UART2): set_pin_mode() failed for pin=P9_22Traceback (most recent call last):
  File "pocket_serial.py", line 4, in <module>
    UART.setup("UART2")
ValueError: Set pin mode failed for uart channel.

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

config-pin P1.08 uart
config-pin P1.10 uart

Then ran the following python script:

import Adafruit_BBIO.UART as UART
import serial

UART.setup("UART2")
ser = serial.Serial(port = "/dev/tty02", baudrate=9600)
ser.close()
ser.open()
if ser.isOpen():
        print "Serial is open"
        ser.read()
ser.close()

pdp7, 2018-03-23:
@tesorrells It appears that there will need to be some conditional logic added.

From `source/common.c`:
```
uart_t uart_table[] = {
  { "UART1", "/dev/ttyO1", "ADAFRUIT-UART1", "P9_26", "P9_24"},
  { "UART2", "/dev/ttyO2", "ADAFRUIT-UART2", "P9_22", "P9_21"},
  { "UART3", "/dev/ttyO3", "ADAFRUIT-UART3", "P9_42", ""},
  { "UART4", "/dev/ttyO4", "ADAFRUIT-UART4", "P9_11", "P9_13"},
  { "UART5", "/dev/ttyO5", "ADAFRUIT-UART5", "P8_38", "P8_37"},
  { NULL, NULL, 0, 0, 0 }
};
```
I will work on a solution and followup here when I have something to test.

pdp7, 2018-08-27:
From common.c:
```
BBIO_err lookup_uart_by_name(const char *input_name, char *dt)
{
    uart_t *p;
    for (p = uart_table; p->name != NULL; ++p) {
        if (strcmp(p->name, input_name) == 0) {
            strncpy(dt, p->dt, FILENAME_BUFFER_SIZE);
            dt[FILENAME_BUFFER_SIZE - 1] = '\0';
            return BBIO_OK;
        }
    }
    return BBIO_INVARG;
}
```

pdp7, 2018-08-27:
From `source/py_uart.c`:
```
#ifdef BBBVERSION41
    uart_t *p;
    for (p = uart_table; p->name != NULL; ++p) {
        if (strcmp(p->name, channel) == 0) {
            err = set_pin_mode(p->rx, "uart");
            //Check if set_pin_mode() returned no error
            if (err != BBIO_OK) {
                fprintf(stderr, "py_setup_uart(%s): set_pin_mode() failed for pin=%s", channel, p->rx);
                PyErr_SetString(PyExc_ValueError, "Set pin mode failed for uart channel.");
                return NULL;
            }

            err = set_pin_mode(p->tx, "uart");
            //Check if set_pin_mode() returned no error
            if (err != BBIO_OK) {
                fprintf(stderr, "py_setup_uart(%s): set_pin_mode() failed for pin=%s", channel, p->tx);
                PyErr_SetString(PyExc_ValueError, "Set pin mode failed for uart channel.");
                return NULL;
            }
        }
    }
#endif
```

pdp7, 2018-08-27:
From `source/py_uart.c`:
```
typedef struct uart_t {
    const char *name;
    const char *path;
    const char *dt;
    const char *rx;
    const char *tx;
} uart_t;
```

pdp7, 2018-08-28:
```
	P1_08_INFO="spi0_sclk default gpio0_2 gpio0_2 gpio0_2 gpio0_2 spi0_sclk uart2_rxd i2c2_sda ehrpwm0a pru_uart"
	P1_10_INFO="spi0_d0 default gpio0_3 gpio0_3 gpio0_3 gpio0_3 spi0_d0 uart2_txd i2c2_scl ehrpwm0b pru_uart"
	P1_30_INFO="uart0_txd default gpio1_11 gpio1_11 gpio1_11 gpio1_11 spi1_cs1 uart0_txd dcan0_rx i2c2_scl pru1_out15 pru1_in15"
	P1_32_INFO="uart0_rxd default gpio1_10 gpio1_10 gpio1_10 gpio1_10 spi1_cs0 uart0_rxd dcan0_tx i2c2_sda pru1_out14 pru1_in14"
	P2_09_INFO="i2c1_scl default gpio0_15 gpio0_15 gpio0_15 gpio0_15 uart1_txd dcan1_rx i2c1_scl pru_uart pru0_in16"
	P2_11_INFO="i2c1_sda default gpio0_14 gpio0_14 gpio0_14 gpio0_14 uart1_rxd dcan1_tx i2c1_sda pru_uart pru1_in16"
	P2_29_INFO="spi1_sclk default gpio0_7 gpio0_7 gpio0_7 gpio0_7 spi1_cs1 spi1_sclk uart3_txd ecap0_in_pwm0_out pru_ecap"
```

pdp7, 2018-08-30:
```
diff --git a/source/common.c b/source/common.c
index e8fbbec..0661340 100644
--- a/source/common.c
+++ b/source/common.c
@@ -263,12 +263,29 @@ pins_t table[] = {
   { NULL, NULL, 0, 0, 0 }
 };
 
+/*
+       P1_30 uart0_txd
+       P1_32 uart0_rxd
+
+       P2_11 uart1_rxd
+       P2_09 uart1_txd
+
+       P1_08 uart2_rxd
+       P1_10 uart2_txd
+
+       P2_29 uart3_txd
+*/
+
 uart_t uart_table[] = {
   { "UART1", "/dev/ttyO1", "ADAFRUIT-UART1", "P9_26", "P9_24"},
   { "UART2", "/dev/ttyO2", "ADAFRUIT-UART2", "P9_22", "P9_21"},
   { "UART3", "/dev/ttyO3", "ADAFRUIT-UART3", "P9_42", ""},
   { "UART4", "/dev/ttyO4", "ADAFRUIT-UART4", "P9_11", "P9_13"},
   { "UART5", "/dev/ttyO5", "ADAFRUIT-UART5", "P8_38", "P8_37"},
+  { "PB-UART0", "/dev/ttyO0", "ADAFRUIT-UART0", "P1_30", "P1_32"},
+  { "PB-UART1", "/dev/ttyO1", "ADAFRUIT-UART1", "P2_11", "P2_09"},
+  { "PB-UART2", "/dev/ttyO2", "ADAFRUIT-UART2", "P1_08", "P1_10"},
+  { "PB-UART3", "/dev/ttyO3", "ADAFRUIT-UART3", "P2_29", ""},
   { NULL, NULL, 0, 0, 0 }
 };
 
@@ -354,7 +371,13 @@ int lookup_ain_by_name(const char *name)
 BBIO_err lookup_uart_by_name(const char *input_name, char *dt)
 {
     uart_t *p;
+
+    fprintf(stderr, "DEBUG: lookup_uart_by_name(): input_name=%s\n", input_name);
+    fprintf(stderr, "DEBUG: lookup_uart_by_name(): dt=%s\n", dt);
+
     for (p = uart_table; p->name != NULL; ++p) {
+        fprintf(stderr, "DEBUG: lookup_uart_by_name(): p->name=%s\n", p->name);
+        fprintf(stderr, "DEBUG: lookup_uart_by_name(): p->dt=%s\n", p->dt);
         if (strcmp(p->name, input_name) == 0) {
             strncpy(dt, p->dt, FILENAME_BUFFER_SIZE);
             dt[FILENAME_BUFFER_SIZE - 1] = '\0';
diff --git a/source/py_uart.c b/source/py_uart.c
index c70d327..b92d421 100644
--- a/source/py_uart.c
+++ b/source/py_uart.c
@@ -64,7 +64,12 @@ static PyObject *py_setup_uart(__attribute__ ((unused)) PyObject *self, PyObject
 
 #ifdef BBBVERSION41
     uart_t *p;
+    if(pocketbeagle()) {
+        fprintf(stderr, "DEBUG: py_uart.c: py_setup_uart(): pocketbeagle() is TRUE\n");
+    }
     for (p = uart_table; p->name != NULL; ++p) {
+        fprintf(stderr, "DEBUG: py_uart.c: py_setup_uart(): p->name=%s\n", p->name);
+        fprintf(stderr, "DEBUG: py_uart.c: py_setup_uart(): channel=%s\n", channel);
         if (strcmp(p->name, channel) == 0) {
             err = set_pin_mode(p->rx, "uart");
             //Check if set_pin_mode() returned no error
debian@beaglebone:~/adafruit-beaglebone-io-python$ git  diff > p.diff
debian@beaglebone:~/adafruit-beaglebone-io-python$ cat p.diff 
diff --git a/source/common.c b/source/common.c
index e8fbbec..0661340 100644
--- a/source/common.c
+++ b/source/common.c
@@ -263,12 +263,29 @@ pins_t table[] = {
   { NULL, NULL, 0, 0, 0 }
 };
 
+/*
+	P1_30 uart0_txd
+	P1_32 uart0_rxd
+
+	P2_11 uart1_rxd
+	P2_09 uart1_txd
+
+ 	P1_08 uart2_rxd
+	P1_10 uart2_txd
+
+	P2_29 uart3_txd
+*/
+
 uart_t uart_table[] = {
   { "UART1", "/dev/ttyO1", "ADAFRUIT-UART1", "P9_26", "P9_24"},
   { "UART2", "/dev/ttyO2", "ADAFRUIT-UART2", "P9_22", "P9_21"},
   { "UART3", "/dev/ttyO3", "ADAFRUIT-UART3", "P9_42", ""},
   { "UART4", "/dev/ttyO4", "ADAFRUIT-UART4", "P9_11", "P9_13"},
   { "UART5", "/dev/ttyO5", "ADAFRUIT-UART5", "P8_38", "P8_37"},
+  { "PB-UART0", "/dev/ttyO0", "ADAFRUIT-UART0", "P1_30", "P1_32"},
+  { "PB-UART1", "/dev/ttyO1", "ADAFRUIT-UART1", "P2_11", "P2_09"},
+  { "PB-UART2", "/dev/ttyO2", "ADAFRUIT-UART2", "P1_08", "P1_10"},
+  { "PB-UART3", "/dev/ttyO3", "ADAFRUIT-UART3", "P2_29", ""},
   { NULL, NULL, 0, 0, 0 }
 };
 
@@ -354,7 +371,13 @@ int lookup_ain_by_name(const char *name)
 BBIO_err lookup_uart_by_name(const char *input_name, char *dt)
 {
     uart_t *p;
+
+    fprintf(stderr, "DEBUG: lookup_uart_by_name(): input_name=%s\n", input_name);
+    fprintf(stderr, "DEBUG: lookup_uart_by_name(): dt=%s\n", dt);
+
     for (p = uart_table; p->name != NULL; ++p) {
+        fprintf(stderr, "DEBUG: lookup_uart_by_name(): p->name=%s\n", p->name);
+        fprintf(stderr, "DEBUG: lookup_uart_by_name(): p->dt=%s\n", p->dt);
         if (strcmp(p->name, input_name) == 0) {
             strncpy(dt, p->dt, FILENAME_BUFFER_SIZE);
             dt[FILENAME_BUFFER_SIZE - 1] = '\0';
diff --git a/source/py_uart.c b/source/py_uart.c
index c70d327..b92d421 100644
--- a/source/py_uart.c
+++ b/source/py_uart.c
@@ -64,7 +64,12 @@ static PyObject *py_setup_uart(__attribute__ ((unused)) PyObject *self, PyObject
 
 #ifdef BBBVERSION41
     uart_t *p;
+    if(pocketbeagle()) {
+        fprintf(stderr, "DEBUG: py_uart.c: py_setup_uart(): pocketbeagle() is TRUE\n");
+    }
     for (p = uart_table; p->name != NULL; ++p) {
+        fprintf(stderr, "DEBUG: py_uart.c: py_setup_uart(): p->name=%s\n", p->name);
+        fprintf(stderr, "DEBUG: py_uart.c: py_setup_uart(): channel=%s\n", channel);
         if (strcmp(p->name, channel) == 0) {
             err = set_pin_mode(p->rx, "uart");
             //Check if set_pin_mode() returned no error
```

pdp7, 2018-08-30:
```python
debian@beaglebone:~/adafruit-beaglebone-io-python$ cat ~/uart.py 
import Adafruit_BBIO.UART as UART
import serial

# Add UART entries for the PocketBeagle (issue #243)
#    
# PocketBeagle pins with UART pin mode:
#    
#      P1_30 uart0_txd
#      P1_32 uart0_rxd
#      P2_11 uart1_rxd
#      P2_09 uart1_txd
#      P1_08 uart2_rxd
#      P1_10 uart2_txd
#    
#    Available UART names on PocketBeagle:
#    
#     PB-UART0: /dev/ttyO0, Rx: P1_30, Tx: P1_32
#     PB-UART1: /dev/ttyO1, Rx: P2_11, Tx: P2_09
#     PB-UART2: /dev/ttyO2, Rx: P1_08, Tx: P1_10

value = open('/proc/device-tree/model').read()
if(value.startswith("TI AM335x PocketBeagle")):
  print("detected board is a PocketBeagle")
  uarts = {
    'PB-UART0': '/dev/ttyO0',
    'PB-UART1': '/dev/ttyO1',
    'PB-UART2': '/dev/ttyO2',
  }
else:
  print("detected board is NOT a PocketBeagle")
  uarts = {
    'UART1': '/dev/ttyO1',
    'UART2': '/dev/ttyO2',
    'UART3': '/dev/ttyO3',
    'UART4': '/dev/ttyO4',
    'UART5': '/dev/ttyO5',
  }

for name, device in sorted(uarts.items()):
    print("name: {0}, device: {1}".format(name, device))
    UART.setup(name)
    ser = serial.Serial(port = device, baudrate=9600)
    ser.close()
    ser.open()
    if ser.isOpen():
        print("name: {0}, device: {1}: Serial is open".format(name, device))
        ser.write("Hello World!")
    ser.close()
```

pdp7, 2018-08-31:
Testing from BeagleBone Black:
```
debian@beaglebone:~/adafruit-beaglebone-io-python$ sudo python ~/uart.py 
[sudo] password for debian: 
Sorry, try again.
[sudo] password for debian: 
detected board is NOT a PocketBeagle
name: UART1, device: /dev/ttyO1
name: UART1, device: /dev/ttyO1: Serial is open
name: UART2, device: /dev/ttyO2
name: UART2, device: /dev/ttyO2: Serial is open
```

pdp7, 2018-08-31:
@tesorrells I have pushed a fix to the master branch: https://github.com/adafruit/adafruit-beaglebone-io-python/commit/c749402003d9514ac23ed4885dc9806a68924fc3

Please install from the master branch [(using the instructions in the README)](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/README.md#installation-on-debian) and please test the UARTs again on the PocketBeagle.  Thanks!

pdp7, 2018-09-01:
@tesorrells please reopen if you are still having this issue