adafruit/adafruit-beaglebone-io-python #246: BeagleboneBlack Wireless with Ubuntu 16.04 : PWM produces "RuntimeError: Problem with the cape manager"

NickGhi, 2018-03-26:
I have a beagleboneblack wireless with Ubuntu xenial 16.04. I am trying to use the Adafruit_BBIO library and I am following these instructions : https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/installation-on-ubuntu .

Following the tutorials provided in adafruit's webpage, I set up some GPIO ports successfully. However, when I try to set up a PWM port, the following runtime error occurs:
https://www.dropbox.com/s/nw8dbwl04rjgums/PMW_cape_manager_runtimeError.png?dl=0

I have found similar threads here: https://github.com/adafruit/adafruit-beaglebone-io-python/issues/136, https://github.com/adafruit/adafruit-beaglebone-io-python/issues/125, https://github.com/adafruit/adafruit-beaglebone-io-python/issues/230. I have tried most of the solutions suggested in these forums, without any luck so far.

pdp7, 2018-03-28:
I would suggest running the [latest BeagleBoard.org Debian image](https://beagleboard.org/latest-images) which @RobertCNelson maintains.

If you do need to stay on Ubuntu 16.04, then please paste the output for the following commands:
```
cat /etc/dogtag 

cat /etc/debian_version 

cat /boot/uEnv.txt 

sudo /opt/scripts/tools/version.sh

sudo python -c "import Adafruit_BBIO; print Adafruit_BBIO"
```

Please paste the code for the Python script that you trying to run.

It would also be good to see what system calls the Python script makes.  Install strace and run python with strace:
```
sudo apt-get install strace
sudo strace -f -o /tmp/strace-python.txt python filename.py
```
Paste the contents of `/tmp/strace-python.txt` into a GitHub gist and post the link here.

pdp7, 2018-04-04:
@NickGhi is this still an issue for you?

NickGhi, 2018-04-07:
@pdp7 I apologize for not replying all this time. The problem I described above was solved by flashing the latest ubuntu armhf console image (i.e. the 2018-03-09 version), instead of the older 2018-02-09 version. I did not investigate any further what caused the problem with the older version. Thank you for your immediate response! I believe you can close this issue.