adafruit/adafruit-beaglebone-io-python #247: "ValueError: Set gpio mode failed, missing file or invalid permissions." when using GPIO.IN but only when using a cape.

Irvin08, 2018-03-29:
Hello, I have seen similar issues but the solutions to the other issues have not fixed my problem. I am running Debian 9.3 2018-01-28 4GB SD LXQT. I am able to run the following code without a cape attached:

import Adafruit_BBIO.GPIO as GPIO
import time
 
GPIO.setup("P8_10", GPIO.IN)
 
old_switch_state = 0
 
while True:
    new_switch_state = GPIO.input("P8_10")
    if new_switch_state == 1 and old_switch_state == 0 :
        print('Do not press this button again!')
        time.sleep(0.1)
    old_switch_state = new_switch_state

However, when I attach a TI DLP2000, I receive the error "ValueError: Set gpio mode failed, missing file or invalid permissions." The error does not occur if I use GPIO 10 as an output.

This is the result from running version.sh:

debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLT00C02917BBBK1187]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]
kernel:[4.9.78-ti-r94]
nodejs:[v6.14.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180322.0-0rcnee0~stretch+20180322]
pkg:[bb-wl18xx-firmware]:[1.20180328-0rcnee2~stretch+20180328]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    1.405757] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END

RobertCNelson, 2018-03-29:
@Irvin08 correct, once you plug in a cape, all the other pinmux'ing disappears..

This is being tracked here:

https://github.com/beagleboard/bb.org-overlays/milestone/1

Commits with: "universal rewrite: " are capes completed..

https://github.com/beagleboard/bb.org-overlays/commits/master

Regards,

Irvin08, 2018-03-29:
I see, thank you for your reply, this is my first time using a BeagleBone and I am a bit unfamiliar but I see there is a file "DLPDLCR2000-00A0.dts". I can see from the DLP2000 documentation that the used ports are listed here so would I be able to simply add GPIO 10 for example to this file and use it in the example python program?

pdp7, 2018-03-30:
@Irvin08 please paste the output of this command:
```
perl /opt/scripts/device/bone/show-pins.pl | grep -i p8.10
```
@RobertCNelson does an overlay needs to be loaded to set pinmux for P8.10?

Irvin08, 2018-03-30:
@pdp7 The output is
`debian@beaglebone:~$ perl /opt/scripts/device/bone/show-pins.pl | grep -i p8.10`
`P8.10                             38  U6 fast rx  up  7 gpio 2.04`

pdp7, 2018-04-04:
@RobertCNelson do you think an universal overlay is needed to mux P8_10 for GPIO?

RobertCNelson, 2018-04-04:
@pdp7 this needs a working v4.16.0 in the apt repo and then i need to patch bb.org-overlays..

Everything should be ready by tomorrow..

Regards,

RobertCNelson, 2018-04-04:
Fixed in git:

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[70edebd65fe6ea7de16ea8efe3c48b2a4062034a]
eeprom:[A335BNLT0A5A1513BBBK3949]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-03-28]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-g254339602c]:[location: dd MBR]
kernel:[4.16.0-bone7]
nodejs:[v6.14.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180322.0-0rcnee0~stretch+20180322]
pkg:[bb-wl18xx-firmware]:[1.20180328-0rcnee2~stretch+20180328]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait uboot_detected_capes=DLPDLCR2000, coherent_pool=1M net.ifnames=0]
dmesg | grep pinctrl-single
[    0.920278] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.932961] gpio-of-helper ocp:cape-universal: ready
END
debian@beaglebone:~$ uname -r
4.16.0-bone7
debian@beaglebone:~$ ls /sys/devices/platform/ocp/ | grep pinmux
ocp:P8_07_pinmux
ocp:P8_08_pinmux
ocp:P8_09_pinmux
ocp:P8_10_pinmux
ocp:P8_18_pinmux
ocp:P9_11_pinmux
ocp:P9_12_pinmux
ocp:P9_13_pinmux
ocp:P9_14_pinmux
ocp:P9_16_pinmux
ocp:P9_17_pinmux
ocp:P9_18_pinmux
ocp:P9_19_pinmux
ocp:P9_20_pinmux
ocp:P9_21_pinmux
ocp:P9_22_pinmux
ocp:P9_23_pinmux
ocp:P9_24_pinmux
ocp:P9_25_pinmux
ocp:P9_26_pinmux
ocp:P9_27_pinmux
ocp:P9_28_pinmux
ocp:P9_29_pinmux
ocp:P9_30_pinmux
ocp:P9_31_pinmux
ocp:P9_41_pinmux
ocp:P9_42_pinmux
ocp:P9_91_pinmux
ocp:P9_92_pinmux
```

RobertCNelson, 2018-04-04:
btw, wish we know how to "reinit" the DLPDLCR2000..  You always have to power cycle the cape, "sudo reboot" never reinit's the display..

Regards,

Irvin08, 2018-04-07:
Hi @pdp7 , thanks for the continued help. I tried updating with the new version but running the commands "sudo apt update ; sudo apt install bb-cape-overlays" from the github page said that the installed version was the newest already and when I opened the .dts file for the DLP2000 it was still the old one. I tried to manually change it and then ran the install.sh in bb.org-overlays which said the new version was successfully installed. After rebooting I am still receiving the same error message but I am not sure if I updated the file properly. I tried running the commands @RobertCNelson posted and this is what I received:

debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLT00C02917BBBK1187]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]
kernel:[4.9.78-ti-r94]
nodejs:[v6.14.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180405.0-0rcnee0~stretch+20180405]
pkg:[bb-wl18xx-firmware]:[1.20180328-0rcnee2~stretch+20180328]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    1.410131] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END

debian@beaglebone:~$ uname -r
4.9.78-ti-r94

debian@beaglebone:~$ ls /sys/devices/platform/ocp/ | grep pinmux
ocp:P9_19_pinmux
ocp:P9_20_pinmux

RobertCNelson, 2018-04-08:
@Irvin08 needs u-boot v2018.03 and v4.16.x:

```
debian@test-bbb-2:~$ sudo /opt/scripts/tools/update_kernel.sh --bone-channel --testing
```
and

```
debian@test-bbb-2:~$ sudo /opt/scripts/tools/developers/update_bootloader.sh 
```

then everything should be good..

Regards,

Irvin08, 2018-04-08:
I put the first command and this was the output:
info: checking archive
2018-04-08 18:31:37 URL:https://rcn-ee.com/repos/latest/stretch-armhf/LATEST-omap-psp [193/193] -> "LATEST-omap-psp" [1]
-----------------------------
Kernel Options:
ABI:1 LTS41 4.1.38-bone24
ABI:1 LTS44 4.4.126-bone22
ABI:1 LTS49 4.9.92-bone10
ABI:1 LTS414 4.14.32-bone14
ABI:1 STABLE 3.8.13-bone86
ABI:1 TESTING 4.16.0-bone8
ABI:1 EXPERIMENTAL 4.16.0-bone7
-----------------------------
Kernel version options:
-----------------------------
LTS44: --lts-4_4
LTS49: --lts-4_9
LTS414: --lts-4_14
STABLE: --stable
TESTING: --testing
-----------------------------
info: you are running: [4.9.78-ti-r94], latest is: [4.16.0-bone8] updating...
Ign:1 http://deb.debian.org/debian stretch InRelease
Get:2 http://deb.debian.org/debian stretch-updates InRelease [91.0 kB]
Hit:3 http://repos.rcn-ee.com/debian stretch InRelease
Get:4 http://deb.debian.org/debian-security stretch/updates InRelease [63.0 kB]
Hit:5 http://deb.debian.org/debian stretch Release
Fetched 154 kB in 9s (16.5 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Checking dpkg...
Checking apt-cache...
debug: installing: [linux-image-4.16.0-bone8]
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  enchant fonts-dejavu fonts-dejavu-extra gir1.2-atk-1.0 gir1.2-freedesktop
  gir1.2-gdkpixbuf-2.0 gir1.2-glib-2.0 gir1.2-gtk-3.0 gir1.2-pango-1.0
  gir1.2-peas-1.0 gnome-desktop3-data gnome-user-guide
  gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-pulseaudio
  gstreamer1.0-x gthumb-data hunspell-en-us libaa1 libavc1394-0 libcaca0
  libcdparanoia0 libclutter-1.0-0 libclutter-1.0-common libclutter-gtk-1.0-0
  libcogl-common libcogl-pango20 libcogl-path20 libcogl20 libdv4
  libenchant1c2a libexempi3 libexiv2-14 libgirepository-1.0-1
  libgnome-desktop-3-12 libgstreamer-plugins-base1.0-0 libgstreamer1.0-0
  libharfbuzz-icu0 libhunspell-1.4-0 libhyphen0 libiec61883-0 libjack-jackd2-0
  libjavascriptcoregtk-4.0-18 libnotify4 liborc-0.4-0 libpangoxft-1.0-0
  libpeas-1.0-0 libpeas-common libraw1394-11 libraw15 libshout3 libtag1v5
  libtag1v5-vanilla libvisual-0.4-0 libwebkit2gtk-4.0-37 libyelp0 man-db
  notification-daemon yelp yelp-xsl
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  linux-firmware-image-4.16.0-bone8
The following NEW packages will be installed:
  linux-image-4.16.0-bone8
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 28.0 MB of archives.
After this operation, 36.2 MB of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian stretch/main armhf linux-image-4.16.0-bone8 armhf 1stretch [28.0 MB]
Fetched 28.0 MB in 7s (3,853 kB/s)
Selecting previously unselected package linux-image-4.16.0-bone8.
(Reading database ... 90983 files and directories currently installed.)
Preparing to unpack .../linux-image-4.16.0-bone8_1stretch_armhf.deb ...
Unpacking linux-image-4.16.0-bone8 (1stretch) ...
Setting up linux-image-4.16.0-bone8 (1stretch) ...
update-initramfs: Generating /boot/initrd.img-4.16.0-bone8
zz-uenv_txt: Updating /boot/uEnv.txt [uname_r=4.16.0-bone8]
[... 50 lines cut ...]
432064 bytes (432 kB, 422 KiB) copied, 0.0514906 s, 8.4 MB/s
-----------------------------
Bootloader Updated
Bootloader Recovery Complete
 
Lastly I ran
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLT00C02917BBBK1187]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]
kernel:[4.16.0-bone8]
nodejs:[v6.14.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180405.0-0rcnee0~stretch+20180405]
pkg:[bb-wl18xx-firmware]:[1.20180328-0rcnee2~stretch+20180328]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    0.878405] pinctrl-single 44e10800.pinmux: 142 pins, size 568
END

However after rebooting I have been unable to connect to the beaglebone.

pdp7, 2018-04-08:
@Irvin08 
> However after rebooting I have been unable to connect to the beaglebone.
Sorry to hear that.  Might you have an USB to serial cable so that you could view the bootloader output?

@RobertCNelson any ideas?

Irvin08, 2018-04-08:
@pdp7 I do not have a USB to serial available, I was unable to get it connect after power cycling so I just reflashed but I got the same issue after trying again. I have reflashed once more and currently have this configuration again:

git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLT00C02917BBBK1187]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]
kernel:[4.9.78-ti-r94]
nodejs:[v6.14.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180405.0-0rcnee0~stretch+20180405]
pkg:[bb-wl18xx-firmware]:[1.20180328-0rcnee2~stretch+20180328]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    1.409101] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END

RobertCNelson, 2018-04-09:
@Irvin08 

ah you are starting with a flashed eMMC, please follow these steps then:

Remove the cape. ;)

Step 1: Update scripts directory

debian@beaglebone:~$ cd /opt/scripts/tools/
debian@beaglebone:/opt/scripts/tools$ git pull

Step 2: Check version:

```
debian@beaglebone:/opt/scripts/tools$ sudo ./version.sh
git:/opt/scripts/:[ee77aa2aa361474be4b0dde88f4d93c766691565]
eeprom:[A335BNLT000C1727BBBG1071]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]:[location: dd MBR]
kernel:[4.9.78-ti-r94]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.428761] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.435906] gpio-of-helper ocp:cape-universal: ready
END
```

Step 3: Upgrade bb-cape-overlays

```
debian@beaglebone:/opt/scripts/tools$ sudo apt update
debian@beaglebone:/opt/scripts/tools$ sudo apt install --only-upgrade bb-cape-overlays
```

Step 4: Update Bootloader

```
debian@beaglebone:/opt/scripts/tools$ sudo ./developers/update_bootloader.sh
```

Step 5: Update Kernel

```
debian@beaglebone:/opt/scripts/tools$ sudo ./update_kernel.sh --bone-channel --testing
```

Step 6: reboot and verify

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[ee77aa2aa361474be4b0dde88f4d93c766691565]
eeprom:[A335BNLT000C1727BBBG1071]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.16.1-bone8]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180408.0-0rcnee0~stretch+20180409]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.912502] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.925156] gpio-of-helper ocp:cape-universal: ready
END
```

Step 7: Shutdown, reinsert cape, and power up..

```
debian@beaglebone:~$ uname -r
4.16.1-bone8
debian@beaglebone:~$ cat /proc/cmdline
console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait uboot_detected_capes=DLPDLCR2000, coherent_pool=1M net.ifnames=0 quiet
debian@beaglebone:~$ ls /sys/devices/platform/ocp/ | grep pinmux
ocp:P8_07_pinmux
ocp:P8_08_pinmux
ocp:P8_09_pinmux
ocp:P8_10_pinmux
ocp:P8_18_pinmux
ocp:P9_11_pinmux
ocp:P9_12_pinmux
ocp:P9_13_pinmux
ocp:P9_14_pinmux
ocp:P9_16_pinmux
ocp:P9_17_pinmux
ocp:P9_18_pinmux
ocp:P9_19_pinmux
ocp:P9_20_pinmux
ocp:P9_21_pinmux
ocp:P9_22_pinmux
ocp:P9_23_pinmux
ocp:P9_24_pinmux
ocp:P9_25_pinmux
ocp:P9_26_pinmux
ocp:P9_27_pinmux
ocp:P9_28_pinmux
ocp:P9_29_pinmux
ocp:P9_30_pinmux
ocp:P9_31_pinmux
ocp:P9_41_pinmux
ocp:P9_42_pinmux
ocp:P9_91_pinmux
ocp:P9_92_pinmux
```

Regards,

RobertCNelson, 2018-04-09:
Ps remember, you need to use two "5vDC" jacks for power, both the cape and bone needs a 5volt dc source..

Regards,

Irvin08, 2018-04-09:
@RobertCNelson I am following along with the provided steps and when I reboot after updating the kernel all 4 leds light up and stay on, they have been like this for a couple of minutes. Is this supposed to happen?

RobertCNelson, 2018-04-09:
@Irvin08 nope, you should see something, can you plugin a serial cable so we can what u-boot is doing?

https://elinux.org/Beagleboard:BeagleBone_Black_Serial

Regards,

Irvin08, 2018-04-09:
@RobertCNelson I was in the process of reflashing when I saw your last suggestion so I tried just going through the process again and I never connected the serial cable. I followed the steps again except I used the "shutdown now" command instead of "reboot" after updating the kernel and after that the board turned on like normal, though I'm not sure if that was the issue or if I made a mistake the first time. I was then able to finish the steps provided and I am now able to set the GPIO ports as inputs and run the original code while the DLP2000 is connected. Thank you very much for all your help.

pdp7, 2018-04-10:
>  was then able to finish the steps provided and I am now able to set the GPIO ports as inputs and run the original code while the DLP2000 is connected. Thank you very much for all your help.
@Irvin08 is it ok to close this issue now?

@RobertCNelson thanks for helping!

Irvin08, 2018-04-10:
@pdp7 Yes, my issue is resolved.

closed 2018-04-10 by pdp7