adafruit/adafruit-beaglebone-io-python #248: "ValueError: Set gpio mode failed, missing file or invalid permissions." when using GPIO.IN --NO CAPE

cren14, 2018-04-01:
Hello,

I am having a similar issue to previous posters but all the recommendations list still don't work. I have worked through issue #206, #207, and #247. Trying to just run a simple python program to set GPIO.IN

import Adafruit_BBIO.GPIO as GPIO
tb="P9_15"
GPIO.setup(tb,GPIO.IN)

Eventually, I want to use a more complicated code for reading a flow meter but need to be able to set GPIO.IN first.

sudo /opt/scripts/tools/version.sh 

root@beaglebone:/home/debian/Desktop# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[70edebd65fe6ea7de16ea8efe3c48b2a4062034a]
eeprom:[A335BNLT00C02917BBBK27D1]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-g9aa111a004]:[location: dd MBR]
kernel:[4.14.31-ti-r41]
nodejs:[v6.14.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo]
pkg:[bb-cape-overlays]:[4.4.20180322.0-0rcnee0~stretch+20180322]
pkg:[bb-wl18xx-firmware]:[1.20180328-0rcnee2~stretch+20180328]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.945936] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.947011] gpio-of-helper ocp:cape-universal: ready
END

sudo apt list --upgradable
root@beaglebone:/home/debian/Desktop# sudo apt list --upgradable
Listing... Done
gpiod/unknown 1.0-1rcnee0~stretch+20180221 armhf [upgradable from: 0.3.2+git20171201-2rcnee2~stretch+20180104]
N: There is 1 additional version. Please use the '-a' switch to see it
root@beaglebone:/home/debian/Desktop#

sudo strace -e open python ./ft.py 2>&1 | grep /sys/
root@beaglebone:/home/debian/Desktop# sudo strace -e open python ./ft.py 2>&1 | grep /sys/
open("/sys/class/gpio/export", O_WRONLY) = 4
open("/sys/class/gpio/gpio48/direction", O_WRONLY) = 4
open("/sys/devices/platform/ocp/ocp:P9_15_pinmux/state", O_WRONLY|O_CREAT|O_TRUNC, 0666) = -1 ENOENT (No such file or directory)
open("/sys/class/gpio/unexport", O_WRONLY) = 4
root@beaglebone:/home/debian/Desktop#

cat /boot/uEnv.txt
root@beaglebone:/home/debian/Desktop# cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.31-ti-r41
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo
root@beaglebone:/home/debian/Desktop#

pdp7, 2018-04-07:
From the strace output, it looks like pinmux helper is not setup for P9.15:
```
open("/sys/devices/platform/ocp/ocp:P9_15_pinmux/state", O_WRONLY|O_CREAT|O_TRUNC, 0666) = -1
```
And from `/opt/scripts/tools/version.sh`, I see that `BB-W1-P9.12-00A0.dtbo` overlay is being loaded:
```
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo]
```
@cren14  *Could you try commenting that out in `/boot/uEnv.txt`?*

@RobertCNelson any other suggestions?

fivdi, 2018-04-07:
Removing `uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo` from `/boot/uEnv.txt` if it isn't needed should also resolve the issue as @pdp7 suggests.

pdp7, 2018-04-08:
@fivdi thanks for pointing out those solutions.

@cren14  In #247, @RobertCNelson recommends u-boot v2018.03 and kernel v4.16.x.  Please try this:
```
sudo /opt/scripts/tools/update_kernel.sh --bone-channel --testing
```
and
```
sudo /opt/scripts/tools/developers/update_bootloader.sh 
```
Then reboot and please test again if you able to use `P9_15`.

RobertCNelson, 2018-04-08:
@cren14 Oh and we need this one change for @pdp7 directions, for: BB-W1-P9.12-00A0.dtbo

https://github.com/beagleboard/bb.org-overlays/commit/675e258e08c24417e815984a6c63aad17bba6c66

```
debian@test-bbb-2:~$ cd /opt/source/bb.org-overlays/
debian@test-bbb-2:/opt/source/bb.org-overlays$ git pull
debian@test-bbb-2:/opt/source/bb.org-overlays$ make
debian@test-bbb-2:/opt/source/bb.org-overlays$ sudo make install
debian@test-bbb-2:/opt/source/bb.org-overlays$ sudo reboot
```
Regards,

cren14, 2018-04-10:
@pdp7 @RobertCNelson 

I followed @pdp7 directions and then yours as well @RobertCNelson and the gpio IN is now working properly. I didn't comment out uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-W1-P9.12-00A0.dtbo].

I have been using BB-W1-P9.12-00A0.dtbo to read ds18b20 1 wire temperature sensors but now they keep dropping out. When I first boot up the board and log in I can see them listed sometimes but other times they aren't listed in /sys/bus/w1/devices at all. The times they are listed they drop out as soon as I run my python code to read them.

Could any of the steps above have effected reading the temperature sensors?

pdp7, 2018-04-11:
@cren14 it is hard for me to say whether or not the above changes would have effected 1-wire.

> I have been using BB-W1-P9.12-00A0.dtbo to read ds18b20 1 wire temperature sensors but now they keep dropping out. When I first boot up the board and log in I can see them listed sometimes but other [...]

Are there any mentions of 1-wire or w1 in `dmesg` after boot up?

I am hoping there might be some type of error message to indicate why the 1-wire device failed to appear.

cren14, 2018-04-18:
The problem seems to of disappeared for now. Thank you all for your help.

closed 2018-04-18 by cren14