adafruit/adafruit-beaglebone-io-python #249: PWM frequency cannot be set

jindrichus, 2018-04-09:
Hello, I found the Adafruit_BBIO.PWM not working with other frequency than the default on BeagleBone Black. With the default 2kHz, everything seems to work properly, but when I try to change the frequency i got:

>>> PWM.set_frequency("P9_14", 20000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Could not write to the frequency file

The strace log is: https://gist.github.com/jindrichus/b06d16298777d808bdefe99e6bb2d50b

If I try change the frequency within the PWM.start(), I got:

>>> PWM.start("P9_14", 50, 20000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Problem with a sysfs file

The strace log is: https://gist.github.com/jindrichus/068034cce407754b30c68f2b5afc368e

The output of the version.sh is:
git:/opt/scripts/:[a890d66119f9b84505b4c162ce90ffce977475b5]
eeprom:[A335BNLT00C03214BBBK7006]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.32-ti-r42]
nodejs:[v6.13.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180305.0-0rcnee0~stretch+20180305]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.951706] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.957937] gpio-of-helper ocp:cape-universal: ready
END

I ran python with sudo. I have also tried it with the 4.4, 4.9 and 4.16 kernels. I remember, that I ran PWM with 30kHz in the past, but I am not able to reproduce the configuration I had back then. Must be some parts of the system in the exact version to proper function? 
Thanks for any answer.
Jindrich

pdp7, 2018-04-09:
@jindrichus thanks for reporting this.  I will attempt to reproduce and investigate what the cause might be.

@RobertCNelson @jadonk any insight on how the way userspace sets PWM freq may have changed?

RobertCNelson, 2018-04-09:
Looks fine to me:

```
root@beaglebone:~# cat pwm.py 
import Adafruit_BBIO.PWM as PWM
import time

PWM.start('P9_14',50,1000)
time.sleep(0.01)
PWM.set_frequency("P9_14", 20000)
time.sleep(0.01)
PWM.set_frequency("P9_14", 5000)
time.sleep(0.01)
PWM.stop("P9_14")
PWM.cleanup()
```

```
root@beaglebone:~# uname -r
4.16.1-bone8
```
```
root@beaglebone:~# cat /boot/uEnv.txt | grep BB
uboot_overlay_addr0=/lib/firmware/BB-PWM1-00A0.dtbo
```

fivdi, 2018-04-09:
Running this:
```python
import Adafruit_BBIO.PWM as PWM
import time

PWM.start('P9_16',50,5000)

PWM.start('P9_14',50,1000)
time.sleep(0.01)
PWM.set_frequency("P9_14", 20000)
time.sleep(0.01)
PWM.set_frequency("P9_14", 5000)
time.sleep(0.01)
PWM.stop("P9_14")
PWM.cleanup()
```
Results in this:
```
Traceback (most recent call last):
  File "pwm.py", line 6, in <module>
    PWM.start('P9_14',50,1000)
RuntimeError: Problem with a sysfs file
```
Note the addition of
```python
PWM.start('P9_16',50,5000)
```
This happens because P4_14 and P9_16 are channel 0 and 1 on the same eHRPWM instance and both channels must have the same frequency. If I remember correctly the both channels must be disabled to change the frequency.

@jindrichus is it possible that P9_16 is already in usage somewhere else? Perhaps in a different process?

jindrichus, 2018-04-10:
> @jindrichus is it possible that P9_16 is already in usage somewhere else? Perhaps in a different process?

That can be it. I want to use P9_14 and P9_16, but with the same frequency. So far I observed:
I can run:
>>> PWM.start("P9_14")
>>> PWM.set_frequency("P9_14", 30000)

 and PWM appears working. Then
>>> PWM.start("P9_16")

results in:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Problem with a sysfs file

If I start both channels first, then I am not able to set the frequency of either.
The same behavior is with reversed order of pins. eHRPWM2 exhibits the same.
So, now I can use one channel of each eHRPWM.

pdp7, 2018-04-10:
@jindrichus I see in your reply above that you encountered a Python error: `RuntimeError: Problem with a sysfs file`.

In the issue description, you linked to [this strace output](https://gist.github.com/jindrichus/068034cce407754b30c68f2b5afc368e).  Does that correspond to running `PWM.start("P9_16")`?

jindrichus, 2018-04-11:
The `RuntimeError: Problem with a sysfs file` in my last comment appeared after `PWM.start("P9_16")` when I first started `P9_14` and succesfully set its frequency by `PWM.set_frequency("P9_14", 30000)`.
In the issue description, the mentioned strace output is related to `PWM.start("P9_14", 50, 20000)`.

pdp7, 2018-04-12:
@jindrichus thanks.  I am able to reproduce that behavior and will investigate further to determine that cause.  It also good motivation for me to improve that error message (`RuntimeError: Problem with a sysfs file`) to provide more information.

silver2row, 2018-05-05:
Hello Again Sir,

Seth here.

I typed up some software to run w/ the PWM on my BBB. Here goes it:

    #!/usr/bin/python

    import Adafruit_BBIO.PWM as PWM
    import time

    PWM.start("P9_14", 50, 2000, 1)

    while True:
        eatAtMikes = ("start", 25)
        time.sleep(2)
        print "I Like Mikes!"

If this does not give you an issue, let me know. 

Seth

jindrichus, 2018-05-07:
@silver2row Thanks for your response, but I do not have a problem with outputing the PWM. I have a problem with setting the frequency of it (I am not even able to run the PWM.start("P9_14", 50, 2000, 1) command).

@pdp7 Did you have any success with your investigation? Please let me know if you need any more inputs from me.

silver2row, 2018-05-07:
Hello Everyone,

@jindrichus and @pdp7 

I am getting working order w/ PWM out of the box w/ this software (if this matters):

`import Adafruit_BBIO.PWM as PWM
import time

PWM.start("P9_21", 50, 2000, 1)

while True:

    PWM.stop("P9_21")
    PWM.start("P9_21", 50, 2000, 1)
    PWM.start("P9_22", 50, 2000, 1)
    PWM.start("P9_14", 50, 2000, 1)

    PWM.stop("P9_21")
    PWM.stop("P9_22")
    PWM.stop("P9_14")
    PWM.cleanup()
`
So,

I was thinking this idea...have you set up config-pin P9.14 pwm and if so, have you uninstalled the Adafruit_BBIO library and reinstalled it?

Seth

mallozzi, 2018-05-09:
I have been struggling with the same issue. I am new to both BeagleBone Black and Linux, but I found an odd path to making it work. 

I first installed the latest Debian image (bone-debian-9.3-lxqt-armhf-2018-01-28-4gb.img), and then from a comment from @pdp7 on the discussion https://github.com/adafruit/adafruit-beaglebone-io-python/issues/136 on March 10, 2017 I tried the following:

sudo apt-get update
sudo apt-get install build-essential git-core device-tree-compiler
git clone https://github.com/beagleboard/bb.org-overlays/
cd ./bb.org-overlays/
./install.sh
cd /opt/scripts/
git pull 
sudo shutdown -r now

I then ran a simple test script in python:

import Adafruit_BBIO.PWM as PWM
import time
pwmPin = "P9_14"
PWM.start(pwmPin,0,2000)
print "started pwm"
time.sleep(1)
PWM.set_duty_cycle(pwmPin,20)
print "set duty cycle"
PWM.stop(pwmPin)
PWM.cleanup()

The first time I ran it, I got this error:

Traceback (most recent call last):
  File "testPWM.py", line 7, in <module>
    PWM.start(pwmPin,0,2000)
IOError: could not access a necessary file

I simply ran the script a second time, got the same error, then ran it a third time, and got a different error:

Traceback (most recent call last):
  File "testPWM.py", line 7, in <module>
    PWM.start(pwmPin,0,2000)
RuntimeError: Problem with a sysfs file

The next time I ran it, and all the times after that, it worked. I have no idea why this worked, and am new to both Linux and BBB, but I hope this helps.

pdp7, 2018-05-10:
@mallozzi please paste the output of this command:
```/opt/scripts/tools/version.sh```

mallozzi, 2018-05-10:
@pdp7 I'm afraid I have installed a different image since that test. The follow-up info is that the PWM module worked until I decided to see what happens after rebooting. When I rebooted I got the same errors as before (Problem with sysfs file).  So I don't think this gives much info.

I have tested the last three or four lxqt images (I want the graphics), and none of them have had a working PWM for me (always a sysfs problem). However, if I install the latest iot image, the PWM works perfectly fine with the above test script. I am not measuring the PWM output to confirm that the output is there, only testing that the code runs without error for now.

Since I want the graphical output, I can't use the iot version.

pdp7, 2018-05-10:
@mallozzi please paste the output of:
```/opt/scripts/tools/version.sh```
and
```cat /boot/uEnv.txt```
This will help to determine the configuration of your system.

It would also be useful if you could run the Python script with strace and paste the contents into a GitHub Gist:
```sudo strace -f python ./test-spi.py &> /tmp/strace.log```

pdp7, 2018-08-16:
@jindrichus is this working for you now?

jindrichus, 2018-08-21:
Sorry for late reply. Fresh install on the newest recommended Debian now works well. Thank you.

closed 2018-08-21 by jindrichus