adafruit/adafruit-beaglebone-io-python #250: Delay required after enabling PWMs via udev?

amarburg, 2018-04-12:
- __Platform/operating system:__  Beaglebone Black rev C, BeagleBoard.org Debian Image 2018-01-28, no capes.

- __Python version__ (run `python -version` or `python3 -version`):  Python 2.7.13

- __Error message you are receiving,__ including any Python exception traces: 

Our MWE:

     import Adafruit_BBIO.PWM as PWM
     PWM.start("P9_14",0,10000)

on the first run after reboot, would fail with:

     RuntimeError: Problem with a sysfs file

The second (and all subsequent) runs would work.

Here is the relevant sequence in `strace`:

    stat64("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm0", 0xbe8dcd28) = -1 ENOENT (No such file or directory)
    stat64("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm-3:0", {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
    open("/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip3/pwm-3:0/period", O_RDWR) = -1 EACCES (Permission denied)

The code is failing the stat on [line 435](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/49b2be326267ca3e843b499c41b784cd0ef110a4/source/c_pwm.c#L435),  going successfully through the stat in [line 441](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/e061d1358e189115d3a7fd907470e6b4783c8733/source/c_pwm.c#L441) (middle line), then failing at [line 515](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/49b2be326267ca3e843b499c41b784cd0ef110a4/source/c_pwm.c#L515).

I wondered if there was a delay in udev's operation, I added a 100ms delay (pull request pending) which resolved this issue.   I don't think this is the best solution, but it hints at _a_ solution.

pdp7, 2018-04-16:
Thanks for creating this PR.  I will install on my BeagleBone and run tests.

amarburg, 2018-04-16:
@pdp7.  cheers, just let me know if/how I can help debug.

pdp7, 2018-07-20:
@amarburg I have merged #251.  Is this issue resolved?  thanks

amarburg, 2018-07-20:
I can't replicate the issue with a clean checkout w/ the merge, so yes, I believe it's resolved.

closed 2018-07-20 by amarburg