adafruit/adafruit-beaglebone-io-python #252: RE: Line 10 on the UART Example and The Print Function...

silver2row, 2018-04-14:
Hello Sir,

Seth here, again. Um, that UART example is out of date or something needs to be changed. The main page needs to have the UART example w/ the print function at four spaces only.

Seth

P.S. I tested the software and it works w/ Python 2.7.x w/ the print function at four spaces, i.e. I think it is line 10 that is "over-spaced."

silver2row, 2018-04-14:
If this does not describe well enough what I am saying, please let me know.

Seth

pdp7, 2018-04-16:
Please paste the code that you are attempting to run and the error that you receive.

silver2row, 2018-04-16:
Hello,

The main page has some software under UART1:

`import Adafruit_BBIO.UART as UART
import serial

UART.setup("UART1")

ser = serial.Serial(port = "/dev/ttyO1", baudrate=9600)
ser.close()
ser.open()
if ser.isOpen():
	print "Serial is open!"
    ser.write("Hello World!")
ser.close()`

Seth

P.S. I will get back to you on the error. The line w/ the print statement should be four spaces back, right?

pdp7, 2018-04-16:
I think the problem may be that there is a tab character instead of spaces.  Try to copy and paste this:
```
import Adafruit_BBIO.UART as UART
import serial

UART.setup("UART1")

ser = serial.Serial(port = "/dev/ttyO1", baudrate=9600)
ser.close()
ser.open()
if ser.isOpen():
    print "Serial is open!"
    ser.write("Hello World!")
ser.close()
```

silver2row, 2018-04-19:
Yep...that works. That is all.

Seth

closed 2018-04-19 by silver2row