adafruit/adafruit-beaglebone-io-python #253: Syntax error when using Adafruit BBIO library after upgrading BBB debian image

doananh3937, 2018-04-14:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **Windows 10**

- Python version (run `python -version` or `python3 -version`):  **Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2**

- Error message you are receiving, including any Python exception traces:  **>>> 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SystemError: error return without exception set
>>>
***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**
import
1. start python 2.7 interactive command prompt
2.  >>> import Adafruit_BBIO.GPIO as GPIO
3. >>> GPIO.setup("P9_36", GPIO.IN)

this works on earlier version of BBB debian image, i upgrade to Linux beaglebone 4.9.82-ti-r102 and it stopped working, i tried uninstalling Adafruit_BBIO and reinstalling using pip and same issue.

pdp7, 2018-04-16:
I would like to better understand the system software installed on your BeagleBone.

Please paste the output of:
```
    sudo /opt/scripts/tools/version.sh
    cat /boot/uEnv.txt 
```
If possible, I would recommend that you try a more [recent BeagleBoard.org Debian image](http://beagleboard.org/latest-images) such as "Stretch IoT (without graphical desktop)".  Here is direct download link: [Debian 9.3 2018-03-05 4GB SD IoT](http://debian.beagleboard.org/images/bone-debian-9.3-iot-armhf-2018-03-05-4gb.img.xz)

gert-lauritsen, 2018-04-19:
When I try is I get:
open("/sys/class/gpio/gpio112/direction", O_WRONLY) = 4
open("/sys/devices/platform/ocp/ocp:P9_30_pinmux/state", O_WRONLY|O_CREAT|O_TRUNC, 0666) = -1 ENOENT (No such file or directory)
The device is not there. If one makes "ls /sys/devices/platform/ocp/ocp:*" I get:
/sys/devices/platform/ocp/ocp:cape-universal:
driver  driver_override  modalias  of_node  power  status  subsystem  uevent

/sys/devices/platform/ocp/ocp:l4_wkup@44c00000:
44d00000.wkup_m3  44e10000.scm     modalias  power      uevent
44e00000.prcm     driver_override  of_node   subsystem

/sys/devices/platform/ocp/ocp:P9_19_pinmux:
driver  driver_override  modalias  of_node  power  state  subsystem  uevent

/sys/devices/platform/ocp/ocp:P9_20_pinmux:
driver  driver_override  modalias  of_node  power  state  subsystem  uevent

The system is:
Linux beaglebone 4.9.78-ti-r94 #1 SMP PREEMPT Fri Jan 26 21:26:24 UTC 2018 armv7l GNU/Linux

pdp7, 2018-06-07:
@gert-lauritsen are you still having this issue?

if so, please paste the output of:
> sudo /opt/scripts/tools/version.sh
> cat /boot/uEnv.txt

pdp7, 2018-06-11:
Please re-open if still an issue.

closed 2018-06-11 by pdp7