adafruit/adafruit-beaglebone-io-python #254: RE: PWM/"P9_14" and "P9_16"

silver2row, 2018-04-19:
Hello,

I am using:

- beagleBone Black

- Python 2.7.x

- uname -r
...4.9.78-ti-r94

I have tried to get PWM working on Adafruit_BBIO 1.0.10 and 1.0.3. I have been unsuccessful so far.

Please see the photo below:

[image]

I think that should sum things up for now.

If you need any other info, please let me know. Oh and before I forget, here is the software I have tried while getting an error: **RuntimeError: Problem with a sysfs file**

    import Adafruit_BBIO.PWM as PWM
    #import time

    MotorOne = "P9_14"
    MotorTwo = "P9_16"

    PWM.start("P9_14", 50)
    PWM.start("P9_16", 50)

    PWM.set_duty_cycle("P9_14", 25.5)
    PWM.set_duty_cycle("P9_16", 25.5)

    PWM.set_frequency("P9_14", 50)
    PWM.set_frequency("P9_16", 50)

    while True:
        MotorOne = ("P9_14")
	MotorTwo = ("P9_16")
	PWM.start("P9_14", 50)
        PWM.start("P9_16", 50)
	PWM.stop("P9_14")
	PWM.stop("P9_16")
	PWM.cleanup()
	break

Seth

P.S. This is one of many designs on the software side that I have attempted.

silver2row, 2018-04-19:
Hello,

I forgot to mention something. I am interfacing w/ a L298 on the BeagleBone Black.

Seth

silver2row, 2018-04-20:
Hello...I changed the software from what keeps showing up in my above mentioned software example. I think GitHub.com is having issues.

Seth

P.S. I changed the "four spacing" but the issue w/ showing the software is still incorrect.

silver2row, 2018-04-21:
[image]

This is a photo of the L298 and the rest of the set up for this Open Source board. I got the board from Italy.

Seth

P.S. If you need any other info, please do not hesitate to ask. I will keep trying to make PWM work for me.

silver2row, 2018-04-22:
Hello Once More,

Do I need to use config-pin xx.xx uart or should I stick w/ PWM and set up config-pin xx.xx pwm (like so)?

Seth

P.S. I figured I could get this working out of the box but I have been mistaken so far. What could I be missing?

pdp7, 2018-04-23:
@silver2row could you please paste your `/book/uEnv.txt` ?

silver2row, 2018-04-23:
Sure...I will get that done soon. Sory for the wait.

Seth

silver2row, 2018-04-24:
Hello,

I found #170 and I remembered everything. Odd days.

Seth

P.S. I will revert back to 4.4.x and I will see what happens. I tried 4.14.x but to no avail. I will keep you updated.

silver2row, 2018-04-24:
Hello, 

Otay...I tried 4.4.x, 4.9.x, and 4.14.x. I think I may be out of options. If you come up w/ anything, please let me know. I am still using the BBB, Python 2.7.x, and now uname -r is 4.4.113-ti-r149.

Seth

P.S. I changed the software to resemble something else...seen here:

[image]

pdp7, 2018-04-24:
The settings in your `/boot/uEnv.txt` looks OK.

If you are getting that Python error `RuntimeError: Problem with a sysfs file` then the best approach is to use strace to understand where it is failing.

Please install strace and run python with strace:
```
sudo apt-get install strace
sudo strace -f -o /tmp/strace-python.txt python filename.py
```
Paste the contents of `/tmp/strace-python.txt` into a GitHub gist and post the link here.

silver2row, 2018-04-24:
Hello Again,

I tried to use strace on my other files too. The other strace usage left me w/ more errors, e.g. errors that I was already receiving w/out using strace. I used chmod +x filename.py to set up the file. Should I cancel this idea and try another route?

Seth

silver2row, 2018-04-24:
Hello...I updated my kernel and strace works now. 

    https://gist.github.com/silver2row/32a7ae9b74a8110b45388cf0b3380798

Seth

P.S. Sorry for the confusion.

silver2row, 2018-04-24:
Hello Sir,

Now that we are having this conversation, I remember a while back that I may have changed some software on the BBB for Adafruit_BBIO but since then, I erased Adafruit_BBIO twice and replaced it w/ different versions. I think the "tinkering" may have been obsolete b/c of the erasure and then installation of Adafruit_BBIO over and over (trying to get the different versions working w/ what I know currently).

Seth

silver2row, 2018-04-25:
Hello,

I added some extra info. Sir, there are a bunch of different info. parts to this strace. Here is the updated info.

...

[https://gist.github.com/silver2row/32a7ae9b74a8110b45388cf0b3380798](url)

Seth

P.S. Let me know if you need any additional info.

silver2row, 2018-04-25:
Hello Again Sir,

Just for reference, I am using the P9_21 and P9_22 pins on my BBB now. I think I have tried all pwm pins now and I will stay w/ these two pins for now. 

Seth

silver2row, 2018-04-25:
Yes...I get the "sysfs" error still.

Seth

silver2row, 2018-04-25:
@pdp7 ,

Hello Sir...Seth here. Now, I changed up the software unlike before and got somewhere. I got a print function to print 2500 times at the speed of light. 

Seth

P.S. I have not added battery power yet b/c the print function states that my motors would run too compared to what really needs to happen. If anything, what would you change? See here:

    import Adafruit_BBIO.PWM as PWM
    import time

    MotorOne = "P9_21"
    MotorTwo = "P9_22"

    PWM.start(MotorOne, 50, 2000, 1)
    PWM.start(MotorOne, 50, 2000, 1)

    for i in range(0, 2500):
        taco = (MotorOne, 10)
        bien = (MotorTwo, 10)
        print "I love Lucy!"

        PWM.stop(MotorOne)
        PWM.stop(MotorTwo)
        PWM.cleanup

silver2row, 2018-04-25:
@pdp7 ,

I have wasted your time and mine. I am very sorry. It was software and not on your side of things. 

Seth

P.S. If you know how to slow down the Python script to move slower, please let me know or do not. Either way, "Party on Garth!"

closed 2018-04-26 by silver2row