adafruit/adafruit-beaglebone-io-python #255: enable 2PWM channel withe different frequecy through uboot

dongdianbiao, 2018-04-23:
Dear Mr/Mrs, I need to use two SPI and two different frequency PWM with BBB. we select P9_14 and P8_13. but when I applied the  

> Debian 9.3 2018-01-28 4GB SD LXQT

 image and set the uEnv.txt as 

```
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLTBWA51650BBWG1130]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]
kernel:[4.17.0-rc2-bone0]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-PWM1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/BB-PWM2-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr3=/lib/firmware/<file3>.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    0.985894] pinctrl-single 44e10800.pinmux: 142 pins, size 568
END
```

only P8_13 can work. The results of ls /sys/devices/platform/ocp/ | grep pinmux*:

```
ocp:P8_07_pinmux
ocp:P8_08_pinmux
ocp:P8_09_pinmux
ocp:P8_10_pinmux
ocp:P8_11_pinmux
ocp:P8_12_pinmux
ocp:P8_14_pinmux
ocp:P8_15_pinmux
ocp:P8_16_pinmux
ocp:P8_17_pinmux
ocp:P8_18_pinmux
ocp:P8_26_pinmux
ocp:P8_27_pinmux
ocp:P8_28_pinmux
ocp:P8_29_pinmux
ocp:P8_30_pinmux
ocp:P8_31_pinmux
ocp:P8_32_pinmux
ocp:P8_33_pinmux
ocp:P8_34_pinmux
ocp:P8_35_pinmux
ocp:P8_36_pinmux
ocp:P8_37_pinmux
ocp:P8_38_pinmux
ocp:P8_39_pinmux
ocp:P8_40_pinmux
ocp:P8_41_pinmux
ocp:P8_42_pinmux
ocp:P8_43_pinmux
ocp:P8_44_pinmux
ocp:P8_45_pinmux
ocp:P8_46_pinmux
ocp:P9_11_pinmux
ocp:P9_12_pinmux
ocp:P9_13_pinmux
ocp:P9_15_pinmux
ocp:P9_19_pinmux
ocp:P9_20_pinmux
ocp:P9_23_pinmux
ocp:P9_24_pinmux
ocp:P9_25_pinmux
ocp:P9_26_pinmux
ocp:P9_27_pinmux
ocp:P9_28_pinmux
ocp:P9_29_pinmux
ocp:P9_30_pinmux
ocp:P9_31_pinmux
ocp:P9_41_pinmux
ocp:P9_42_pinmux
ocp:P9_91_pinmux
ocp:P9_92_pinmux
```

 If we change the sequence of BB-PWM1-00A0.dtbo and BB-PWM2-00A0.dtbo in uEnv.txt, then P9_14 can give output instead of P8_13. it seems the final one will be activated and the first one wont work. Do you guys know how to fix this? I really cost a lot of days to activate the function of BBB. thanks everyone

RobertCNelson, 2018-04-23:
Fixed with:

https://github.com/beagleboard/bb.org-overlays/commit/06e9132ef3bb539573af551125844e132ad480fa

and

https://github.com/beagleboard/bb.org-overlays/commit/06e9132ef3bb539573af551125844e132ad480fa

```
#!/bin/bash

#P9_14
echo 0 > /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip*/export
#P8_13
echo 1 > /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip*/export

sleep 2

echo 1000 > /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip*/pwm-*0/period
echo 500 > /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip*/pwm-*0/duty_cycle
echo 1 > /sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/pwm/pwmchip*/pwm-*0/enable

echo 2000 > /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip*/pwm-*1/period
echo 1000 > /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip*/pwm-*1/duty_cycle
echo 1 > /sys/devices/platform/ocp/48304000.epwmss/48304200.pwm/pwm/pwmchip*/pwm-*1/enable

```

[image]

dongdianbiao, 2018-04-24:
@RobertCNelson Dear Robert, the two PWM can be successfully enable now!! thanks a lot. but when I try to enable two SPI at the same time as 

```
root@beaglebone:/lib/firmware# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
eeprom:[A335BNLTBWA51650BBWG1130]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]
kernel:[4.17.0-rc2-bone0]
nodejs:[v6.12.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-PWM1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/BB-PWM2-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr3=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180423.0-0rcnee0~stretch+20180423]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
dmesg | grep pinctrl-single
[    0.986877] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.196736] pinctrl-single 44e10800.pinmux: pin PIN100 already requested by ocp:P9_31_pinmux; cannot claim for 481a0000.spi
[    1.208027] pinctrl-single 44e10800.pinmux: pin-100 (481a0000.spi) status -22
[    1.215223] pinctrl-single 44e10800.pinmux: could not request pin 100 (PIN100) from group pinmux_bb_spi1_pins  on device pinctrl-single
END

```

the first SPI (0,0 ) can work but the SPI (1,0) cannot work.
the result of ls /sys/devices/platform/ocp/ | grep pinmux*:
`ocp:P8_07_pinmux
ocp:P8_08_pinmux
ocp:P8_09_pinmux
ocp:P8_10_pinmux
ocp:P8_11_pinmux
ocp:P8_12_pinmux
ocp:P8_14_pinmux
ocp:P8_15_pinmux
ocp:P8_16_pinmux
ocp:P8_17_pinmux
ocp:P8_18_pinmux
ocp:P8_26_pinmux
ocp:P8_27_pinmux
ocp:P8_28_pinmux
ocp:P8_29_pinmux
ocp:P8_30_pinmux
ocp:P8_31_pinmux
ocp:P8_32_pinmux
ocp:P8_33_pinmux
ocp:P8_34_pinmux
ocp:P8_35_pinmux
ocp:P8_36_pinmux
ocp:P8_37_pinmux
ocp:P8_38_pinmux
ocp:P8_39_pinmux
ocp:P8_40_pinmux
ocp:P8_41_pinmux
ocp:P8_42_pinmux
ocp:P8_43_pinmux
ocp:P8_44_pinmux
ocp:P8_45_pinmux
ocp:P8_46_pinmux
ocp:P9_11_pinmux
ocp:P9_12_pinmux
ocp:P9_13_pinmux
ocp:P9_15_pinmux
ocp:P9_19_pinmux
ocp:P9_20_pinmux
ocp:P9_23_pinmux
ocp:P9_24_pinmux
ocp:P9_25_pinmux
ocp:P9_26_pinmux
ocp:P9_27_pinmux
ocp:P9_28_pinmux
ocp:P9_29_pinmux
ocp:P9_30_pinmux
ocp:P9_31_pinmux
ocp:P9_41_pinmux
ocp:P9_42_pinmux
ocp:P9_91_pinmux
ocp:P9_92_pinmux`

while if the sencond spi can work, there shouldnt be P9_28, P9_29, P9_30 and P9_31 in this list. Could you help with this problem? thanks a lot!

RobertCNelson, 2018-04-24:
The issue is shown above here:
```
dmesg | grep pinctrl-single
[    0.986877] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.196736] pinctrl-single 44e10800.pinmux: pin PIN100 already requested by ocp:P9_31_pinmux; cannot claim for 481a0000.spi
[    1.208027] pinctrl-single 44e10800.pinmux: pin-100 (481a0000.spi) status -22
[    1.215223] pinctrl-single 44e10800.pinmux: could not request pin 100 (PIN100) from group pinmux_bb_spi1_pins  on device pinctrl-single
END
```
this should be fixed via:

https://github.com/beagleboard/bb.org-overlays/commit/604c0926a4f7505dfc3d501301413c821e59febe

Regards,

dongdianbiao, 2018-04-24:
Dear Robert @RobertCNelson , when I try to use your code to update the overlays, this happened : 

```
root@beaglebone:/usr/mycode# sudo apt install bb-cape-overlays
Reading package lists... Done
Building dependency tree
Reading state information... Done
bb-cape-overlays is already the newest version (4.4.20180423.0-0rcnee0~stretch+20180423).
0 upgraded, 0 newly installed, 0 to remove and 121 not upgraded.
root@beaglebone:/usr/mycode#
```
and after reboot the BBB, the question is still there. 
```
root@beaglebone:/usr/mycode# python encoder.py
Traceback (most recent call last):
  File "encoder.py", line 22, in <module>
    spi=SPI(1,0)#
IOError: [Errno 2] No such file or directory
```
and pinmux : 
`
root@beaglebone:/usr/mycode# ls /sys/devices/platform/ocp/ | grep pinmux*
ocp:P8_07_pinmux
ocp:P8_08_pinmux
ocp:P8_09_pinmux
ocp:P8_10_pinmux
ocp:P8_11_pinmux
ocp:P8_12_pinmux
ocp:P8_14_pinmux
ocp:P8_15_pinmux
ocp:P8_16_pinmux
ocp:P8_17_pinmux
ocp:P8_18_pinmux
ocp:P8_26_pinmux
ocp:P8_27_pinmux
ocp:P8_28_pinmux
ocp:P8_29_pinmux
ocp:P8_30_pinmux
ocp:P8_31_pinmux
ocp:P8_32_pinmux
ocp:P8_33_pinmux
ocp:P8_34_pinmux
ocp:P8_35_pinmux
ocp:P8_36_pinmux
ocp:P8_37_pinmux
ocp:P8_38_pinmux
ocp:P8_39_pinmux
ocp:P8_40_pinmux
ocp:P8_41_pinmux
ocp:P8_42_pinmux
ocp:P8_43_pinmux
ocp:P8_44_pinmux
ocp:P8_45_pinmux
ocp:P8_46_pinmux
ocp:P9_11_pinmux
ocp:P9_12_pinmux
ocp:P9_13_pinmux
ocp:P9_15_pinmux
ocp:P9_19_pinmux
ocp:P9_20_pinmux
ocp:P9_23_pinmux
ocp:P9_24_pinmux
ocp:P9_25_pinmux
ocp:P9_26_pinmux
ocp:P9_27_pinmux
ocp:P9_28_pinmux
ocp:P9_29_pinmux
ocp:P9_30_pinmux
ocp:P9_31_pinmux
ocp:P9_41_pinmux
ocp:P9_42_pinmux
ocp:P9_91_pinmux
ocp:P9_92_pinmux
`

sorry I have so many questions. 😁

RobertCNelson, 2018-04-24:
```
sudo apt update
sudo apt install --only-upgrade bb-cape-overlays
```

dongdianbiao, 2018-04-24:
@RobertCNelson it's solved!!! thank you sosososososo much!. I cost really a lot of days to seach information and find solutions until I saw your post in Git!.

closed 2018-04-29 by pdp7