adafruit/adafruit-beaglebone-io-python #256: kernel version selection uses build machine rather than target

jharvell, 2018-05-05:
library version:  master:

the library supports two kernel interfaces, and the detection of this is in setup.py:

kernel = platform.release() 

if kernel >= '4.1.0': 
   kernel41 = [('BBBVERSION41', None)] 
else: 
   kernel41 = None

In my case, my build machine is running a 4.16.3 kernel, so the library gets built correctly for

pdp7, 2018-07-24:
@RobertCNelson any suggestions on how this could be handled?  thanks!

RobertCNelson, 2018-07-24:
@pdp7 sorry, i hack around that too..

one of my fast arm builders used a v3.14.x based kernel..

https://github.com/RobertCNelson/omap-image-builder/blob/master/target/chroot/beagleboard.org-stretch.sh#L189

Yes, it should be an export option.. (defaults to 4.1.x+ ;) )

Regards,