adafruit/adafruit-beaglebone-io-python #258: Controlling motor channels in BeagleBone BLue

sampathkmurthy, 2018-06-01:
Hello @pdp7,

I am beginner with the BeagleBone Blue. I am trying to control DC motor channels on BeagleBone Blue with python script written on it. I have a query about the working of Adafruit_BBIO library with the BeagleBone Blue. 

In the link attached https://github.com/adafruit/adafruit-beaglebone-io-python/issues/236 , I saw
In one of the issues raised at you regarding the controlling of servo motor by beagle bone, you mentioned that Adafruit_BBIO does work with the BeagleBone Blue.  How can I work with the Adafruit_BBIO library on BeagleBone Blue? and I tried with rc_test_motors command to  check the response and it is working for the given duty cycle value and motor channel. 

My aim is to control the DC motor channels of Beagle Bone Blue with Python API script. Please, do guide me with this issue.

 Thanks in advance,

Regards,

Sampath Kumar Krishna Murthy

pdp7, 2018-06-05:
@sampathkmurthy  Thanks for opening this issue.  One suggestion I got when talking with @jadon was to use `rcpy` by @mcdeoliveira:
https://github.com/mcdeoliveira/rcpy

@jadon also suggested that I add the ability for Adafruit_BBIO to call `rcpy`.

I'm going to also check with @MarkAYoder to see what he thinks.

sampathkmurthy, 2018-06-07:
Hallo @pdp7 ,

Thank you for your response. I have one more query related to Adafruit_BBIO on Beaglebone Blue. Is it possible to work with Adafruit_BBIO library on BeagleBone Blue? is there any procedure for this. I followed the guidelines to install rcpy on my Beaglebone blue with the link you provided.

Board : Beaglebone Blue
Platform : Linux distribution debian image (latest IOT image from beagleboard).
kernel verson : 4.9.82

In the issue #236 , you mentioned Adafruit_BBIO does work with Beaglebone Blue. I am confused reagrding this, whethere the Adafruit_BBIO supports Beaglebone Blue or not? if it supports, how can I install Adafruit_BBIO on BBblue? 

Please clarify this question.

Thanks in advance.

Regards,
Sampath Kumar

pdp7, 2018-07-20:
@sampathkmurthy 
please try the latest Debian 9.x (Stretch) image on:
https://beagleboard.org/latest-images

It will contain the Adafruit_BBIO library.

@RobertCNelson is there a package that contains rcpy?

sampathkmurthy, 2018-07-24:
@pdp7 ,

Thank you for the response,
I have this new image installed and I am successfull in controlling the motor channels. Thank you for your suggestions.
Regards,

closed 2018-07-24 by sampathkmurthy