adafruit/adafruit-beaglebone-io-python #259: runtimeerror: problem with a sysfs file

ududj, 2018-06-19:
I already found a resolution of my problem and already tried some solution propouse, but it persisty.

I don´t have pratice to write in english, sorry my lot of mistakes.

I am using Beagle bone black, with a remote terminal (Putty), from windows 10 (64 bit)
it is runing Debian9.3/Linux, python 2.7.13

I am trying to use a simple pwm comand, but the error message show me.
**runtimeerror: problem with a sysfs file**

my script:
 Import Adafruit_BBIO.PWM as pwm
 myPWM = "P8_19"
 pwm.start(myPWM, 0, 1000)

put other trings before time, but deteted and the error continue:
runtimeerror: problem with a sysfs file

the comand that I'm using is: python teste.py

ududj, 2018-06-19:
I discovered that debian no have privilegies, i change the user and it worked.
how about the comand i can use to grant privileges to debian user?

pdp7, 2018-07-08:
New Debian image and Linux kernel should solve this issue.

I'd like to know more about what is installed. Please paste the output of:
```
sudo /opt/scripts/tools/version.sh
```
and:
```
cat /boot/uEnv.txt
```

pdp7, 2018-07-24:
@ududj are you still experiencing this issue?

pdp7, 2018-08-16:
Please reopen if still an issue.

closed 2018-08-16 by pdp7