adafruit/adafruit-beaglebone-io-python #261: BeagleBone Blue: Set gpio direction failed, missing file or invalid permissions.

vmasn, 2018-06-24:
Hi,

I'm using Cloud9 IDE to write code for BeagleBone Blue with Debian 9.4 2018-06-17 4GB SD LXQT installed (found [there](https://beagleboard.org/latest-images))

Here is the code:

```
import Adafruit_BBIO.GPIO as GPIO

GPIO.setup("P8_9", GPIO.IN)
GPIO.setup("P8_7", GPIO.OUT)

while True:
    if GPIO.input("P8_9"):
        GPIO.output("P8_7", GPIO.HIGH)
    else:
        GPIO.output("P8_7", GPIO.LOW)
```
According to [this link](https://github.com/beagleboard/beaglebone-blue/blob/master/BeagleBone_Blue_Pin_Table.csv), P8_9 is pause button and P8_7 is red LED.

But I get this error:
```
Traceback (most recent call last):
  File "/var/lib/cloud9/examples/blink.py", line 4, in <module>
    GPIO.setup("P8_9", GPIO.IN)
ValueError: Set gpio direction failed, missing file or invalid permissions.

```
I have tried updating kernel and bootloader according to issue #247 , but that did not solve the problem. 
Here is sudo /opt/scripts/tools/version.sh output:

```
git:/opt/scripts/:[31292bce1d4b35b497cc7a013d6d57e7e1f4f5c4]
eeprom:[A335BNLTBLA21712EL003553]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-06-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.17.2-bone3]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180622.0-0rcnee0~stretch+20180622]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.937803] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
```

Also, i have to mention that there's no problem runing this code:
```
import Adafruit_BBIO.GPIO as GPIO

GPIO.setup("P8_7", GPIO.OUT)
GPIO.output("P8_7", GPIO.HIGH)
```
So, there's no problem with LED's.

pdp7, 2018-07-08:
@vmasn It would be helpful if you could run strace so that I can better understand what is happening.

Install strace (if it doesn't already exist):
```
sudo apt-get install strace
```
Then run strace, replace test.py with the name of your test program:
```
strace -f -o strace.log python test.py
```
Paste the contents of strace.log into a GitHub Gist and post the link in a reply here.

pdp7, 2018-07-24:
@ududj are you still experiencing this issue?

pdp7, 2018-07-26:
Here is the output of version.sh on my BeagleBone:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[31292bce1d4b35b497cc7a013d6d57e7e1f4f5c4]
eeprom:[A335BNLTBBG1BBG217110442]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Image 2018-06-17]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.49-ti-r54]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180628.0-0rcnee0~stretch+20180628]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.137499] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.147969] gpio-of-helper ocp:cape-universal: ready
END
```

I am able to run this program without any issues:
```
import Adafruit_BBIO.GPIO as GPIO

GPIO.setup("P8_9", GPIO.IN)
GPIO.setup("P8_7", GPIO.OUT)

while True:
  if GPIO.input("P8_9"):
    GPIO.output("P8_7", GPIO.HIGH)
  else:
    GPIO.output("P8_7", GPIO.LOW)
```

pdp7, 2018-07-26:
@vmasn  Please reopen if you are still having this problem.

closed 2018-07-26 by pdp7