adafruit/adafruit-beaglebone-io-python #262: Usage of PWM fails with "Problem with the cape manager"

andrecurvello, 2018-06-26:
Hi guys,

I'm having a problem here with the Adafruit_BBIO PWM, in which I'm unable to activate and use the PWM module of a BeagleBone Green Wireless.

I tried to replicate all the steps detailed in the [Issue 136](https://github.com/adafruit/adafruit-beaglebone-io-python/issues/136), that is similar as the problem I'm facing now, but with no success.

I followed all the commands that @pdp7 mentioned.

My setup is this one:

- SeeedStudio BeagleBone Green Wireless
- Debian 9.4 2018-06-17 4GB SD IoT
- Linux beaglebone 4.14.49-ti-r54 #1 SMP PREEMPT Fri Jun 15 22:14:13 UTC 2018 armv7l GNU/Linux
- Python 2.7.13
- Adafruit_BBIO 1.0.10

**Proving it:**
`root@beaglebone:~/bb.org-overlays# cat /etc/dogtag
BeagleBoard.org Debian Image 2018-06-17
root@beaglebone:~/bb.org-overlays# cat /etc/debian_version
9.4
root@beaglebone:~/bb.org-overlays# uname -a
Linux beaglebone 4.14.49-ti-r54 #1 SMP PREEMPT Fri Jun 15 22:14:13 UTC 2018 armv7l GNU/Linux
oot@beaglebone:~/bb.org-overlays# pip install Adafruit_BBIO
Requirement already satisfied: Adafruit_BBIO in /usr/local/lib/python2.7/dist-packages/Adafruit_BBIO-1.0.10-py2.7-linux-armv7l.egg
`

Output from dmesg:
[dmesg output from BeagleBone Green Wireless with Debian 9.4 2018-06-17 4GB SD IoT](https://gist.github.com/andrecurvello/018f573a190677de82fccb6dcdbab037)

Output from /boot/uEnv.txt:
[uEnv.txt from Debian 9.4 IoT](https://gist.github.com/andrecurvello/da681cdd63ed0f6f6226100bcd717aef)

Output from # cat /sys/devices/platform/bone_capemgr/slots:
`root@beaglebone:~# cat /sys/devices/platform/bone_capemgr/slots
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
`

Cape Overlays are already installed:
`root@beaglebone:~/bb.org-overlays# dpkg-query -l |grep overlay
ii  bb-cape-overlays                      4.4.20180625.0-0rcnee0~stretch+20180625           armhf        Device tree overlays for Beaglebone.
`

Output from # sudo strace -f -o trace.txt python and entering the following
```
from Adafruit_BBIO import PWM
PWM.start('P9_14', 50)
```
[Strace Output from Python execution of Adafruit_BBIO test of PWM Module](https://gist.github.com/andrecurvello/affbdb9e1bcf409c71a82dc2a1befff6)

Git log from /opt/scritps (showing that it is updated):
`commit 99d7b0cd3213e38cc58597a7d3406640a2469952
Author: Robert Nelson <robertcnelson@gmail.com>
Date:   Mon Jun 25 11:29:21 2018 -0500

    version.sh: also report dtb_overlay if used

    Signed-off-by: Robert Nelson <robertcnelson@gmail.com>

commit c190498f25b51d83e452a296de31e2e65883d001
Author: Robert Nelson <robertcnelson@gmail.com>
Date:   Tue Jun 19 13:30:17 2018 -0500

    version: add parsing of machinekit user

    Signed-off-by: Robert Nelson <robertcnelson@gmail.com>

commit 31292bce1d4b35b497cc7a013d6d57e7e1f4f5c4
Author: Robert Nelson <robertcnelson@gmail.com>
Date:   Mon Jun 11 14:28:11 2018 -0500

    nginx: add options both buster/bionic

    Signed-off-by: Robert Nelson <robertcnelson@gmail.com>
`

And with all of that:
`root@beaglebone:~# python pwm.py
Traceback (most recent call last):
  File "pwm.py", line 2, in <module>
    PWM.start('P9_14', 50)
RuntimeError: Problem with the cape manager`

RobertCNelson, 2018-06-26:
@abachman please run:

```
sudo /opt/scripts/tools/version.sh
```

Regards,

andrecurvello, 2018-06-26:
Hi @RobertCNelson and @pdp7,

After executing this script, I've got this output:
```
root@beaglebone:~# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[99d7b0cd3213e38cc58597a7d3406640a2469952]
eeprom:[A335BNLTGW1ABBGW16054907]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-06-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]:[location: dd MBR]
kernel:[4.14.49-ti-r54]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180625.0-0rcnee0~stretch+20180625]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet cape_universal=enable]
dmesg | grep pinctrl-single
[    1.058866] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.060334] gpio-of-helper ocp:cape-universal: ready
END
root@beaglebone:~#
```

andrecurvello, 2018-06-26:
I'm also facing a strange issue when calling the ADC methods from the Adafruit_BBIO, but I think that it is correlated with this cape_universal thing.

RobertCNelson, 2018-06-26:
U-Boot in the eMMC is blocking the new image from working..
```
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]:[location: dd MBR]
k
```

Just run:

```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

and reboot..

Regards,

andrecurvello, 2018-07-10:
Thanks @RobertCNelson and @pdp7, it worked!

@RobertCNelson, I flashed the eMMC with the Debian 9.4 IoT, to update the onboard U-Boot as well.

Thanks,
Andre Curvello

closed 2018-07-10 by andrecurvello