adafruit/adafruit-beaglebone-io-python #263: SPIDEV ABI change on mainline Linux kernel

pdp7, 2018-06-26:
FYI - a solution will need to be decided upon for libraries like Adafruit_BBIO

```
---------- Forwarded message ----------
From: Robert Nelson 
Date: Tue, Jun 26, 2018 at 11:03 AM
Subject: [beagle-alpha] SPIDEV abi change on mainline

Mainline:

debian@beaglebone:~$ ls /sys/devices/platform/ocp/48030000.spi/spi_master/spi*/
device  of_node  power  spi0.0  spi0.1  statistics  subsystem  uevent
debian@beaglebone:~$ ls /sys/devices/platform/ocp/481a0000.spi/spi_master/spi*/
device  of_node  power  spi1.0  spi1.1  statistics  subsystem  uevent
debian@beaglebone:~$ ls /dev/spidev*
/dev/spidev0.0  /dev/spidev0.1  /dev/spidev1.0  /dev/spidev1.1

v4.1.x -> v4.9.x

debian@beaglebone:~$ ls /sys/devices/platform/ocp/48030000.spi/spi_master/spi*/
device  of_node  power  spi1.0  spi1.1  statistics  subsystem  uevent
debian@beaglebone:~$ ls /sys/devices/platform/ocp/481a0000.spi/spi_master/spi*/
device  of_node  power  spi2.0  spi2.1  statistics  subsystem  uevent
debian@beaglebone:/sys/devices/platform/ocp$ ls /dev/spi*
/dev/spidev1.0  /dev/spidev1.1  /dev/spidev2.0  /dev/spidev2.1

v3.8.13:

debian@beaglebone:~$ ls /sys/devices/ocp.*/48030000.spi/spi_master/spi*/
device    power  spi1.0  spi1.1  subsystem  uevent
debian@beaglebone:~$ ls /sys/devices/ocp.*/481a0000.spi/spi_master/spi*/
device  power  spi2.0  spi2.1  subsystem  uevent
debian@beaglebone:~$ ls /dev/spidev*
/dev/spidev1.0  /dev/spidev1.1  /dev/spidev2.0  /dev/spidev2.1

What to Do:

In v4.14.x i'm patching things so it's v4.1.x > v4.9.x compabile..

But, we should change all the libr's to use the "ocp" directory
lookup's shown above..
```

cc @RobertCNelson

pdp7, 2018-10-09:
For example:
```/sys/devices/platform/ocp/48030000.spi/spi_master/spi1/spi1.0/```
who do I get to the actual character device file that I would open?

@RobertCNelson wrote:
```spi1.0 -> /dev/spidev1.0```

```/sys/devices/platform/ocp/*.spi/spi_master/spiX/spiX.Y/ =
/dev/spidevX.Y```

pdp7, 2019-04-04:
For reference, here is a commit to add the /dev/spi/X.Y properties to device tree:
*SPIDEV: use symlink label to help udev*
https://github.com/rcn-ee/repos/commit/3efd567a81bbdd262a50d6167f93c0e7765379cf

And udev script that create the corresponding symlinks:
https://github.com/mvduin/py-uio/blob/master/etc/udev/rules.d/10-of-symlink.rules

pdp7, 2019-12-24:
FYI - this will solve issues like this one:
SPI No such file or directory
#322

matthijskooijman, 2020-05-01:
I also ran into this problem, which got me a bit by surprise.

However, it does seem the new names make more sense: Now it uses spidev0 and spidev1, which match the labels on the pinout and the overlays (e.g. `BB-SPIDEV0-00A0.dtbo`). However, it does seem that the `/dev/spidev.x.x` is rather arbitrary (e.g. like ethx labels used to be), so the new symlinks are probably the best way to target specific spi devices.