adafruit/adafruit-beaglebone-io-python #264: GPIO.IN mode failed, missing file or invalid permissions

rjdyk, 2018-06-27:
Hi, I'm having trouble getting GPIO input on the Beaglebone Black using the Adafruit_BBIO library.
Setting a pin to GPIO.OUT works fine:

```
	debian@beaglebone:~/Desktop$ python -i
	Python 2.7.13 (default, Nov 24 2017, 17:33:09)
	[GCC 6.3.0 20170516] on linux2
	Type "help", "copyright", "credits" or "license" for more information.
	>>> import Adafruit_BBIO.GPIO as GPIO
	>>>
	>>> GPIO.setup("P8_7", GPIO.OUT)
	>>> print(GPIO.input("P8_7"))
	0
	>>> GPIO.output("P8_7", GPIO.HIGH)
	>>> print(GPIO.input("P8_7"))
	1
	>>> exit()
```

When I try to set the pin to GPIO.IN, it gives the error
	ValueError: Set gpio mode failed, missing file or invalid permissions.

```
	debian@beaglebone:~/Desktop$ python -i
	Python 2.7.13 (default, Nov 24 2017, 17:33:09)
	[GCC 6.3.0 20170516] on linux2
	Type "help", "copyright", "credits" or "license" for more information.
	>>> import Adafruit_BBIO.GPIO as GPIO
	>>>
	>>> GPIO.setup("P8_7", GPIO.IN)
	Traceback (most recent call last):
	  File "<stdin>", line 1, in <module>
	ValueError: Set gpio mode failed, missing file or invalid permissions.
	>>> exit()
```

How do I read in GPIO input?

This is the output to sudo /opt/scripts/tools/version.sh

	```
debian@beaglebone:~/Desktop$ sudo /opt/scripts/tools/version.sh
	[sudo] password for debian:
	git:/opt/scripts/:[ea6ea9fca05f36f5044398884375b0231348d6e2]
	eeprom:[A335BNLT00C02916BBBK1F31]
	model:[TI_AM335x_BeagleBone_Black]
	dogtag:[BeagleBoard.org Debian Image 2018-01-28]
	bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]
	kernel:[4.9.78-ti-r94]
	nodejs:[v6.12.3]
	uboot_overlay_options:[enable_uboot_overlays=1]
	uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo]
	uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-UART4-00A0.dtbo]
	uboot_overlay_options:[enable_uboot_cape_universal=1]
	pkg:[bb-cape-overlays]:[4.4.20180126.0-0rcnee0~stretch+20180126]
	pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
	pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
	groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
	dmesg | grep pinctrl-single
	[    1.385802] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
	END
```

and dmesg | grep bone

```
	debian@beaglebone:~/Desktop$ dmesg | grep bone
	[    0.000000] Kernel command line: console=ttyO0,115200n8 bone_capemgr.enable_partno=BB-UART1,BB-UART4 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
	[    6.620318] systemd[1]: Set hostname to <beaglebone>.
```

some additional information

```
	debian@beaglebone:~/Desktop$ cat /etc/dogtag
	BeagleBoard.org Debian Image 2018-01-28
```

	debian@beaglebone:~/Desktop$ cat /boot/uEnv.txt
```
	#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

	uname_r=4.9.78-ti-r94
	#uuid=
	#dtb=

	###U-Boot Overlays###
	###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
	###Master Enable
	enable_uboot_overlays=1
	###
	###Overide capes with eeprom
	uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo
	uboot_overlay_addr1=/lib/firmware/BB-UART4-00A0.dtbo
	#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
	#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
	###
	###Additional custom capes
	#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
	#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
	#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
	#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
	###
	###Custom Cape
	#dtb_overlay=/lib/firmware/<file8>.dtbo
	###
	###Disable auto loading of virtual capes (emmc/video/wireless/adc)
	#disable_uboot_overlay_emmc=1
	#disable_uboot_overlay_video=1
	#disable_uboot_overlay_audio=1
	#disable_uboot_overlay_wireless=1
	#disable_uboot_overlay_adc=1
	###
	###PRUSS OPTIONS
	###pru_rproc (4.4.x-ti kernel)
	#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
	###pru_uio (4.4.x-ti & mainline/bone kernel)
	#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
	###
	###Cape Universal Enable
	enable_uboot_cape_universal=1
	###
	###Debug: disable uboot autoload of Cape
	#disable_uboot_overlay_addr0=1
	#disable_uboot_overlay_addr1=1
	#disable_uboot_overlay_addr2=1
	#disable_uboot_overlay_addr3=1
	###
	###U-Boot fdt tweaks... (60000 = 384KB)
	#uboot_fdt_buffer=0x60000
	###U-Boot Overlays###

	cmdline=coherent_pool=1M net.ifnames=0 quiet

	#In the event of edid real failures, uncomment this next line:
	#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

	##Example v3.8.x
	#cape_disable=capemgr.disable_partno=
	cape_enable=bone_capemgr.enable_partno=BB-UART1,BB-UART4

	##Example v4.1.x
	#cape_disable=bone_capemgr.disable_partno=
	cape_enable=bone_capemgr.enable_partno=BB-UART1,BB-UART4

	##enable Generic eMMC Flasher:
	##make sure, these tools are installed: dosfstools rsync
```

RobertCNelson, 2018-06-27:
```bash
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]
```
Run, this script to update your bootloader to 2018.03 with more fixes..
```bash
debian@beaglebone:~$ sudo /opt/scripts/tools/developers/update_bootloader.sh
```

```bash
kernel:[4.9.78-ti-r94]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-UART4-00A0.dtbo]
```
So by setting uboot_overlay_addr0/uboot_overlay_addr1 here in v4.9.x that disables config-pin, which makes bbb.io fail, you can either jump to v4.16.x+ based kernel, or define the UAR1/UAR4 via config-pin in v4.9.x (and thus removing the uboot_overlay_addr0/uboot_overlay_addr1 define in /boot/uEnv.txt)

Regards,

rjdyk, 2018-06-27:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/developers/update_bootloader.sh
bootloader:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]
Bootloader Recovery
mount: /dev/mmcblk1p1 is already mounted or /tmp/uboot busy
       /dev/mmcblk1p1 is already mounted on /
       /dev/mmcblk1p1 is already mounted on /tmp/uboot
```

```
debian@beaglebone:~/Desktop$ sudo python -i
Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO.GPIO as GPIO
>>>
>>> GPIO.setup("P8_12", GPIO.IN)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Set gpio mode failed, missing file or invalid permissions.
>>> exit()

```

closed 2018-06-28 by rjdyk