adafruit/adafruit-beaglebone-io-python #265: RuntimeError add_event_detect P8_7 and P8_8

edkirk30, 2018-07-02:
- Platform/operating system: Linux beaglebone 4.14.51-bone16 #1 PREEMPT Fri Jun 22 21:49:08 UTC 2018 armv7l GNU/Linux - Debian 9.2 iot image

- Python version:  '2.7.13 (default, Nov 24 2017, 17:33:09) \n[GCC 6.3.0 20170516]'

Getting RuntimeError: Failed to add edge detection when using add_event_detect on P8_7 and P8_8 but works fine with P8_9 and P8_10. input() executes but is reading high in all circumstances.

Test script:
`import Adafruit_BBIO.GPIO as GPIO

pin = "P8_7"

GPIO.setup(pin, GPIO.IN)
if GPIO.input(pin):
    print("HIGH")
else:
    print("LOW")

GPIO.add_event_detect(pin, GPIO.BOTH)

if GPIO.event_detected(pin):
    print("DETECT")`

edkirk30, 2018-07-02:
strace:
https://gist.github.com/0a4cb4851ec4de95c2517cca358835c5

pdp7, 2018-07-03:
@edkirk30 please paste the output of:
```
sudo /opt/scripts/tools/version.sh
```
and:
```
cat /boot/uEnv.txt
```

edkirk30, 2018-07-04:
root@beaglebone:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[ed7dd4c61a9c9307b08db8a7dcbaffac77a7f776]
eeprom:[A335BNLT00C05116BBBK1BA5]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.52-bone16]
nodejs:[v6.11.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20171009.0-0rcnee1~stretch+20171009]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.3.4-git20170602-0rcnee1~stretch+20170602]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.905044] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.910432] gpio-of-helper ocp:cape-universal: ready
END

edkirk30, 2018-07-04:
root@beaglebone:/tmp# cat /boot/uEnv.txt 
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.52-bone16
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks...
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##Example v3.8.x
#cape_disable=capemgr.disable_partno=
#cape_enable=capemgr.enable_partno=

##Example v4.1.x
#cape_disable=bone_capemgr.disable_partno=
#cape_enable=bone_capemgr.enable_partno=

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

pdp7, 2018-07-19:
The program you pasted runs OK on my BeagleBone with **BeagleBoard.org Debian Image 2018-06-17** and **4.14.49-ti-r54 kernel**.

I see you are running the **BeagleBoard.org Debian Image 2017-10-10** image.

Please try upgrade to the [https://beagleboard.org/latest-images](**Debian 9.4 2018-06-17 4GB SD IoT**):
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[31292bce1d4b35b497cc7a013d6d57e7e1f4f5c4]
eeprom:[A335BNLTBBG1BBG217110442]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Image 2018-06-17]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.49-ti-r54]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180628.0-0rcnee0~stretch+20180628]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.137530] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.148104] gpio-of-helper ocp:cape-universal: ready
END
```

pdp7, 2018-07-24:
@edkirk30  are you still experiencing this issue?

edkirk30, 2018-07-24:
Sorry for the delay. 9.4 seems to fix it (and I see you've updated the docs to reference 9.4 :+1:) and input() works as expected.

closed 2018-07-24 by edkirk30