adafruit/adafruit-beaglebone-io-python #266: Unable to import module gpio

sampathkmurthy, 2018-07-05:
Hallo @pdp7 ,

I am facing a problem with the gpio module while working with the Beagle Bone Blue. I am trying to set one of the gpio pin to high. While importing rcpy.gpio as gpio, there is an error while importing the module GPIO. The error showed, there is no module 
named 'gpiod'. I am surprised to see this gpiod. Is there any solution for this?, why am I unable to export the module. I am using the latest debian image on my BBBlue released on 27-06-2018. Please guide me with this issue.

Thanks and regards,
Sampath Kumar Krishna Murthy

pdp7, 2018-07-08:
@sampathkmurthy 
Please paste the exact error message.

And also please paste the output of:
```
sudo /opt/scripts/tools/version.sh
```
and:
```
cat /boot/uEnv.txt
```

sampathkmurthy, 2018-07-10:
@pdp7 
Thank you for your response. Here are my outputs:
- This is the output when i was trying to set one of the GPIO pin high.
/usr/local/lib/python3.5/dist-packages/rcpy-0.5.0-py3.5-linux-armv7l.egg/rcpy/__init__.py:116: UserWarning: > Robotics cape initialized
  warnings.warn('> Robotics cape initialized')
/usr/local/lib/python3.5/dist-packages/rcpy-0.5.0-py3.5-linux-armv7l.egg/rcpy/__init__.py:127: UserWarning: > Installing signal handlers
  warnings.warn('> Installing signal handlers')
Traceback (most recent call last):
  File "gpiotest2.py", line 1, in <module>
    import rcpy.gpio as gpio
  File "/usr/local/lib/python3.5/dist-packages/rcpy-0.5.0-py3.5-linux-armv7l.egg/rcpy/gpio.py", line 4, in <module>
    import gpiod
ImportError: No module named 'gpiod'
Initiating cleanup...
Done with cleanup

-This is the output of sudo /opt/scripts/tools/version.h

debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[31292bce1d4b35b497cc7a013d6d57e7e1f4f5c4]
eeprom:[A335BNLTBLA21712EL001427]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-06-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7                                                                                                                     c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.03-rc2-00002-g11d4fd]:[l                                                                                                                     ocation: dd MBR]
kernel:[4.14.49-ti-r54]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-                                                                                                                     00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180628.0-0rcnee0~stretch+20180628]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev us                                                                                                                     ers systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk                                                                                                                      weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait                                                                                                                      coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.052300] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper

-This is the output of cat /boot/uEnv.txt
debian@beaglebone:~$ cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.49-ti-r54
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.9.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.9.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

I was trying to set one of the GPIO pin high, but the module is not getting imported. I also installed libgpiod as the rcpy relies on it for GPIO access.

RobertCNelson, 2018-07-10:
```
This is the output when i was trying to set one of the GPIO pin high.
/usr/local/lib/python3.5/dist-packages/rcpy-0.5.0-py3.5-linux-armv7l.egg/rcpy/init.py:116: UserWarning: > Robotics cape initialized
warnings.warn('> Robotics cape initialized')
/usr/local/lib/python3.5/dist-packages/rcpy-0.5.0-py3.5-linux-armv7l.egg/rcpy/init.py:127: UserWarning: > Installing signal handlers
warnings.warn('> Installing signal handlers')
Traceback (most recent call last):
File "gpiotest2.py", line 1, in 
import rcpy.gpio as gpio
File "/usr/local/lib/python3.5/dist-packages/rcpy-0.5.0-py3.5-linux-armv7l.egg/rcpy/gpio.py", line 4, in 
import gpiod
ImportError: No module named 'gpiod'
Initiating cleanup...
Done with cleanup
```

python bindings where added in gpiod 1.1

https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/log/

Which haven't been packaged yet..

Regards,

sampathkmurthy, 2018-07-11:
@RobertCNelson 
Thank you for your response. Is there any other way to access gpio pin on BBblue without rcpy module.
Regards,

RobertCNelson, 2018-07-12:
@sampathkmurthy i've got something if you want to try testing it...

Grab this "buster-iot" image:

https://rcn-ee.net/rootfs/bb.org/testing/2018-07-08/buster-iot/bone-debian-buster-iot-armhf-2018-07-08-4gb.img.xz

run: (this will bring gpio 1.1.0 with python3 support) 
```bash
sudo apt update ; sudo apt upgrade ; sudo apt dist-upgrade
```

Then run:
```bash
cd /opt/source/rcpy
sudo python3 setup.py install
```
```bash
cd /opt/source/pyctrl
sudo python3 setup.py install
```
This will need some testing before i backport gpiod to stretch/bionic...

Regards,

sampathkmurthy, 2018-07-18:
@RobertCNelson 
Thank you for your reply. Perhaps, I used Adafruit_BBIO.GPIO library on Beaglebone Blue and it worked. I was successful in accessing GPIO pin.
Regards,

closed 2018-07-18 by sampathkmurthy