adafruit/adafruit-beaglebone-io-python #267: how receiving data thoght beagle bone uart

ududj, 2018-07-07:
Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.): pocket beagle / Debian 9.3 by terminal putty, Windows 10(64)

Python version (run python --version or python3 -version): python 2.7.13

Error message you are receiving, including any Python exception traces: my program stoped. i should cancel by "^c" keys.
I need that beagleBone interface with PIC16F877a through serial port. but i am not found receiving information with beaglebone through serial port by python.
all exemple that i found show me as transmit data, but not receive.

List the steps to reproduce the problem below (if possible attach code or commands
to run):

import Adafruit_BBIO.UART as uart
import serial

uart.setup("UART4")
ser = serial.Serial (port = "/dev/ttyO4", baudrate = 9600)
ser.close()
letra = raw_input("digite a letra que quer enviar:")
ser.open()
if ser.isOpen():
  ser.write (letra)
  print ser.read()
ser.close

could you help me?

pdp7, 2018-07-08:
@ududj  please paste the output of:
```
sudo /opt/scripts/tools/version.sh
```
and:
```
cat /boot/uEnv.txt
```

ududj, 2018-07-11:
i found sotution in:
https://github.com/zhanglongqi/python-study/blob/master/uart.py

i configurated timeout, and use readline.

thanks.

closed 2018-07-11 by ududj