adafruit/adafruit-beaglebone-io-python #268: Attribute error:adafruit i2c object has no attribute writeList

RadhaKrishnanme, 2018-07-11:
Hi,
I am trying to use Motor Bridge cape for stepper motor. On running the code, I received the following error:
Traceback (most recent call last):
  File "Valve_test.py", line 20, in <module>
    Valveopen()
  File "Valve_test.py", line 8, in Valveopen
    valve.StepperMotorAInit()
  File "/home/debian/tutorials/MotorBridge.py", line 233, in StepperMotorAInit
    WriteByte(TB_1A_MODE,TB_SPM) #Stepper
  File "/home/debian/tutorials/MotorBridge.py", line 205, in WriteByte
    MotorBridge.writeList(WriteMode,data)
AttributeError: 'Adafruit_I2C' object has no attribute 'writeList'

Any help in this regard is much appreciated.

pdp7, 2018-07-16:
@RadhaKrishnanme  were you able to find a solution?

RadhaKrishnanme, 2018-07-17:
Hi,
Yes, I managed to find a solution for the issue. Thanks