adafruit/adafruit-beaglebone-io-python #269: RuntimeError: Unable to setup ADC system

sampathkmurthy, 2018-07-12:
Hallo @pdp7 

I am trying to read the position of the linear actuator with the ADC port of Beagle Bone Blue.  I am importing Adafruit_BBIO.ADC as ADC and then I am setting up the ADC also, but I am getting the following runtimr error. This is my output on terminal.
Traceback (most recent call last):
  File "ADCtest.py", line 2, in <module>
    ADC.setup()
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper

please clarify this and is there a module rcpy.adc?, Is there any method to get the position of the actuator?

Thanks and Regards,
Sampath Kumar Krishna Murthy

pdp7, 2018-07-16:
@sampathkmurthy Hello, I'd like to know more about the setup on your BeagleBone.

Please paste the output of:
```sudo /opt/scripts/tools/version.sh```
and
```
cat /boot/uEnv.txt```

sampathkmurthy, 2018-07-17:
@pdp7 

Thank you for your response and here are the outputs:
1. output of sudo /opt/scripts/tools/version.sh

git:/opt/scripts/:[31292bce1d4b35b497cc7a013d6d57e7e1f4f5c4]
eeprom:[A335BNLTBLA21712EL001427]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-06-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7                                                                                                                     c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.03-rc2-00002-g11d4fd]:[l                                                                                                                     ocation: dd MBR]
kernel:[4.14.49-ti-r54]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-                                                                                                                     00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180628.0-0rcnee0~stretch+20180628]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev us                                                                                                                     ers systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk                                                                                                                      weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait                                                                                                                      coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.052227] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper

2. Output of cat /boot/uEnv.txt

#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.49-ti-r54
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Ov                                                                                                                     erlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.9.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-9-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.9.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

Regards,

sampathkmurthy, 2018-07-24:
@pdp7  ,

I am just running the same code as yours to test the Adafruit_BBIO.ADC and I also checked issue #127.  The difference is, I am using Beaglebone blue. My code is like this:
import Adafruit_BBIO.ADC as ADC
ADC.setup()
value =ADC.read("AIN0")
 when I run this code, the exception is thrown and I am using the latest released Debian image 9.4 (stretch).  Is there any method to remove this exception?

Regards,

pdp7, 2018-07-24:
@sampathkmurthy I just talked to @RobertCNelson about this issue and he suggested it might be an issue with an older bootloader on the eMMC.  This could prevent the correct U-Boot Overlay from being loaded which could prevent the ADC from being initialized correctly.

Please try running this:
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10 

sudo apt update 
sudo apt install --only-upgrade bb-cape-overlays 
sudo reboot 
```

sampathkmurthy, 2018-07-26:
@pdp7  and @RobertCNelson,

Thank you for the response, I executed these commands on my terminal. It is still throwing the exception. Is this problem due to the absence of device tree overlay for ADC?. I have two images running with the board, one is running with the Beaglebone blue device which is by default and the other one installed on micro-sd card. Currently, I am using the latest debian image burnt on micro-sd card slot. Is this cause an exception as you said in the previous comment ?. But tried executing your commands, but the exception is same. Please suggest me further.

Thanks and Regards,
Sampath Kumar Krishna Murthy.

pdp7, 2018-07-26:
@sampathkmurthy 
I think it would be helpful if you could run strace so that I can better understand what is happening.

Install strace (if it doesn't already exist):
```
sudo apt-get install strace
```
Then run strace, replace test.py with the name of your test program:
```
strace -f -o strace.log python test.py
```
Paste the contents of strace.log into a GitHub Gist and post the link in a reply here.

sampathkmurthy, 2018-07-27:
@pdp7 ,
Thank you for your reponse.
Here is the link to the contents of strace.log file.

https://gist.github.com/sampathkmurthy/33529effa6d13aaafa0f18605e8831df

Thanks and Regards,
Sampath Kumar Krishna Murthy.

sampathkmurthy, 2018-07-27:
@pdp7 ,

Just an update: The previous link does not contain whole data of strace.log file. Here is the link for new strace.log file. Please follow the link.

https://gist.github.com/sampathkmurthy/1da126257c38b19be657a1dce620c8b9

Regards,

pdp7, 2018-07-27:
@sampathkmurthy I think this is where it fails:
```
1086  open("/sys/devices/platform/bone_capemgr/slots", O_RDWR|O_LARGEFILE) = -1 ENOENT (No such file or directory)
```
This is curious though as it should no longer be checking for the slots file as @RobertCNelson has changed our Debian 9.x images to use the U-Boot Device Tree Overlays.  This means the bootloader hands a fully formed device tree to the Linux kernel and it no longer has to be managed using `/sys/devices/platform/bone_capemgr/slots`.

This makes me wonder if you are using an old version of Adafruit_BBIO.

Please run this command:
```
python -c "import Adafruit_BBIO; print Adafruit_BBIO"
```
The output should look like:
```
debian@beaglebone:~$ python -c "import Adafruit_BBIO; print Adafruit_BBIO"
<module 'Adafruit_BBIO' from '/usr/local/lib/python2.7/dist-packages/Adafruit_BBIO-1.0.10-py2.7-linux-armv7l.egg/Adafruit_BBIO/__init__.pyc'>
```

pdp7, 2018-07-27:
@sampathkmurthy could you try to please update to the latest version of Adafruit_BBIO using:
```sudo pip install --upgrade Adafruit_BBIO```

sampathkmurthy, 2018-08-06:
Hallo @pdp7 
Sorry for the late reply as I was on vacation. I tried upgrading the Adafruit_BBIO library. Still the problem resumes.
debian@beaglebone:~$ sudo pip install --upgrade Adafruit_BBIO
[sudo] password for debian:
Collecting Adafruit_BBIO
  Downloading https://files.pythonhosted.org/packages/28/62/7ecabb522a78606f0d657760257fe8703eb91d580d5a1381e0cea796a5d7/Adafruit_BBIO-1.0.10.tar.gz (68kB)
    100% |████████████████████████████████| 71kB 256kB/s
Building wheels for collected packages: Adafruit-BBIO
  Running setup.py bdist_wheel for Adafruit-BBIO ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-8ocbw191/Adafruit-BBIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpmzyoqpj1pip-wheel- --python-tag cp35:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for Adafruit-BBIO
  Running setup.py clean for Adafruit-BBIO
Failed to build Adafruit-BBIO
Installing collected packages: Adafruit-BBIO
  Found existing installation: Adafruit-BBIO 0.0.30
    Uninstalling Adafruit-BBIO-0.0.30:
      Successfully uninstalled Adafruit-BBIO-0.0.30
  Running setup.py install for Adafruit-BBIO ... done
Successfully installed Adafruit-BBIO-1.0.10

root@beaglebone:/home/debian/bin# emacs adctest.py
root@beaglebone:/home/debian/bin# python adctest.py
Traceback (most recent call last):
  File "adctest.py", line 2, in <module>
    ADC.setup()
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper

What might be causing this exception? I am totally stucked here. Please share your instruction further.

Regards,

pdp7, 2018-08-16:
@sampathkmurthy that is odd.  

Please paste the output off:
```
python -c "import Adafruit_BBIO; print Adafruit_BBIO"
```

And please try installing Adafruit_BBIO manually with these steps from the README:
```
sudo ntpdate pool.ntp.org
sudo apt-get update
sudo apt-get install build-essential python-dev python-pip -y
git clone git://github.com/adafruit/adafruit-beaglebone-io-python.git
cd adafruit-beaglebone-io-python
sudo python setup.py install
```

pdp7, 2018-08-21:
@sampathkmurthy are you still having this issue?

pdp7, 2018-08-23:
@sampathkmurthy please reopen if this is still an issue

closed 2018-08-23 by pdp7

JoeCarmelLopez, 2019-01-29:
Hi @pdp7 
I am also experiencing this problem on my BeagleBone Blue.
`debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLTBLA21736EL001620]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.05-rc1-00002-g35aecb]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.083190] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper

debian@beaglebone:~$ cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.71-ti-r80
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

debian@beaglebone:~$ python -c "import Adafruit_BBIO; print Adafruit_BBIO"
<module 'Adafruit_BBIO' from '/usr/local/lib/python2.7/dist-packages/Adafruit_BBIO-1.0.10-py2.7-linux-armv7l.egg/Adafruit_BBIO/__init__.pyc'>`

Here is the error again.
`debian@beaglebone:~$ python -c "import Adafruit_BBIO.ADC as ADC; ADC.setup()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper`

I've been looking for solutions for a few days but no luck. If this is already solved, I'm sorry and please direct me to the solution. Thank you.

RobertCNelson, 2019-01-29:
@JoeCarmelLopez 

```code
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2017.05-rc1-00002-g35aecb]:[location: dd MBR]
```
Your version of eMMC is too old, just run:

```code
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10 
```
Regards,

JoeCarmelLopez, 2019-01-30:
Thank you for the response @RobertCNelson 
I ran the command you gave and here is the result:
```
debian@beaglebone:~$ sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
10+0 records in
10+0 records out
10485760 bytes (10 MB, 10 MiB) copied, 0.993778 s, 10.6 MB/s
```
Then I checked after running:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLTBLA21736EL001620]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.083098] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
```
The bootloader line for th eMMC is gone.
Trying so setup the ADC still gives the error.
```
debian@beaglebone:~$ python -c "import Adafruit_BBIO.ADC as ADC; ADC.setup()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper
'''
Is there something I'm missing?

iliesaya, 2019-12-03:
Just here to say that I had the same issue on beaglebone black and i just had to **run with sudo**

```

riaps@bbb-9c4a:~$ python3 light.py
Traceback (most recent call last):
  File "light.py", line 6, in <module>
    ADC.setup()
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper

```

```
riaps@bbb-9c4a:~$ sudo python3 light.py

Reading         Volts
0.893040        1.607473
0.892796        1.607033
```

krupal756, 2020-04-07:
This is my first time on github

I am using Beaglebone Enhanced and I too ran into the same issue as @sampathkmurthy with the ADC Error, but I was able to find the problem and also solve it too.

Finding the problem:
I tried to google for the solution and found out this website (link below):
https://www.teachmemicro.com/beaglebone-black-adc/
wherein he has explained for the exact location to probe for ADC raw values on Beaglebone Black on the command line, which is: 

/sys/bus/iio/devices/iio:device0

When I tried to find the same on Enhanced and probed into /devices folder for iio:devices0, there were three folders iio:devices0,1 and 2. The in_voltage#_raw (# is the number) was missing, so I checked the same at other locations and found it in iio:device2.
I cross checked the Adafruit_BBIO library, to find out from which location the library was probing the raw values from, on github and found out the c file (link) on line no 58 which had the probing location:

adafruit-beaglebone-io-python/source/c_adc.c 

"strncat(adc_prefix_dir, "/sys/bus/iio/devices/iio:device0/in_voltage", sizeof(adc_prefix_dir));"

which will surely work for BBB, but may or may not work for other upgraded boards like Beaglebone Blue or Beaglebone Enhanced.

Solution:
What I did is I manually downloaded the complete Adafruit_BBIO library onto the board, and changed the location of the adc folder on the line as below (this is for Enhanced, you may need to find out yours)
 
"strncat(adc_prefix_dir, "/sys/bus/iio/devices/iio:device2/in_voltage", sizeof(adc_prefix_dir));

and then reinstalled the library. After which I restarted the system and ran the adc example from Adafruit and it worked just perfectly fine.

What I would like to suggest to @pdp7 to keep the flexibility of providing the adc folder location through the code, instead of fixed locations, as an update to the library, as most of the new boards as an additional input, if in case the manufacturer have these locations changed, we can fix it in our codes itself.