adafruit/adafruit-beaglebone-io-python #270: PWM change frequency error

tranpro, 2018-07-13:
Hi, I am currently using Beaglebone Black to control a solenoid valve using PWM. **At the moment everything works fine with the default 2k Hz frequency setting**. However, when I tried to change the frequency to others number, it stops working and gives me the error shown below.
```
  File "test.py", line 7, in <module>
    PWM.set_frequency(myPWM,F)
RuntimeError: Could not write to the frequency file
```
This is my test code 
`
import Adafruit_BBIO.PWM as PWM
import time
myPWM = "P9_14"
PWM.start(myPWM, 0, 2000)
for i in range(0,5):
    F= input ("freq?")
    PWM.set_frequency(myPWM,F)
    DC=input("What Duty Cycle Would You Like (0-100)? ")
    PWM.set_duty_cycle(myPWM, DC)
PWM.stop(myPWM)
PWM.cleanup()
`
My current kernel is : 4.14.52-ti-r60
And I used the debian image: BeagleBoard.org Debian Image 2018-07-02

Please help me with this problem ! Thank you very much!

pdp7, 2018-07-24:
@tranpro  are you still experiencing this issue?

pdp7, 2018-08-16:
no reply so I am closing issue.  Please reopen if still an issue for you.

closed 2018-08-16 by pdp7