adafruit/adafruit-beaglebone-io-python #272: Beaglebone sd card dependant

gahigg1, 2018-08-24:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **INSERT PLATFORM/OPERATING
  SYSTEM HERE**

- Python version (run `python -version` or `python3 -version`):  **INSERT PYTHON
  VERSION HERE**

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

RobertCNelson, 2018-08-24:
@gahigg1 

please run:

```bash
sudo /opt/scripts/tools/version.sh
```
and copy the output to this issue.

Regards,

gahigg1, 2018-08-25:
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[31292bce1d4b35b497cc7a013d6d57e7e1f4f5c4]
eeprom:[A335BNLT000C1827BBBG1716]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-06-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7                                     c6a]:[location: dd MBR]
kernel:[4.14.49-ti-r54]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-                                     00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180611.0-0rcnee0~stretch+20180611]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.075900] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.082744] gpio-of-helper ocp:cape-universal: ready
END
debian@beaglebone:~$

pdp7, 2018-08-27:
It would be helpful if you could run strace so that I can better understand what is happening.

Install strace (if it doesn't already exist):
```sudo apt-get install strace```

Then run strace, replace test.py with the name of your test program:
```strace -f -o strace.log python test.py```

Paste the contents of strace.log into a GitHub Gist and post the link in a reply here.

pdp7, 2018-08-31:
@gahigg1 are you still having this issue?

gahigg1, 2018-08-31:
Yes, I do not know what to do after I connect the network cable

pdp7, 2018-09-01:
I am using the Strech IoT image from 2018-07-15 on BeagleBone Black and your program runs ok:
```
debian@beaglebone:~$ sudo python test272.py
debian@beaglebone:~$ cat test272.py 
import Adafruit_BBIO.GPIO as GPIO
import Adafruit_BBIO.PWM as PWM
GPIO.setup("P8_10", GPIO.OUT)
GPIO.output("P8_10", GPIO.HIGH)
GPIO.cleanup()
PWM.start("P9_14",100,1000)
PWM.start("P9_16",100,1000)
PWM.start("P8_13",100,1000)
PWM.start("P9_22",100,1000)
PWM.start("P9_42",100,1000)
PWM.start("P9_21",100,1000)

debian@beaglebone:~$ cat /etc/dogtag 
BeagleBoard.org Debian Image 2018-07-15
```
Could you try the latest Debian Stretch IoT image?
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Stretch_Snapshot_iot

Alternatively, it would be helpful if you could run strace so that I can better understand what is happening.

Install strace (if it doesn't already exist):
```
sudo apt-get install strace
```
Then run strace, replace test.py with the name of your test program:
```
strace -f -o strace.log python test.py
```
Paste the contents of strace.log into a GitHub Gist and post the link in a reply here.

pdp7, 2018-09-06:
please reopen if you are still having this issue

closed 2018-09-06 by pdp7