adafruit/adafruit-beaglebone-io-python #273: PWM pins disabled with 4D LCD Cape w/ Beaglebone Black Rev C

gahigg1, 2018-08-30:
Hi,
I'm using Debian 9.4 with linux 4.14 kernal. Without the lcd attached, my code works utilizing 5 PWM pins. When the LCD is attached, I get an error only with the PWM pins, 'Unknown Error'. Also, when I try to configure them as pwm, an error pops up stating that no pinmux is found for each pin. One last thing, when I try :cat /sys/devices/platform/bone_capemgr/slots; an error says that no such file exist. 

PLEASE HELPPP
thanks

RobertCNelson, 2018-08-30:
@gahigg1 please run:

```bash
sudo /opt/scripts/tools/version.sh
```
Regards,

pdp7, 2018-08-30:
It is import to know what versions of software you have installed on your BeagleBone, especially the version of u-boot bootloader on the eMMC.

Please run:
```sudo /opt/scripts/tools/version.sh```

and copy the output to this issue.

`/sys/devices/platform/bone_capemgr/slots` has been deprecated by u-boot handling device tree overlays.

Please check out this description by @RobertCNelson:
**[Where did the slots file go?](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Where_did_the_slots_file_go.3F)**

Also, it would be good to check the pin mode of the pins you are trying to use.  For example:
```
debian@beaglebone:~$ config-pin -l p9.42 
default gpio gpio_pu gpio_pd gpio_input spi_cs spi_sclk uart pwm pru_ecap
debian@beaglebone:~$ config-pin -q p9.42 
P9_42 Mode: uart
debian@beaglebone:~$ config-pin p9.42 pwm
debian@beaglebone:~$ config-pin -q p9.42 
P9_42 Mode: pwm
```

gahigg1, 2018-08-30:
output from : sudo /opt/scripts/tools/version.sh

   
root@beaglebone:~# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[31292bce1d4b35b497cc7a013d6d57e7e1f4f5c4]
eeprom:[A335BNLT000C1827BBBG1716]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-06-17]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.49-ti-r54]
nodejs:[v6.14.3]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180611.0-0rcnee0~stretch+20180611]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~stretch+20180609]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BB-BONE-4D5C-01, coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.043304] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.044876] gpio-of-helper ocp:cape-universal: ready
END
root@beaglebone:~#

RobertCNelson, 2018-08-30:
@gahigg1 your in pretty good shape just run:

```bash
debian@test-bbb-2:~$ cd /opt/scripts/tools/
debian@test-bbb-2:/opt/scripts/tools$ git pull
Already up-to-date.
debian@test-bbb-2:/opt/scripts/tools$ sudo ./update_kernel.sh
debian@test-bbb-2:/opt/scripts/tools$ sudo apt install --only-upgrade
debian@test-bbb-2:/opt/scripts/tools$ sudo reboot
```
Regards,

gahigg1, 2018-08-30:
The network cable made the board not even turn on

pdp7, 2018-08-31:
@gahigg1 how are you powering the BeagleBone?

gahigg1, 2018-08-31:
I was using a 5V 2A power supply because of the LCD, but also using the micro usb to connect to my computer

pdp7, 2018-09-01:
> The network cable made the board not even turn on
@gahigg1 are you still having that issue?

pdp7, 2018-09-06:
please reopen if this is still an issue

closed 2018-09-06 by pdp7